evcc-io/evcc #3432: Add Stark in Strom Template

andig, 2022-05-20:
Hallo Dirk,
ja, Basis ist die EVSE DIN.
Kannst du einfach mittels des passenden Templates über den Konfigurator (evcc configure) einbinden.
Ansonsten findest du die Beschreibung auch hier: https://docs.evcc.io/docs/devices/chargers/#evse-din

Ein USB-RS485-Interface ist für sehr kleines Geld z. B. bei Reichelt zu bekommen. Alternativ kann man auch RS485-IP-Konverter verwenden. Die sind etwas teurer und machmal auch nicht so einfach richtig einzurichten.

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/3412#discussioncomment-2785810_

closed 2022-05-20 by andig

dirktebie, 2022-05-21:
danke,
hat geklappt mit der Anbindung.
die WB (SIS) spricht als EVSE-DIN mit evcc.

leider lässt sich der Ladestrom (Leistung) nicht in so kleinen Stufen einstellen, wie gedacht.
das kann an mehreren Sachen liegen:
- die WB schaltet ja nur alle 3 Phasen an/aus,
- der ID3 akzeptiert nur wenige Leistungsstufen (3kw, 4kw, 7,5kw) (2-Phasen),

als Beispiel:
laut evcc lädt das Auto jetzt mit 4,1kw, laut Connect-ID-App(Android) lädt der ID3 gerade mit 3kw.

Gruß,
Dirk

andig, 2022-05-21:
Du hast nur ganze A in der Steuerung…

dirktebie, 2022-05-21:
Nachtrag:
Meine Anlage besteht aus:
PV hat 7kwP
SMA-Tripower 7kw
SMA-Home-Manager 2.0

DIY-Speicher:
5kwh
Regler: Victron Multiplus 3000/1700/48
Raspi3B mit Victron-Venus zum Laden des Speichers.

Wallbox SIS-Easy 11kw
Raspi 4, 2GB mit Raspbian bullseye für die evcc.

dirktebie, 2022-05-21:
"Du hast nur ganze A in der Steuerung…"

sind nach meiner Rechnung ca. 4A Differenz...

dirktebie, 2022-05-21:
ich denke, ich habe den Fehler gefunden:

Das Auto lädt mit 6A (6 x 230V = 1380W).
Das Auto kann nur 2-Phasen AC Laden (2 x 1380W = 2760W).
Ist gerundet ~ 3kw (von der APP).
Die evcc geht von 3 Phasen a 6A aus (3 x 1380W = 4140W).
also zeigt mir evcc eine Ladeleistung von ~4,1kw an.

wenn man jetzt noch für das Auto die Anzahl der Phasen eintragen könnte, dann wäre die anzeige genauer...

andig, 2022-05-21:
> eintragen könnte

Kannst Du, siehe Doku ;) Alternativ Zähler zwischenschalten den wir auslesen können.

dirktebie, 2022-05-21:
Ein zusätzlicher Zähler kostet auch Geld.
Z.Bsp. shelly 3em ca. 180€.
3x shelly 1pm je 20€ (60€ für 3 phasen, aber noch nicht direkt auswählbar).

da es aber viele PKW geben soll, die lediglich 2-Pasen AC Laden können, dachte ich, wäre dort der Ansatz gut.

ps: in der section vehicle gebe ich Zugangsdaten für die VW-Cloud an.
wenn diese Daten gar nicht genutzt werden können, kann ich die natürlich besser löschen (große Bedenken wegen Missbrauch).

premultiply, 2022-05-21:
Selbst das Shelly 3EM kostet auch nur so rund 120 EUR.
Übliche Modbus-Zähler liegen ebenso in dem Bereich.
Wenn sie gleich in der Wallbox mit drin sind sind sie oft wesentlich günster.

dirktebie, 2022-05-29:
Hallo,
Urlaub zu Ende.
Hab mir ein shelly 3em bestellt.
Dann ist die Statistik unabhängig vom Auto, das geladen wird.

dirktebie, 2022-05-30:
> Kannst Du, siehe Doku ;) Alternativ Zähler zwischenschalten den wir auslesen können.

Parameter "phases" gefunden und bei "vehicle" eingefügt.
jetzt passt die Leistungsangabe annähernd.

da kommt die nächste Frage auf:
wenn man Überschuss (mode "PV") laden möchte, rechnet evcc jetzt mit 2-phasen ?

Gruß,
Dirk

premultiply, 2022-05-30:
Ja, wenn du beim Fahrzeug `phases: 2` eingetragen hast.