evcc-io/evcc #3435: lp.effectiveCurrent() liefert ggf falschen Wert

hoermto, 2022-05-20:
### Describe the bug

Fehlersituationen:
- StatusD = Laden mit Lüfter gibt immer 0 zurück
- falls ein Charger mit verdrehten Phasen angeschlossen ist (falsch oder rotierend bei mehreren), dann kann ein 3 phasiger Lader auch nur eine Phase benutzen, die nicht Phase 0 ist (zB im Klimatisierungsmodus). Oder ein 3 Phasen Charger hat ein 2Phasen Auto, wo das Ladekabel anders angeschlagen ist.
-  Funktion sollte daher den Maximalwert der 3 Phasen liefern, und nicht nur den von Phase 0

### Steps to reproduce

code review

### Configuration details

standard config

### Log details

no specific log

### What type of operating system are you running?

Docker container

### Version

0.92

andig, 2022-05-20:
> StatusD = Laden mit Lüfter gibt immer 0 zurück

Was heisst das?

> falls ein Charger mit verdrehten Phasen angeschlossen ist (falsch oder rotierend bei mehreren), dann kann ein 3 phasiger Lader auch nur eine Phase benutzen, die nicht Phase 0 ist (zB im Klimatisierungsmodus). Oder ein 3 Phasen Charger hat ein 2Phasen Auto, wo das Ladekabel anders angeschlagen ist.

Diesen Fehlerzustand erkennen und melden wir

> Funktion sollte daher den Maximalwert der 3 Phasen liefern, und nicht nur den von Phase 0

Warum? Zu diesem Zeitpunkt sind wir schon im Fehlerstatus?

hoermto, 2022-05-20:
StatusD: laut `api/api.go`:
```
StatusC    ChargeStatus = "C" // Fzg. angeschlossen:   ja    Laden aktiv:   ja    - Laden
StatusD    ChargeStatus = "D" // Fzg. angeschlossen:   ja    Laden aktiv:   ja    - Laden mit Lüfter
```
Damit würde in C und D Strom fliessen, welcher nach meinem Verständnis von der Funktion berücksichtigt werden muss.

Das 3Phasen Thema habe ich verstanden dass es im Code keine Korrektur braucht.

andig, 2022-05-20:
Stimmt. Ich hab D nie gesehen- denke das ist was aus der Vergangenheit. Auch der Rest vom Code handled das nicht. PR welcome wenn's wichtig ist.

premultiply, 2022-05-20:
Status D ist veraltet. War für Fahrzeuge mit Bleiakkus. Wird von quasi allen aktuellen Wallboxen ohnehin intern direkt abgelehnt und von keinem Fahrzeug heute mehr verwendet. Daher kann man das theoretisch zwar einbauen (der Vollständigkeit halber) aber ohne praktische Relevanz.

Nicht vergessen dass das bei manchen Charger-Implementierungen auch gar nicht erst als gültiger Status gemappt ist.

hoermto, 2022-05-20:
habe den Commit so angepasst dass StatusD wie C als charging betrachtet wird. Alternativ kann ich den auch komplett raushauen. Da kann ich aber die Konsequenzen nicht abschätzen.
Your choice.

andig, 2022-05-20:
Ich schließe das mal als wontfix. Lass uns den Aufwand lieber in echte Issues stecken. Status D gabs jetzt in 2,5 Jahren nie...

closed 2022-05-20 by andig

hoermto, 2022-05-20:
> Stimmt. Ich hab D nie gesehen- denke das ist was aus der Vergangenheit. Auch der Rest vom Code handled das nicht. PR welcome wenn's wichtig ist.

Ich erlaube mir einen separaten PR, wo StatusD dann komplett rausfliegt. Das wäre dann konsequent.