evcc-io/evcc #3439: 'tasmota': configured relay channel doesn't exist on device`

andig, 2022-05-20:
evcc gerade aktualisiert. Habe als grid meter einen Tasmota Lesekopf. Bisher hat das mit dem template Tasmota prima funktioniert. 
Jetzt kommt: `FATAL 2022/05/20 14:44:59 cannot create meter 'grid': cannot create meter 'template': cannot create meter 'tasmota': configured relay channel doesn't exist on device`
Habe es jetzt per HTTP plugin wieder eingebunden. Tut wieder.

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/3428_

thierolm, 2022-05-20:
Einfach im Template den neuen channel Parameter ergänzen ...

Siehe https://github.com/evcc-io/evcc/pull/3368 

Beispiel: 

```yaml
chargers:
- type: template
  template: tasmota 
  host: 192.168.178.95  
  channel: 1  
  standbypower: 15  
  name: wallbox1
```

thierolm, 2022-05-20:
Auch `evcc configure` generiert das Template korrekt:

```bash
thierolm@accffmmt:~/evcc$ ./evcc configure

Die nächsten Schritte führen durch die Einrichtung einer Konfigurationsdatei für evcc.
Beachte dass dieser Prozess nicht alle möglichen Szenarien berücksichtigen kann.
Durch Drücken von CTRL-C kann der Prozess abgebrochen werden.

ACHTUNG: Diese Funktionalität hat experimentellen Status!
  D.h. es kann möglich sein, dass die hiermit erstellte Konfigurationsdatei
  in einem Update nicht mehr funktionieren könnte und neu erzeugt werden müsste.
  Wir freuen uns auf euer Feedback auf https://github.com/evcc-io/evcc/discussions/

Auf geht`s:

 In welchem Modus soll die Konfiguration durchgeführt werden? Fortgeschrittener Modus (Detailliertere Fragen, erfordert jedoch technisches Know-How)

 Was möchtest du machen? Ein einzelnes Gerät konfigurieren (muss manuell in eine Konfigurationsdatei eingetragen werden!)

- Ein Gerät konfigurieren

Wähle eines der folgenden Gerätekategorien aus

 Wähle eines der folgenden Gerätekategorien aus Wallbox

- Konfiguration Wallbox

 Wähle eine Wallbox: Tasmota [Schaltbare Steckdosen]

Führe folgende Einstellungen durch:

 IP-Adresse oder Hostname 192.168.XXX.XXX
 Benutzerkonto 
 Passwort [? for help] 
 Nummer des Schalt-Kanals 1
 Standby-Leistung in W 15

Teste die Konfiguration von Tasmota [Schaltbare Steckdosen] ...

Tasmota [Schaltbare Steckdosen] Tasmota [Schaltbare Steckdosen] wurde erfolgreich hinzugefügt.

Die Konfiguration lautet:

  type: template
  template: tasmota 
  host: 192.168.XXX.XXX  
  channel: 1  
  standbypower: 15  
  name: wallbox1
```

andig, 2022-05-20:
Ich fix es morgen. Mit nem Defaultwert ist es abwärtskompatibel.

thierolm, 2022-05-20:
PR #3445 ist raus und funktioniert. Hab's mit mit meiner Sonoff Pow getestet.
Sorry, hatte deinen einen Review Kommentar übersehen...

VolkerK62, 2022-05-21:
Sorry. Funktioniert nicht. Hab das aktuelle Nightly drauf (da ist es doch schon drin, oder?)

yaml:
```
meters:
- name: grid
  type: template
  template: tasmota
  usage: grid
  host: 192.168.178.27
  channel: 1
  #type: custom   # IR Lesekopf Tasmota
  #power:
    #source: http
    #uri: http://192.168.178.27/cm?cmnd=Status%208
    #jq: .StatusSNS.ENERGY.Power
```

das sagt `evcc meter`
```
i@raspberrypi:~ $ evcc meter
[main  ] INFO 2022/05/21 06:20:24 evcc 0.92 (4121ad43)
[main  ] INFO 2022/05/21 06:20:24 using config file /etc/evcc.yaml
[tasmota] TRACE 2022/05/21 06:20:24 GET http://192.168.178.27/cm?cmnd=Status+0&password=&user=
[tasmota] TRACE 2022/05/21 06:20:24 {"Status":{"Module":0,"DeviceName":"cmatic","FriendlyName":["cmatic"],"Topic":"cmatic_51ACD8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"cmatic","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"24T10:56:37","StartupUTC":"2022-04-26T17:23:48","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2022-03-06T13:33:46","SaveCount":52,"SaveAddress":"F8000"},"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2022.01.19 17:56:20","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"351/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["KAS3","KAS1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]},"StatusMEM":{"ProgramSize":679,"Free":324,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000407","87DAC787","043E8001","000000CF","010013C0","C000F989","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,53"},"StatusNET":{"Hostname":"cmatic-51ACD8-3288","IPAddress":"192.168.178.27","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"40:91:51:51:AC:D8","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0},"StatusMQT":{"MqttHost":"cmatic","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_51ACD8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-05-21T04:20:25","Local":"2022-05-21T05:20:25","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:01","Sunset":"20:32"},"StatusSNS":{"Time":"2022-05-21T05:20:25","ENERGY":{"Power":68,"Total":6704,"Total_out":12676}},"StatusSTS":{"Time":"2022-05-21T05:20:25","Uptime":"24T10:56:37","UptimeSec":2112997,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":2,"SSId":"KAS1","BSSId":"DC:15:C8:9A:61:DE","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":13,"Downtime":"0T00:00:23"}}}
[main  ] FATAL 2022/05/21 06:20:24 cannot create meter 'grid': cannot create meter 'template': cannot create meter 'tasmota': configured relay channel doesn't exist on device
pi@raspberrypi:~ $ 
```

andig, 2022-05-21:
Bitte nightly von heute nochmal testen.

closed 2022-05-21 by andig

thierolm, 2022-05-21:
@VolkerK62 : Das ist offensichtlich ein neues "Flavor of Tasmota Device". Was das für eine Hardware? Kann es sein, dass du damit nur Strom messen kannst, aber nicht schalten? Dein Kommentar "IR Lesekopf Tasmota" in deiner Config deutet datauf hin.

Im Trace deiner Status Response oben, gibt es KEINEN `POWER` Switch Status  ... 
```yaml
  "StatusSTS": {
    "Time": "2022-05-21T05:20:25",
    "Uptime": "24T10:56:37",
    "UptimeSec": 2112997,
    "Heap": 19,
    "SleepMode": "Dynamic",
    "Sleep": 50,
    "LoadAvg": 19,
    "MqttCount": 0,
    "Wifi": {
      "AP": 2,
      "SSId": "KAS1",
      "BSSId": "DC:15:C8:9A:61:DE",
      "Channel": 11,
      "Mode": "11n",
      "RSSI": 58,
      "Signal": -71,
      "LinkCount": 13,
      "Downtime": "0T00:00:23"
    }
```
Deswegen die Fehlermeldung ... (nur zum Messen brauchst du ja keinen Schalter ...)

Bei Tasmota Hardware mit Schalter sieht das zumindest so aus:
```yaml
  "StatusSTS": {
    "Time": "2022-05-21T21:32:18",
    "Uptime": "19T02:59:10",
    "UptimeSec": 1652350,
    "Heap": 24,
    "SleepMode": "Dynamic",
    "Sleep": 50,
    "LoadAvg": 19,
    "MqttCount": 0,
    "POWER": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "THIEROLF-WLAN",
      "BSSId": "1C:ED:6F:A1:C1:F8",
      "Channel": 6,
      "Mode": "11n",
      "RSSI": 80,
      "Signal": -60,
      "LinkCount": 1,
      "Downtime": "0T00:00:05"
    }
  }
```
D.h. auch mit dem neuen nightly wirst du den Fehler bekommen.

thierolm, 2022-05-21:
Lass uns mal in der Tasmota Doku suchen, ob man an einem der Status Response Parameter erkennen kann, dass es keine aktiven Schalter gibt, sondern nur Sensor Werte ... Hab' auf die Schnelle nichts identifizieren können.

VolkerK62, 2022-05-22:
@thierolm  richtig, kein Schalter. Das ist ein IR-Lesekopf, mit dem ich meinen Hauszähler auslese. #2603 Ich glaube damals (da war ich Neuling) stand Tasmota in der Doku (Hausinstallation) nicht unter schaltbare Steckdosen. Das Template hat wunderbar funktioniert. Aber mittlerweile kenn ich mich etwas besser aus. Ich habe es jetzt mit HTTP Plugin eingebunden. Wegen mir braucht ihr da nix ändern. Unter den Begriff "Tasmota" kann sich ja sehr viel verbergen. Alle Fälle in einem Template abzubilden stelle ich mir schwierig vor.

thierolm, 2022-05-22:
@VolkerK62 Hab' da schon den Ergeiz alles über das Template abzubilden ;-)
Schau mal den #3459 oben. Wenn @andig ihn merged, solltest du mit dem nächsten nightly keine Probleme mehr mit dem Template bekommen.

VolkerK62, 2022-05-22:
Hätt ich mir ja denken können ;-) 
Ich werd es dann testen.
Danke

thierolm, 2022-05-22:
Eine Bitte: Kannst du kurz Infos zu deiner Tasmota grid Meter Lösung sharen? Ich suche noch nach einer einfachen Möglichkeit für den Stromzähler bei meinen Eltern ... (Muss auch mal schauen, was die für einen Stromzähler haben).

VolkerK62, 2022-05-22:
Ich habe damals diese Teil gekauft. https://cmatic.de/?accessid=2&id=127
Damals gab es die von "Hichie" nur mit USB und nicht mit wlan. 
Meine PV ist von 2019 und der neue Zähler hat eine IR-Schnittstelle.
Das war für ich der einfachste Weg. Mein SolarLog funzt nicht richtig, beim powerfox störte mich die cloud und für den Shelly 3EM hätt ich im vollen Zählerschrank runfummeln müssen.

premultiply, 2022-05-22:
Powerfox Poweropti ?
Dürfte sogar deutlich preiswerter sein.

VolkerK62, 2022-05-22:
@thierolm   hab 0.93 installiert. Tasmota Template als grid meter funktioniert wieder :-)   Prima. Danke