evcc-io/evcc #3444: Zielladen Unterschiedliches Datum Local zu UTC nach Mitternacht

Chris591, 2022-05-20:
### Describe the bug

Zeitpunkt liegt in der Vergangenheit Fehlermeldung in der UI
[image]

### Steps to reproduce

Zeitzone <> UTC (Vermutung das es auf UTC gerechnet wird)
lokale Zeit 0:45
versucht man dann ein Zielladen heute um 7:45 zu planen, kann man es nicht speichern da es in der Vergangenheit ist.
man muss es für morgen 7:45 planen damit es am selben tag um 7:45 fertig ist.

### Configuration details

I guess it's not required as it's only a date calculation issue

### Log details

no log available as you can't save it

### What type of operating system are you running?

Linux

### Version

See in screenshot if needed

andig, 2022-05-20:
Aktuelles nightly? Aonst bitte nochmal dort probieren.

Chris591, 2022-06-25:
[image]

Heute morgen nochmal getestet mit dem nightly von gestern.

Immernoch der gleiche Fehler.
Das Datum heute ist zwar richtig aber man kann die Uhrzeit nicht setzen da sie angeblich in der Vergangenheit ist.

andig, 2022-11-03:
@naltatis fixen oder schließen?

naltatis, 2022-11-03:
Lass ruhig offen. Ist ein Edge Case aber sollten wir trotzdem sauber machen.

PR #5164 (2022-11-17): Improve target time handling
fixes #3444
closed 2022-11-17 by naltatis