evcc-io/evcc #3457: Add confirmation for TargetSOC change

ExecuteGK, 2022-05-21:
When operating the UI from a mobile device, it happens quite fast that the target SOC get's changed by an unintentional touch at the UI. specially when swiping between multiple chargers at one site. If the vehicle has reached the "unintentional" SOC already, charging stops immediately and has to be re-started by changing the targetSOC back to it's original value. 

As a solution, it would be beneficial if a change of the targetSOC requires an additional confirmation. At least when using it from a mobile device.

Regards,

Sebastian

andig, 2022-05-22:
An additional confirmation is not the way to go. We're not MS Windows ;)

andig, 2022-05-22:
Duplikat zu https://github.com/evcc-io/evcc/issues/3272

closed 2022-05-22 by andig

ExecuteGK, 2022-05-22:
Then go for push and hold. Setting it unintentionally is bad.Sebastian.Von meinem/meiner Galaxy gesendet

  #3457)