evcc-io/evcc #3469: template for Fronius Symo inverters

Macrisu, 2022-05-22:
**Fronius Symo 1/3 phase** + smart meter are very popular, but a template is missing from the docs.

This is my working template for 3 phase inverters, please add it to docs.

It's working on 
- Fronius Symo 8.2-3 with 63A-3 Smartmeter (old smartmeter, the new one it's TS 65A-3)
- Fronius Symo 20-3 with TS 5kA Smartmeter

```
meters:  
- name: grid
  type: modbus
  model: sunspec
  uri: 192.168.1.xxx:502 #inverter fix IP
  id: 240
  power: 213:W # sunspec model 213 meter

- name: pv
  type: modbus
  model: sunspec
  uri: 192.168.1.xxx:502 #inverter fix IP (same ip as grid meter)
  id: 1
```

Modbus data export must be activated on the inverter (service menu) as in this image:
https://user-images.githubusercontent.com/16325429/160246056-d8a0f13a-2a2f-4cd0-a599-8bdee74674a7.jpg

andig, 2022-05-22:
https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-gen24.yaml  Doesn‘t work for you?

/cc @premultiply

Macrisu, 2022-05-22:
> https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-gen24.yaml Doesn‘t work for you?
> 
> /cc @premultiply

evcc 9.93

Working config

```
meters:
- name: grid_v28
  type: modbus
  model: sunspec
  uri: 192.168.1.65:502
  id: 240
  power: 213:W # sunspec model 213 meter

- name: pv_v28
  type: modbus
  model: sunspec
  uri: 192.168.1.65:502
  id: 1
```
```
grid_v28
--------
[modbus] TRACE 2022/05/22 17:17:11 modbus: send 00 10 00 00 00 06 f0 03 9c 87 00 7c
[modbus] TRACE 2022/05/22 17:17:11 modbus: recv 00 10 00 00 00 fb f0 03 f8 41 57 85 1f 40 9e 4d d3 40 a5 4f df 40 56 d9 17 43 70 22 22 43 6c b3 33 43 71 4c cd 43 72 66 66 43 cf f7 77 43 cf 00 00 43 d1 73 33 43 cf 73 33 42 48 00 00 c5 48 7e 14 c4 91 f8 a4 c4 9b 43 d7 c4 47 7f 5c 45 48 e0 00 44 92 5e 8f 44 9b d1 c9 44 4b 6f 28 c3 59 02 8f c1 8c 28 f6 c2 8b 38 52 c3 01 e1 48 bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 bf 7a e1 48 48 3b ec 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 46 28 14 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 00 00 00 00
Power: -3208W

pv_v28
------
[modbus] TRACE 2022/05/22 17:17:11 modbus: send 00 11 00 00 00 06 01 03 9c 87 00 3c
[modbus] TRACE 2022/05/22 17:17:11 modbus: recv 00 11 00 00 00 7b 01 03 78 41 70 f5 c2 40 a0 a3 d7 40 a0 a3 d7 40 a0 a3 d7 43 cf 19 9a 43 d0 33 33 43 ce b3 33 43 6c 99 9a 43 71 99 9a 43 70 00 00 45 60 d0 00 42 48 14 7b 45 60 d0 1c 40 e0 00 00 c2 c7 ff e7 48 51 21 81 7f c0 00 00 7f c0 00 00 45 6a 04 5d 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 00 04 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Power: 3597W
```

**New test config**

```
meters:
  - name: grid_v28
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 
    
  - name: pv_v28
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 
```
Result:

```
./evcc meter --log trace
[main  ] INFO 2022/05/22 17:25:39 evcc 0.93
[main  ] INFO 2022/05/22 17:25:39 using config file /volume1/docker/evcc28/evcc.yaml
[modbus] TRACE 2022/05/22 17:25:40 modbus: send 00 01 00 00 00 06 c8 03 9c 40 00 02
[modbus] TRACE 2022/05/22 17:25:40 modbus: recv 00 01 00 00 00 03 c8 83 0b
[modbus] TRACE 2022/05/22 17:25:40 modbus: send 00 02 00 00 00 06 c8 03 c3 50 00 02
[modbus] TRACE 2022/05/22 17:25:40 modbus: recv 00 02 00 00 00 03 c8 83 0b
[modbus] TRACE 2022/05/22 17:25:40 modbus: send 00 03 00 00 00 06 c8 03 00 00 00 02
[modbus] TRACE 2022/05/22 17:25:40 modbus: recv 00 03 00 00 00 03 c8 83 0b
[main  ] FATAL 2022/05/22 17:25:40 cannot create meter 'grid_v28': cannot create meter 'template': cannot create meter 'modbus': not a SunSpec device
```

andig, 2022-05-22:
Does pv work if bot grid? Seems you have a different slave id for grid.

Macrisu, 2022-05-22:
There is no id in the template

https://docs.evcc.io/docs/devices/meters#symo-gen24-plus

Macrisu, 2022-05-22:
```
meters:
  - name: grid_v28
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 
    id: 240
    
  - name: pv_v28
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 
    id: 1
```

```
root@Xpenology_P:/volume1/docker/evcc28# ./evcc meter --log trace
[main  ] INFO 2022/05/22 17:57:53 evcc 0.93
[main  ] INFO 2022/05/22 17:57:53 using config file /volume1/docker/evcc28/evcc.yaml
[modbus] TRACE 2022/05/22 17:57:53 modbus: send 00 01 00 00 00 06 c8 03 9c 40 00 02
[modbus] TRACE 2022/05/22 17:57:53 modbus: recv 00 01 00 00 00 03 c8 83 0b
[modbus] TRACE 2022/05/22 17:57:53 modbus: send 00 02 00 00 00 06 c8 03 c3 50 00 02
[modbus] TRACE 2022/05/22 17:57:53 modbus: recv 00 02 00 00 00 03 c8 83 0b
[modbus] TRACE 2022/05/22 17:57:53 modbus: send 00 03 00 00 00 06 c8 03 00 00 00 02
[modbus] TRACE 2022/05/22 17:57:53 modbus: recv 00 03 00 00 00 03 c8 83 0b
[main  ] FATAL 2022/05/22 17:57:53 cannot create meter 'grid_v28': cannot create meter 'template': cannot create meter 'modbus': not a SunSpec device
```

Macrisu, 2022-05-22:
I've tried fronius-solarapi-v1

```
meters:
  - name: grid_v28
    type: template
    template: fronius-solarapi-v1
    usage: grid
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 

    
  - name: pv_v28
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 
```

Results...

```
./evcc meter --log trace
[main  ] INFO 2022/05/22 18:06:06 evcc 0.93
[main  ] INFO 2022/05/22 18:06:06 using config file /volume1/docker/evcc28/evcc.yaml
pv_v28
------
[http  ] TRACE 2022/05/22 18:06:06 GET http://192.168.1.65/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[http  ] TRACE 2022/05/22 18:06:06 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 123,
               "E_Day" : 42690,
               "E_Total" : 216087.015625,
               "E_Year" : 216087.40625,
               "P" : 2657
            }
         },
         "Site" : {
            "E_Day" : 42690,
            "E_Total" : 216087.015625,
            "E_Year" : 216087.40625,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : -2118.52,
            "P_Load" : -538.48000000000002,
            "P_PV" : 2657,
            "rel_Autonomy" : 100,
            "rel_SelfConsumption" : 20.266465939028979
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2022-05-22T18:06:12+03:00"
   }
}
Power: 2657W

grid_v28
--------
[http  ] TRACE 2022/05/22 18:06:06 GET http://192.168.1.65/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[http  ] TRACE 2022/05/22 18:06:06 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 123,
               "E_Day" : 42690,
               "E_Total" : 216087.015625,
               "E_Year" : 216087.40625,
               "P" : 2657
            }
         },
         "Site" : {
            "E_Day" : 42690,
            "E_Total" : 216087.015625,
            "E_Year" : 216087.40625,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : -2118.52,
            "P_Load" : -538.48000000000002,
            "P_PV" : 2657,
            "rel_Autonomy" : 100,
            "rel_SelfConsumption" : 20.266465939028979
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2022-05-22T18:06:12+03:00"
   }
}
Power: -2119W
```

andig, 2022-05-22:
I meant 

    usage: pv

not solar api.

Macrisu, 2022-05-22:
```
meters:
  - name: grid_v28
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 
    
  - name: pv_v28
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 

```

```
./evcc meter --log trace
[main  ] INFO 2022/05/22 18:35:00 evcc 0.93
[main  ] INFO 2022/05/22 18:35:00 using config file /volume1/docker/evcc28/evcc.yaml
[modbus] TRACE 2022/05/22 18:35:00 modbus: send 00 01 00 00 00 06 c8 03 9c 40 00 02
[modbus] TRACE 2022/05/22 18:35:00 modbus: recv 00 01 00 00 00 03 c8 83 0b
[modbus] TRACE 2022/05/22 18:35:00 modbus: send 00 02 00 00 00 06 c8 03 c3 50 00 02
[modbus] TRACE 2022/05/22 18:35:00 modbus: recv 00 02 00 00 00 03 c8 83 0b
[modbus] TRACE 2022/05/22 18:35:00 modbus: send 00 03 00 00 00 06 c8 03 00 00 00 02
[modbus] TRACE 2022/05/22 18:35:00 modbus: recv 00 03 00 00 00 03 c8 83 0b
[main  ] FATAL 2022/05/22 18:35:00 cannot create meter 'grid_v28': cannot create meter 'template': cannot create meter 'modbus': not a SunSpec device
```

andig, 2022-05-22:
Remove grid. We know it doesn‘t work. Please test pv.

Macrisu, 2022-05-22:
```
meters:
#   - name: grid_v28
#     type: template
#     template: fronius-gen24
#     usage: grid
#     host: 192.168.1.65 # IP-Adresse oder Hostname
#     port: 502 # Port # Optional 

    
  - name: pv_v28
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.1.65 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 
```

results:

```
./evcc meter --log trace
[main  ] INFO 2022/05/22 18:47:04 evcc 0.93
[main  ] INFO 2022/05/22 18:47:04 using config file /volume1/docker/evcc28/evcc.yaml
[modbus] TRACE 2022/05/22 18:47:04 modbus: send 00 01 00 00 00 06 01 03 9c 40 00 02
[modbus] TRACE 2022/05/22 18:47:04 modbus: recv 00 01 00 00 00 07 01 03 04 53 75 6e 53
[modbus] TRACE 2022/05/22 18:47:04 modbus: send 00 02 00 00 00 06 01 03 9c 42 00 02
[modbus] TRACE 2022/05/22 18:47:04 modbus: recv 00 02 00 00 00 07 01 03 04 00 01 00 41
[modbus] TRACE 2022/05/22 18:47:04 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
[modbus] TRACE 2022/05/22 18:47:05 modbus: recv 00 03 00 00 00 07 01 03 04 00 71 00 3c
[modbus] TRACE 2022/05/22 18:47:05 modbus: send 00 04 00 00 00 06 01 03 9c c3 00 02
[modbus] TRACE 2022/05/22 18:47:05 modbus: recv 00 04 00 00 00 07 01 03 04 00 78 00 1a
[modbus] TRACE 2022/05/22 18:47:05 modbus: send 00 05 00 00 00 06 01 03 9c df 00 02
[modbus] TRACE 2022/05/22 18:47:05 modbus: recv 00 05 00 00 00 07 01 03 04 00 79 00 1e
[modbus] TRACE 2022/05/22 18:47:05 modbus: send 00 06 00 00 00 06 01 03 9c ff 00 02
[modbus] TRACE 2022/05/22 18:47:05 modbus: recv 00 06 00 00 00 07 01 03 04 00 7a 00 2c
[modbus] TRACE 2022/05/22 18:47:05 modbus: send 00 07 00 00 00 06 01 03 9d 2d 00 02
[modbus] TRACE 2022/05/22 18:47:05 modbus: recv 00 07 00 00 00 07 01 03 04 00 7b 00 18
[modbus] TRACE 2022/05/22 18:47:05 modbus: send 00 08 00 00 00 06 01 03 9d 47 00 02
[modbus] TRACE 2022/05/22 18:47:06 modbus: recv 00 08 00 00 00 07 01 03 04 00 a0 00 30
[modbus] TRACE 2022/05/22 18:47:06 modbus: send 00 09 00 00 00 06 01 03 9d 79 00 02
[modbus] TRACE 2022/05/22 18:47:06 modbus: recv 00 09 00 00 00 07 01 03 04 ff ff 00 00
[modbus] TRACE 2022/05/22 18:47:06 modbus: send 00 0a 00 00 00 06 01 03 9c 44 00 42
[modbus] TRACE 2022/05/22 18:47:06 modbus: recv 00 0a 00 00 00 87 01 03 84 46 72 6f 6e 69 75 73 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 79 6d 6f 20 38 2e 32 2d 33 2d 4d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 2e 32 31 2e 34 2d 31 00 00 00 00 00 00 00 00 30 2e 33 2e 32 37 2e 32 00 00 00 00 00 00 00 00 33 31 33 32 39 35 30 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 71
[modbus] TRACE 2022/05/22 18:47:06 modbus: send 00 0b 00 00 00 06 01 03 9c 40 00 02
[modbus] TRACE 2022/05/22 18:47:06 modbus: recv 00 0b 00 00 00 07 01 03 04 53 75 6e 53
[modbus] TRACE 2022/05/22 18:47:06 modbus: send 00 0c 00 00 00 06 01 03 9c 42 00 02
[modbus] TRACE 2022/05/22 18:47:07 modbus: recv 00 0c 00 00 00 07 01 03 04 00 01 00 41
[modbus] TRACE 2022/05/22 18:47:07 modbus: send 00 0d 00 00 00 06 01 03 9c 85 00 02
[modbus] TRACE 2022/05/22 18:47:07 modbus: recv 00 0d 00 00 00 07 01 03 04 00 71 00 3c
[modbus] TRACE 2022/05/22 18:47:07 modbus: send 00 0e 00 00 00 06 01 03 9c c3 00 02
[modbus] TRACE 2022/05/22 18:47:07 modbus: recv 00 0e 00 00 00 07 01 03 04 00 78 00 1a
[modbus] TRACE 2022/05/22 18:47:07 modbus: send 00 0f 00 00 00 06 01 03 9c df 00 02
[modbus] TRACE 2022/05/22 18:47:07 modbus: recv 00 0f 00 00 00 07 01 03 04 00 79 00 1e
[modbus] TRACE 2022/05/22 18:47:07 modbus: send 00 10 00 00 00 06 01 03 9c ff 00 02
[modbus] TRACE 2022/05/22 18:47:08 modbus: recv 00 10 00 00 00 07 01 03 04 00 7a 00 2c
[modbus] TRACE 2022/05/22 18:47:08 modbus: send 00 11 00 00 00 06 01 03 9d 2d 00 02
[modbus] TRACE 2022/05/22 18:47:08 modbus: recv 00 11 00 00 00 07 01 03 04 00 7b 00 18
[modbus] TRACE 2022/05/22 18:47:08 modbus: send 00 12 00 00 00 06 01 03 9d 47 00 02
[modbus] TRACE 2022/05/22 18:47:08 modbus: recv 00 12 00 00 00 07 01 03 04 00 a0 00 30
[modbus] TRACE 2022/05/22 18:47:08 modbus: send 00 13 00 00 00 06 01 03 9d 79 00 02
[modbus] TRACE 2022/05/22 18:47:08 modbus: recv 00 13 00 00 00 07 01 03 04 ff ff 00 00
[modbus] TRACE 2022/05/22 18:47:08 modbus: send 00 14 00 00 00 06 01 03 9c 44 00 42
[modbus] TRACE 2022/05/22 18:47:09 modbus: recv 00 14 00 00 00 87 01 03 84 46 72 6f 6e 69 75 73 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 79 6d 6f 20 38 2e 32 2d 33 2d 4d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 2e 32 31 2e 34 2d 31 00 00 00 00 00 00 00 00 30 2e 33 2e 32 37 2e 32 00 00 00 00 00 00 00 00 33 31 33 32 39 35 30 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 71
[modbus] TRACE 2022/05/22 18:47:09 modbus: send 00 15 00 00 00 06 01 03 9d 49 00 08
[modbus] TRACE 2022/05/22 18:47:09 modbus: recv 00 15 00 00 00 13 01 03 10 ff fe ff fe ff ff 00 00 00 00 00 00 00 02 ff ff
[modbus] TRACE 2022/05/22 18:47:09 modbus: send 00 16 00 00 00 06 01 03 9d 51 00 14
[modbus] TRACE 2022/05/22 18:47:09 modbus: recv 00 16 00 00 00 2b 01 03 28 00 01 53 74 72 69 6e 67 20 31 00 00 00 00 00 00 00 00 00 87 c4 18 1a 79 00 02 a3 00 2a 1d 33 a3 80 00 00 04 ff ff ff ff
[modbus] TRACE 2022/05/22 18:47:09 160:1:DCW: 677.7
[modbus] TRACE 2022/05/22 18:47:09 modbus: send 00 17 00 00 00 06 01 03 9d 49 00 08
[modbus] TRACE 2022/05/22 18:47:09 modbus: recv 00 17 00 00 00 13 01 03 10 ff fe ff fe ff ff 00 00 00 00 00 00 00 02 ff ff
[modbus] TRACE 2022/05/22 18:47:09 modbus: send 00 18 00 00 00 06 01 03 9d 65 00 14
[modbus] TRACE 2022/05/22 18:47:10 modbus: recv 00 18 00 00 00 2b 01 03 28 00 02 53 74 72 69 6e 67 20 32 00 00 00 00 00 00 00 00 00 49 4c 2c 05 8f 00 00 ad 2f 2a 1d 33 a4 80 00 00 04 ff ff ff ff
[modbus] TRACE 2022/05/22 18:47:10 160:2:DCW: 142.3
Power: 820W

```

kalaws, 2022-05-22:
Exactly the same issue for me (only with grid, pv is fine). Fronius Symo 12.5-M

andig, 2022-05-22:
What is the difference between Symo and Symo Gen24? Are the Gen24 newer hybrids? Wondering how to differentiate the two types?

Macrisu, 2022-05-22:
The current EU versions are:

Symo is On-grid system 3 phase
Primo is On-grid system 1 phase
Gen24 Hybrid (on-grid + 3 kW emergency backup socket)

https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42%2C0410%2C2012.pdf

6.2 Inverter Device Type List

andig, 2022-05-23:
From what I understand:

- both configs are more or less identical
- @Macrisu uses slave id 240 instead of 200 for grid (why- does that depend on the older meter?). We could probably make the grid id configurable
- @Macrisu uses `power` as pv reading instead of the sum over the DC currents. He can because it's not a hybrid inverter.

That said we could technically split the current setup into 3 different templates:
- Fronius Hybrid/Gen24 for PV, Battery
- Fronius (no hybrid) for PV
- Fronius Smartmeter with configurable slave id for Grid

Since we should leave the current template as-is I'd probably slice them differently:

- Fronius Hybrid/Gen24 for PV, Battery, Grid - this is the current template
- Fronius (no hybrid) for PV
- Fronius Smartmeter (legacy) with configurable slave id for Grid- this would only be used for non-hybrid installations as those always have a meter attached.

/cc @premultiply

Macrisu, 2022-05-23:
I have two smartmeters Fronius:
- old direct measurements 63a-3
- New indirect measurements ts 5kA-3

both have the same id, but a configurable id is a good idea.

andig, 2022-05-23:
Made a suggestion. Not specifically Fronius, but rather generic inverter/ hybrid inverter. Maybe @thierolm could give it a try. Removed `sunspec` template as well.

thierolm, 2022-05-23:
`pv` meter is working, `grid` meter not. But I have an external S0 () meter connected to the external connector like shown below.

For this meter I'm not able to find any in the attached Fronius register Excel sheets. 

[Smart_Meter_Register_Map_Float.xlsx](https://github.com/evcc-io/evcc/files/8758428/Smart_Meter_Register_Map_Float.xlsx)
[Gen24_Primo_Symo_Inverter_Register_Map_Float.xlsx](https://github.com/evcc-io/evcc/files/8758429/Gen24_Primo_Symo_Inverter_Register_Map_Float.xlsx)

[image]
[image]

Macrisu, 2022-05-24:
S0 is a dumb impulse meter.
It adds X impulses for each kwh.

andig, 2022-05-24:
In the end, it may not be exposed at all. I wouldn't know. We could still let @Macrisu try by merging into nightly which I'll do now.

Macrisu, 2022-05-24:
So I must install Nightly (easy in docker), but what template should I use for config?

andig, 2022-05-24:
See pr or use configure. Should be generic inverter.

Macrisu, 2022-05-24:
Where can I find the nightly releases (amd_x64)? 
I know how to doit in Docker, but I prefer executable for configuration :(

Macrisu, 2022-05-24:
I use Synology, so no apt.
I've installed it on docker - nightly 565582fb, but I think is old.

From what I see [here](https://github.com/evcc-io/evcc/commit/c201029d1141b277532bc942bca3e561788d28a1)
[templates/definition/meter/sunspec-inverter.yaml](https://github.com/evcc-io/evcc/commit/c201029d1141b277532bc942bca3e561788d28a1#diff-27fbca1b75ee49f3d6f59cbafd48aa13de3ae6853405f653886a2faded089f67)

I've made this config

```
- name: grid
  type: template
  template: sunspec-inverter
  usage: grid
  modbus: tcpip
  integrer: float
  advanced: true
  valuetype: bool
  uri: 192.168.1.190
  port: 502
```

Error

`[main  ] FATAL 2022/05/24 10:59:05 cannot create meter 'grid': cannot create meter 'template': template not found: sunspec-inverter`

thierolm, 2022-05-24:
By using `evcc configure` U have to select the "generic inverter". 

Check my log:

```bash
Auf geht`s:

 In welchem Modus soll die Konfiguration durchgeführt werden? Fortgeschrittener Modus (Detailliertere Fragen, erfordert jedoch technisches Know-How)

 Was möchtest du machen? Ein einzelnes Gerät konfigurieren (muss manuell in eine Konfigurationsdatei eingetragen werden!)

- Ein Gerät konfigurieren

Wähle eines der folgenden Gerätekategorien aus

 Wähle eines der folgenden Gerätekategorien aus Netz-Stromzähler

- Konfiguration Netz-Stromzähler

 Wähle einen Netz-Stromzähler: Wechselrichter [Generische Unterstützung]

Führe folgende Einstellungen durch:

 Modbus ID 1
 IP-Adresse oder Hostname 192.168.178.18
 Port 502
-------------------------------------------------
Hilfe:
Einstellung Float/Integer im Wechselrichter überprüfen
-------------------------------------------------
 Integer Registerzugriff (nur für Netzzähler) Nein

Teste die grid Konfiguration von Wechselrichter [Generische Unterstützung] ...
   Fehler: sunspec: 213:0:W not found
```

Macrisu, 2022-05-24:
evcc version 0.93 (565582fb) apt unstale installed on rPi zero 2w

```
+--------------+----------+-------------+----------------------------------------------------------------------------------------------------------+
|      IP      | HOSTNAME |    TASK     |                                                 DETAILS                                                  |
+--------------+----------+-------------+----------------------------------------------------------------------------------------------------------+
| 192.168.1.65 |          | fronius-web | {"IP":"192.168.1.65","Port":80}                                                                          |
+              +----------+-------------+----------------------------------------------------------------------------------------------------------+
|              |          | meter       | {"IP":"192.168.1.65","Port":502,"ModbusResult":{"SlaveID":240,"Model":213,"Point":"W","Value":359.22}}   |
+              +----------+-------------+----------------------------------------------------------------------------------------------------------+
|              |          | strings     | {"IP":"192.168.1.65","Port":502,"ModbusResult":{"SlaveID":1,"Model":160,"Point":"N","Value":2}}          |
+              +----------+-------------+----------------------------------------------------------------------------------------------------------+
|              |          | sunspec     | {"IP":"192.168.1.65","Port":502,"ModbusResult":{"SlaveID":240,"Model":1,"Point":"Mn","Value":"Fronius"}} |
+              +----------+             +----------------------------------------------------------------------------------------------------------+
|              |          |             | {"IP":"192.168.1.65","Port":502,"ModbusResult":{"SlaveID":1,"Model":1,"Point":"Mn","Value":"Fronius"}}   |
+              +----------+-------------+----------------------------------------------------------------------------------------------------------+
|              |          | meter       | {"IP":"192.168.1.65","Port":502,"ModbusResult":{"SlaveID":240,"Model":213,"Point":"W","Value":358.29}}   |
+              +----------+-------------+----------------------------------------------------------------------------------------------------------+
|              |          | strings     | {"IP":"192.168.1.65","Port":502,"ModbusResult":{"SlaveID":1,"Model":160,"Point":"N","Value":2}}          |
+--------------+----------+-------------+----------------------------------------------------------------------------------------------------------+
```

```
evcc configure
[main  ] INFO 2022/05/24 21:15:46 evcc 0.93 (565582fb)

The next steps will guide you through the creation of a configuration file for evcc.
Please note that this process cannot cover all possible scenarios.
You can cancel the process by pressing CTRL-C anytime.

NOTE: This functionality is in an experimental state!
  This means that the created configuration file may not work
  in a future update of evcc and would need to be recreated.
  We are looking forward to your feedback at https://github.com/evcc-io/evcc/discussions/

Let's go:

 In which mode should this process guide you through the configuration? Advanced mode (Ask more details, requires technical know-how)

 What do you want to do? Configure a single device (has to be added manually to a configuration file!)

- Setup a device

Choose one of the following device categories

 Choose one of the following device categories grid meter

- Configuration grid meter

 Choose a grid meter: Inverter (SunSpec) [Generic support]

Please provide the following settings:

 Modbus ID 240
 IP-Address or Hostname 192.168.1.65
 Port 502

Testing the grid configuration of Inverter (SunSpec) [Generic support] ...
   Error: cannot create meter 'modbus': sunspec: subdevice 1 not found

 Testing of the grid configuration of Inverter (SunSpec) [Generic support] failed. Do you want to add it anyway? (y/N) 

```

andig, 2022-05-27:
Ich hab den Überblick verloren. Die neuen "generischen" Templates sind ok?

Macrisu, 2022-05-27:
If i use the grid template, is not working
It’s ok If I use the pv template

andig, 2022-05-27:
Looking at your scan, grid seems to be:

    {"IP":"192.168.1.65","Port":502,"ModbusResult":{"SlaveID":240,"Model":213,"Point":"W","Value":359.22}}  

Correct? That should work! What is the grid config you're using?

Macrisu, 2022-05-27:
Evcc configure

Inverter (SunSpec) [Generic support] Inverter (SunSpec) [Generic support] was successfully added.

The configuration:

  type: template
  template: sunspec 
  id: 240  
  host: 192.168.1.65  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1

andig, 2022-05-28:
@Macrisu it's quite confusing when you keep changing versions

> template: sunspec

That template doesn't exist anymore in the nightly. Could you as suggested please use nightly version and try

    template: sunspec-inverter

which is the "Inverter" that can also deliver grid power?

Macrisu, 2022-05-28:
I've updated to the latest evcc unstable: evcc version 0.93 (22ecf153)

I have a Fronius Symo 8.2-3-m (ongrid only, no battery)

**Using this is working!**

```
  type: template
  template: sunspec-inverter
  id: 240
  host: 192.168.1.65
  port: 502
  usage: grid
  modbus: tcpip
  name: grid
```

```
root@pi0:~# evcc meter --log trace
[main  ] INFO 2022/05/28 14:45:42 evcc 0.93 (22ecf153)
[main  ] INFO 2022/05/28 14:45:42 using config file /etc/evcc.yaml
[modbus] TRACE 2022/05/28 14:45:42 modbus: send 00 01 00 00 00 06 f0 03 9c 40 00 02
[modbus] TRACE 2022/05/28 14:45:43 modbus: recv 00 01 00 00 00 07 f0 03 04 53 75 6e 53
[modbus] TRACE 2022/05/28 14:45:43 modbus: send 00 02 00 00 00 06 f0 03 9c 42 00 02
[modbus] TRACE 2022/05/28 14:45:43 modbus: recv 00 02 00 00 00 07 f0 03 04 00 01 00 41
[modbus] TRACE 2022/05/28 14:45:43 modbus: send 00 03 00 00 00 06 f0 03 9c 85 00 02
[modbus] TRACE 2022/05/28 14:45:43 modbus: recv 00 03 00 00 00 07 f0 03 04 00 d5 00 7c
[modbus] TRACE 2022/05/28 14:45:43 modbus: send 00 04 00 00 00 06 f0 03 9d 03 00 02
[modbus] TRACE 2022/05/28 14:45:44 modbus: recv 00 04 00 00 00 07 f0 03 04 ff ff 00 00
[modbus] TRACE 2022/05/28 14:45:44 modbus: send 00 05 00 00 00 06 f0 03 9c 44 00 42
[modbus] TRACE 2022/05/28 14:45:44 modbus: recv 00 05 00 00 00 87 f0 03 84 46 72 6f 6e 69 75 73 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 6d 61 72 74 20 4d 65 74 65 72 20 36 33 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 2e 32 31 2e 34 2d 31 00 00 00 00 00 00 00 00 32 2e 39 00 00 00 00 00 00 00 00 00 00 00 00 00 31 37 30 32 31 32 38 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f0 00 d5
[modbus] TRACE 2022/05/28 14:45:44 modbus: send 00 06 00 00 00 06 f0 03 9c 87 00 7c
[modbus] TRACE 2022/05/28 14:45:44 modbus: recv 00 06 00 00 00 fb f0 03 f8 41 50 2d 0e 40 86 56 04 40 8e 4d d3 40 8b b6 46 43 72 88 89 43 72 b3 33 43 72 66 66 43 72 80 00 43 d2 08 89 43 d2 0c cd 43 d1 f3 33 43 d2 19 9a 42 47 99 9a c5 44 6a 14 c4 7d b8 52 c4 86 3c 7b c4 83 bb 85 45 44 a0 00 44 7e b6 b2 44 86 be 7d 44 84 58 29 c3 27 61 48 c1 f0 51 ec c2 61 8f 5c c2 a1 e6 66 bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 48 c4 bb 40 7f c0 00 00 7f c0 00 00 7f c0 00 00 46 cb bc 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 00 00 00 00
[modbus] TRACE 2022/05/28 14:45:44 modbus: send 00 07 00 00 00 06 f0 03 9c 87 00 7c
[modbus] TRACE 2022/05/28 14:45:45 modbus: recv 00 07 00 00 00 fb f0 03 f8 41 58 66 66 40 8b ce d9 40 93 e7 6d 40 91 16 87 43 72 b3 33 43 72 cc cd 43 72 99 9a 43 72 b3 33 43 d2 33 33 43 d2 33 33 43 d2 26 66 43 d2 40 00 42 47 99 9a c5 4c 61 9a c4 84 1f 0a c4 8b b6 66 c4 88 ed c3 45 4c a0 00 44 84 99 62 44 8c 29 83 44 89 8c dc c3 28 d1 ec c1 fa f5 c3 c2 60 a3 d7 c2 a2 94 7b bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 48 c4 bb 60 7f c0 00 00 7f c0 00 00 7f c0 00 00 46 cb bc 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 00 00 00 00
[modbus] TRACE 2022/05/28 14:45:45 modbus: send 00 08 00 00 00 06 f0 03 9c 87 00 7c
[modbus] TRACE 2022/05/28 14:45:45 modbus: recv 00 08 00 00 00 fb f0 03 f8 41 58 66 66 40 8b ce d9 40 93 e7 6d 40 91 16 87 43 72 b3 33 43 72 cc cd 43 72 99 9a 43 72 b3 33 43 d2 33 33 43 d2 33 33 43 d2 26 66 43 d2 40 00 42 47 99 9a c5 4c 61 9a c4 84 1f 0a c4 8b b6 66 c4 88 ed c3 45 4c a0 00 44 84 99 62 44 8c 29 83 44 89 8c dc c3 28 d1 ec c1 fa f5 c3 c2 60 a3 d7 c2 a2 94 7b bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 48 c4 bb 60 7f c0 00 00 7f c0 00 00 7f c0 00 00 46 cb bc 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 00 00 00 00
[modbus] TRACE 2022/05/28 14:45:45 modbus: send 00 09 00 00 00 06 f0 03 9c 87 00 7c
[modbus] TRACE 2022/05/28 14:45:45 modbus: recv 00 09 00 00 00 fb f0 03 f8 41 58 66 66 40 8b ce d9 40 93 e7 6d 40 91 16 87 43 72 b3 33 43 72 cc cd 43 72 99 9a 43 72 b3 33 43 d2 33 33 43 d2 33 33 43 d2 26 66 43 d2 40 00 42 47 99 9a c5 4c 61 9a c4 84 1f 0a c4 8b b6 66 c4 88 ed c3 45 4c a0 00 44 84 99 62 44 8c 29 83 44 89 8c dc c3 28 d1 ec c1 fa f5 c3 c2 60 a3 d7 c2 a2 94 7b bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 bf 7d 70 a4 48 c4 bb 60 7f c0 00 00 7f c0 00 00 7f c0 00 00 46 cb bc 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 7f c0 00 00 00 00 00 00
Power:          -3143W
Current L1..L3: 4.37A 4.62A 4.53A
```

I'm exporting into the grid 3.144 kW

andig, 2022-05-28:
Nice. Then I'll close this one as resolved.

closed 2022-05-28 by andig