evcc-io/evcc #3470: vehicleOdometer not updated anymore

e-frog, 2022-05-22:
### Describe the bug

Since evcc 0.92 vehicleOdometer is not updated anymore. It works when restarting evcc but doesn't update when connecting the car and start charging. This seems to be related to the changes in #3309.

### Steps to reproduce

see above

### Configuration details

vehicles:
- type: template
  template: vw
  title: e-Golf
  user: <username>
  password: <pw>
  vin: WVWZxxxxxxxxxxx
  capacity: 31.5
  phases: 2
  mode: pv
  minSoC: 0
  targetSoC: 100
  minCurrent: 6
  maxCurrent: 16
  identifiers:
  - 
  name: ev3

### Log details

nothing in the logs as DEBUG code was removed in #3309 :-)

### What type of operating system are you running?

Linux

### Version

Tested in 0.92 and 0.93

andig, 2022-05-22:
@e-frog it should still be visible from the trace log that the odometer API is being called when the vehicle is identified. Could we please check that?

e-frog, 2022-05-22:
at startup

```
[lp-1  ] INFO 2022/05/22 21:44:34 vehicle updated: unknown -> e-Golf
[cache ] TRACE 2022/05/22 21:44:34 lp-1/hasVehicle: true
[cache ] TRACE 2022/05/22 21:44:34 lp-1/mode: pv
[cache ] TRACE 2022/05/22 21:44:34 lp-1/targetSoC: 100
[cache ] TRACE 2022/05/22 21:44:34 lp-1/minSoC: 0
[cache ] TRACE 2022/05/22 21:44:34 lp-1/vehiclePresent: true
[influx] TRACE 2022/05/22 21:44:34 write targetSoC=100 (map[loadpoint:Garage vehicle:])
[influx] TRACE 2022/05/22 21:44:34 write minSoC=0 (map[loadpoint:Garage vehicle:])
[cache ] TRACE 2022/05/22 21:44:34 lp-1/vehicleTitle: e-Golf
[influx] TRACE 2022/05/22 21:44:34 write vehicleCapacity=31 (map[loadpoint:Garage vehicle:e-Golf])
[vw    ] TRACE 2022/05/22 21:44:34 GET https://mal-1a.prd.ece.vwg-connect.com/api/bs/vsr/v1/vehicles/***/status
[cache ] TRACE 2022/05/22 21:44:34 lp-1/vehicleCapacity: 31
[vw    ] TRACE 2022/05/22 21:44:34 {"StoredVehicleDataResponse":{"vin":"***","vehicleData":{"data": [{"id":"0x0101010001","field": [{"id":"0x0101010001","tsCarSentUtc":"2022-05-22T16:14:50Z","tsCarSent":"2000-01-01T00:00:00","tsCarCaptured":"2000-01-01T00:00:00","tsTssReceivedUtc":"2022-05-22T16:14:50Z","milCarCaptured":0,"milCarSent":0,"value":"echo"}]},{"id":"0x0101010002","field": [{"id":"0x0101010002","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T14:30:10","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16990,"milCarSent":16990,"value":"16990"}]},{"id":"0x0203FFFFFF","field": [{"id":"0x0203010001","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":null},{"id":"0x0203010002","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":null},{"id":"0x0203010003","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":"-13100","unit":"km","textId":"interval.inspection.distance"},{"id":"0x0203010004","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":"-49","textId":"interval.inspection.time"}]},{"id":"0x0204040001","field": [{"id":"0x0204040001","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":"0.0","unit":"l"}]},{"id":"0x0204040003","field": [{"id":"0x0204040003","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":"0.0","unit":"%","textId":"oil.level"}]},{"id":"0x0204040004","field": [{"id":"0x0204040004","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":"0","textId":"oilDisplay.0"}]},{"id":"0x02040C0001","field": [{"id":"0x02040C0001","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T00:32:09","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16947,"milCarSent":16990,"value":null,"textId":"not_supported"}]},{"id":"0x030101FFFF","field": [{"id":"0x0301010001","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T14:30:10","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16990,"milCarSent":16990,"value":"2","textId":"status_parking_light_off"}]},{"id":"0x030102FFFF","field": [{"id":"0x0301020001","tsCarSentUtc":"2022-05-22T16:14:50Z","tsCarSent":"2000-01-01T00:00:00","tsCarCaptured":"2000-01-01T00:00:00","tsTssReceivedUtc":"2022-05-22T16:14:50Z","milCarCaptured":0,"milCarSent":0,"value":"2985","unit":"dK"}]},{"id":"0x030103FFFF","field": [{"id":"0x0301030001","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T14:30:10","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16990,"milCarSent":16990,"value":"1","textId":"parking_brake_active"},{"id":"0x0301030002","tsCarSentUtc":"2022-05-22T16:14:41Z","tsCarSent":"2000-01-01T00:00:00","tsCarCaptured":"2000-01-01T00:00:00","tsTssReceivedUtc":"2022-05-22T16:14:53Z","milCarCaptured":0,"milCarSent":0,"value":"90","unit":"%","textId":"soc_ok"},{"id":"0x0301030003","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T14:30:10","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16990,"milCarSent":16990,"textId":"bem_error"},{"id":"0x0301030004","tsCarSentUtc":"2022-05-22T14:30:20Z","tsCarSent":"2022-05-22T14:30:10","tsCarCaptured":"2022-05-22T14:30:10","tsTssReceivedUtc":"2022-05-22T14:30:25Z","milCarCaptured":16990,"milCarSent":16990,"value":"0","unit":"km/h","textId":"speed_ok"},{"id":"0x0301030005","tsCarSentUtc":"2022-05-22T16:14:41Z","tsCarSent":"2000-01-01T00:00:00","tsCarCaptured":"2000-01-01T00:00:00","tsTssReceivedUtc":"2022-05-22T16:14:53Z","milCarCaptured":0,"milCarSent":0,"value":"248","unit":"k [... cut ...]
[go-e  ] TRACE 2022/05/22 21:46:05 {"alw":true,"eto":131555,"car":1,"wh":0,"trx":null,"nrg":[233,233,231,0,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"User 1","energy":0,"cardId":true},{"name":"User 2","energy":0,"cardId":false},{"name":"User 3","energy":0,"cardId":false},{"name":"User 4","energy":0,"cardId":false},{"name":"User 5","energy":0,"cardId":false},{"name":"User 6","energy":0,"cardId":false},{"name":"User 7","energy":0,"cardId":false},{"name":"User 8","energy":0,"cardId":false},{"name":"User 9","energy":0,"cardId":false},{"name":"User 10","energy":0,"cardId":false}]}
[lp-1  ] DEBUG 2022/05/22 21:46:05 charger status: A
[lp-1  ] INFO 2022/05/22 21:46:05 car disconnected
```

andig, 2022-05-23:
Ich sehe in Deinem Log keine 

    vehicle updated

Meldung. Wenns nicht erkannt wird dann wird auch kein km-Stand gelesen.

e-frog, 2022-05-23:
`vehicle updated` ist nur nach Neutstart von evcc in den logs.

`[lp-1  ] INFO 2022/05/22 21:44:34 vehicle updated: unknown -> e-Golf`

Dort wird das das Fahrzeug erkannt.
Sollte das nochmal neu erkannt werden beim Anstecken/Ladestart?

andig, 2022-05-23:
Sollte, man weiß ja sonst nicht welches nach Abstecken dran hängt.

e-frog, 2022-05-23:
`vehicle updated`  kommt beim Anstecken nur wenn mehrere Fahrzeuge konfiguriert sind. Bei nur einem Fahrzeug wird setActiveVehicle() sofort wieder verlassen und daher der km-Stand niemals aktualisiert.

```
// setActiveVehicle assigns currently active vehicle and configures soc estimator
func (lp *LoadPoint) setActiveVehicle(vehicle api.Vehicle) {
	if lp.vehicle == vehicle {
		return
	}
```

andig, 2022-05-23:
Klasse Analyse, danke!

andig, 2022-05-24:
Beim zweiten drauf schauen: das liegt an der unglücklichen Behandlung der Sondersituation "nur ein Auto konfiguriert". Ich glaube wir sollten das vorerst gar nicht weiter anpassen, sondern lieber zusammen mit dem Parkplatzfeature grundsätzlich aufräumen. Eine Diskussion zum Design wird es in Kürze geben.

/cc @naltatis

kaluxx, 2022-06-19:
Vielleicht könnt ihr mir mal auf die Sprünge helfen. Ich habe dasselbe Problem, dass es keinen Update auf den vehicleOdometer nach dem Start mehr gibt. Also Odo nur einmalig nach Start von evcc. Blöd, da ich damit meine gefahrenen KM darstelle.
Kann ich etwas in meiner Konfiguration ändern, damit das wieder upgedated wird? 
Wieso ist es denn ein Sonderfall, nur ein Auto konfiguriert zu haben??

andig, 2022-06-19:
> Wieso ist es denn ein Sonderfall, nur ein Auto konfiguriert zu haben??

Ist im Moment einfach so... Schau mal hier für Workaround: https://github.com/evcc-io/evcc/issues/3323#issuecomment-1152190224

kaluxx, 2022-06-28:
...habe dieses Dummy Fahrzeug angelegt und bekomme immer noch keine Odometer Daten (bzw. nur einmalig nach Neustart). Muss ich das Fzg. noch einem Loadpoint hinzufügen??
Ist auch nicht so schlimm. Falls das Problem in einem der nächsten Releases gelöst wird, würde ich da gar nicht weiter rumbasteln.

andig, 2022-07-13:
Please verify with tomorrow's nightly.

closed 2022-07-16 by andig