evcc-io/evcc #3471: missing loadpoints if only vehicle is configured

rr19-hub, 2022-05-22:
### Describe the bug

Wenn ich das evcc-Addon lediglich mit Daten für vehicle starte, erhalte ich folgende Fehlermeldung:
When I start the evcc addon with data for vehicle only, I get the following error message (see below).

Wichtig: mit der gleichen YAML klappt der Abruf der Daten über die Kommandozeile (Win) mit `evcc vehicle`!
(Important: with the same YAML the retrieval of the data via the command line (Win) with `evcc vehicle` works!)

### Steps to reproduce

- install evcc-addon in hass.io (supervised installation)
- create and modify yaml as described above
- start evcc-addon in hass.io

### Configuration details

~~~
# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
log: info
levels:
  core: debug
  lp-1: debug
  lp-2: debug

vehicles:
  - name: fiat500e
    type: template
    template: fiat
    title: Fiat 500e # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    # vin: ZFAE... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 44 # Akku-Kapazität in kWh # Optional
    pin: xxxx # Optional

# mqtt message broker
mqtt:
  broker: 192.168.178.80:1883
  topic: evcc/fiat # root topic for publishing, set empty to disable
  user: xxxxx
  password: xxxx
~~~

### Log details

Using config file:  /config/evcc.yaml
starting evcc: 'evcc --config  /config/evcc.yaml'
[main  ] INFO 2022/05/22 20:03:34 evcc 0.93
[main  ] INFO 2022/05/22 20:03:34 using config file /config/evcc.yaml
[main  ] INFO 2022/05/22 20:03:34 listening at :7070
[mqtt  ] INFO 2022/05/22 20:03:34 connecting evcc-677703514 at tcp://192.168.178.80:1883
[main  ] FATAL 2022/05/22 20:03:35 missing loadpoints

### What type of operating system are you running?

Linux

### Version

0.93 on:
Home Assistant Supervisor 2022.05.2
Kernel version 5.10.0-14-amd64
Agent version 1.2.2

andig, 2022-05-22:
evcc requires a loadpoint to run.

closed 2022-05-22 by andig

rr19-hub, 2022-05-22:
This is a pity, because evcc (in the terminal) can also retrieve the status of a vehicle and thus actions or notifications could be triggered in HA. It would be helpful if this was also possible in the addon. Now I have to install evcc separately and try to send the status to HA via MQTT.

premultiply, 2022-05-22:
The is not true. There is only the component test mode and its output on console. There is no notification or anything else.

rr19-hub, 2022-05-22:
Ah, okay, sorry. I didn't know this one is just for testing. Then I'll see how I can use the values via the console. My attempt to create a generic loadpoint as a dummy is unfortunately not successful so far.