evcc-io/evcc #3474: Installation error

ano302, 2022-05-22:
Hi, I had to reinstalled my raspbian because of a buggy SD card. I had evcc 0.88 already instlalled and that worked without problems.
Now I got a problem during installation on a Raspberry Pi Zero 2 W

```

sudo apt-get install -y nodejs
sudo apt-get install -y golang
git clone https://github.com/evcc-io/evcc.git
```

Next step was a `make` where I got the following error. 
```

go: downloading cloud.google.com/go/compute v1.6.1
main.go:4:2: package embed is not in GOROOT (/usr/lib/go-1.15/src/embed)
tools.go:6:2: import "github.com/alvaroloes/enumer" is a program, not an importable package
tools.go:7:2: import "github.com/golang/mock/mockgen" is a program, not an importable package
../../go/pkg/mod/github.com/evcc-io/eebus@v0.0.0-20220403153356-c4f2bab5546f/cert/cert.go:16:2: package io/fs is not in GOROOT (/usr/lib/go-1.15/src/io/fs)
make: *** [Makefile:67: build] Error 1

```
Any idea what is going on?

thierolm, 2022-05-22:
Why are U not simply using the pi tarball installation?
See https://docs.evcc.io/docs/installation/manual#erstinstallation

By doing this, there is no need to clone the repo, install go, node etc ... no make necessary ...

andig, 2022-05-23:
Your go version is outdated.

closed 2022-05-23 by andig

PeterPablo, 2023-01-20:
On a raspbian bullseye `sudo apt install golang nodejs` yields installation of these packages
```
Get:1 http://deb.debian.org/debian bullseye/main arm64 golang-1.15-src arm64 1.15.15-1~deb11u4 [13.9 MB]
Get:2 http://security.debian.org/debian-security bullseye-security/main arm64 libnode72 arm64 12.22.12~dfsg-1~deb11u1 [7,693 kB]
Get:3 http://security.debian.org/debian-security bullseye-security/main arm64 nodejs arm64 12.22.12~dfsg-1~deb11u1 [147 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main arm64 nodejs-doc all 12.22.12~dfsg-1~deb11u1 [2,547 kB]
Get:5 http://deb.debian.org/debian bullseye/main arm64 golang-1.15-go arm64 1.15.15-1~deb11u4 [43.0 MB]
Get:6 http://deb.debian.org/debian bullseye/main arm64 golang-1.15-doc all 1.15.15-1~deb11u4 [2,455 kB]
Get:7 http://deb.debian.org/debian bullseye/main arm64 golang-1.15 all 1.15.15-1~deb11u4 [27.9 kB]
Get:8 http://deb.debian.org/debian bullseye/main arm64 golang-src arm64 2:1.15~1 [4,856 B]
Get:9 http://deb.debian.org/debian bullseye/main arm64 golang-go arm64 2:1.15~1 [23.9 kB]
Get:10 http://deb.debian.org/debian bullseye/main arm64 golang-doc all 2:1.15~1 [4,888 B]
Get:11 http://deb.debian.org/debian bullseye/main arm64 golang arm64 2:1.15~1 [4,844 B]
Get:12 http://deb.debian.org/debian bullseye/main arm64 libc-ares2 arm64 1.17.1-1+deb11u1 [99.9 kB]
Get:13 http://deb.debian.org/debian bullseye/main arm64 libjs-highlight.js all 9.18.5+dfsg1-1 [397 kB]
```
and invoking `make`
```
pi@raspberrypi:~/bin/evcc $ make
Version: 0.111.1 f38353bf 2023-01-20_07:41:26
CGO_ENABLED=0 go build -v -tags=release -ldflags='-X github.com/evcc-io/evcc/server.Version=0.111.1 -X github.com/evcc-io/evcc/server.Commit=f38353bf -s -w'
go: downloading github.com/golang/mock v1.6.0
go: downloading github.com/dmarkham/enumer v1.5.7
go: downloading github.com/Masterminds/sprig/v3 v3.2.3
go: downloading github.com/joho/godotenv v1.4.0
go: downloading github.com/eclipse/paho.mqtt.golang v1.4.2
go: downloading github.com/skratchdot/open-golang v0.0.0-20200116055534-eef842397966
go: downloading github.com/andig/gosunspec v0.0.0-20211108155140-af2e73b86e71
go: downloading github.com/olekukonko/tablewriter v0.0.5
go: downloading github.com/prometheus/client_golang v1.14.0
go: downloading github.com/libp2p/zeroconf/v2 v2.2.0
go: downloading gitlab.com/bboehmke/sunny v0.15.1-0.20211022160056-2fba1c86ade6
go: downloading golang.org/x/exp v0.0.0-20230116083435-1de6713980de
go: downloading github.com/jinzhu/copier v0.3.5
go: downloading github.com/dylanmei/iso8601 v0.1.0
go: downloading github.com/grid-x/modbus v0.0.0-20230105075906-8b31e0d1647e
go: downloading golang.org/x/crypto v0.5.0
go: downloading golang.org/x/tools v0.5.0
go: downloading github.com/nicksnyder/go-i18n/v2 v2.2.1
go: downloading github.com/nirasan/go-oauth-pkce-code-verifier v0.0.0-20220510032225-4f9f17eaec4c
go: downloading github.com/avast/retry-go/v3 v3.1.1
go: downloading github.com/jpfielding/go-http-digest v0.0.0-20211006141426-fbc93758452e
go: downloading github.com/imdario/mergo v0.3.13
go: downloading github.com/cenkalti/backoff v2.2.1+incompatible
go: downloading github.com/bogosj/tesla v1.1.0
go: downloading github.com/volkszaehler/mbmd v0.0.0-20221223133217-870b3b92b81e
go: downloading github.com/spf13/cobra v1.6.1
go: downloading github.com/go-telegram-bot-api/telegram-bot-api/v5 v5.5.1
go: downloading github.com/pascaldekloe/name v1.0.1
go: downloading golang.org/x/net v0.5.0
go: downloading github.com/google/go-github v17.0.0+incompatible
go: downloading github.com/joeshaw/carwings v0.0.0-20220711204617-d90801cf2848
go: downloading github.com/miekg/dns v1.1.50
go: downloading github.com/google/uuid v1.3.0
go: downloading google.golang.org/protobuf v1.28.1
go: downloading golang.org/x/text v0.6.0
go: downloading gopkg.in/yaml.v3 v3.0.1
go: downloading github.com/pkg/browser v0.0.0-20210911075715-681adbf594b8
go: downloading github.com/grid-x/serial v0.0.0-20211107191517-583c7356b3aa
go: downloading github.com/korylprince/ipnetgen v1.0.1
go: downloading github.com/google/go-github/v32 v32.1.0
go: downloading golang.org/x/sync v0.1.0
go: downloading github.com/enbility/cemd v0.1.5
go: downloading github.com/prometheus/common v0.39.0
go: downloading github.com/BurntSushi/toml v1.2.1
go: downloading github.com/fatih/structs v1.1.0
go: downloading github.com/Masterminds/goutils v1.1.1
go: downloading github.com/manifoldco/promptui v0.9.0
go: downloading github.com/spf13/cast v1.5.0
go: downloading github.com/tv42/httpunix v0.0.0-20191220191345-2ba4b9c3382c
[... 121 lines cut ...]
go: downloading github.com/mergermarket/go-pkcs7 v0.0.0-20170926155232-153b18ea13c9
go: downloading github.com/remyoudompheng/bigfft v0.0.0-20220927061507-ef77025ab5aa
go: downloading modernc.org/memory v1.5.0
go: downloading github.com/jmespath/go-jmespath v0.4.0
go: downloading github.com/jeremywohl/flatten v1.0.1
go: downloading gopkg.in/ini.v1 v1.67.0
go: downloading go.opencensus.io v0.24.0
go: downloading github.com/googleapis/gax-go/v2 v2.7.0
go: downloading github.com/googleapis/enterprise-certificate-proxy v0.2.1
go: downloading cloud.google.com/go v0.107.0
go: downloading cloud.google.com/go/compute v1.15.1
go: downloading github.com/golang/groupcache v0.0.0-20210331224755-41bb18bfe9da
go: downloading cloud.google.com/go/compute/metadata v0.2.3
main.go:4:2: package embed is not in GOROOT (/usr/lib/go-1.15/src/embed)
[... cut ...]
../../go/pkg/mod/github.com/mattn/go-runewidth@v0.0.14/runewidth.go:7:2: found packages uniseg (doc.go) and main (gen_breaktest.go) in /home/pi/go/pkg/mod/github.com/rivo/uniseg@v0.4.3
provider/javascript/registry.go:11:2: import "github.com/robertkrimen/otto/underscore" is a program, not an importable package
server/assets/assets.go:3:8: package io/fs is not in GOROOT (/usr/lib/go-1.15/src/io/fs)
make: *** [Makefile:72: build] Error 1
```

Anyhow, I'm now trying to find Go 1.18 and Nodejs 18 for Raspbian to replace the apt versions golang 1.15 and nodejs 12.22...

@ano302, how did you obtain a more recent version of Go?