evcc-io/evcc #3479: Make bufferSoC accessable by API

premultiply, 2022-05-22:
### Discussed in https://github.com/evcc-io/evcc/discussions/3458

<div type='discussions-op-text'>

<sup>Originally posted by **DrGalvany** May 22, 2022</sup>
Hallo zusammen,
ich nutze "IoBroker" mit dem Adapter "mqtt".
In der Auflistung ist der Punkt "prioritySoC" vorhanden, den man auch schön im Broker "lesen/schreiben" kann.
Ist es möglich, bei der nächsten Version, evtl. auch den Punkt "bufferSoC" in den "mqtt" kommunizieren zu lassen?

Viele Grüße
Micha</div>

ThomDietrich, 2022-05-24:
Kann dies bestätigen und würde die Ergänzung ebenso begrüßen (auch wenn von mir konkret nicht benötigt).

Für mich allerdings sehr interessant wäre `residualPower`. Im Bezug auf die Diskussion in https://github.com/evcc-io/evcc/discussions/3255 wäre es wünscheswert diese via API/mqtt setzen zu können.

andig, 2022-05-24:
Ich tendiere dazu, keine weiteren Einzel-APIs mehr zu bauen, sondern ein `settings` API einzurichten dass über einen JSON POST zu bedienen wäre. Analog auch JSON über MQTT.

ThomDietrich, 2022-05-24:
Fyi ich habe die Integration via MQTT in Home Assistant dokumentiert ([hier](https://community.home-assistant.io/t/evcc-integration-via-mqtt-sensor-entity-config/407959))
JSON ist hier prinzipiell kein Problem, würde allerdings unnötig kompliziert wenn jeder POST das vollständige Settings Objekt enthalten muss. Ich denke nicht dass du davon sprichst aber wollte das mal erwähnt haben.

VolkerK62, 2022-05-26:
Wäre es in dem Zusammenhang möglich, für den einfachen Nutzer (IT-Dummie) ein settings Menü im GUI einzurichten?  Die Einstellungen auswählen, OK klicken und der POST wird gesendet.

premultiply, 2022-05-26:
PR welcome 😉

VolkerK62, 2022-05-27:
@premultiply  sorry, als Unwissender in Sachen Programmierung, kann ich leider nur mit der Idee dienen.

andig, 2022-06-26:
See https://github.com/evcc-io/evcc/issues/3715

closed 2022-06-26 by andig