evcc-io/evcc #3482: Allow disabling phase switching

andig, 2022-05-23:
Mit dem Update 0.92 ist sind mir ein paar Meldungen im UI zur Phasensteuerung aufgefallen. Im Bereich des Fahrzeugs stand gestern bei mir, dass auf die Phasenumschaltung von 1P auf 3P gewechselt wird. Meine Easee Home ist auf eine Phase fixiert und auch in der evcc.yaml ist eine Phase festhinterlegt. 
Dies nur mal als möglichen Hinweis auf eventuelle Kommunikationsfehler zwischen Easee Home und evcc. 
Laden funktioniert bei mir super, obwohl auch ab und an Fehlermeldungen kommen.

_Originally posted by @torstenwiese in https://github.com/evcc-io/evcc/discussions/2795#discussioncomment-2803702_

urmelaus, 2022-05-27:
Habe das gleiche Problem bei der Easee. Wäre nicht schlecht, wenn man das einstellen könnten. Habe aber gerade eine Email von Easee bekommen:
Easee Cloud API update with backwards compatibility break
Scheduled for Jul 1, 10:00 - 11:00 CEST
Scheduled
We will remove one of our login endpoints during this maintenance. If you have any systems relying on 'https://api.easee.cloud/api/accounts/token' please move this to 'https://api.easee.cloud/api/accounts/login' before this maintenance.

The two endpoints should be identical from an external pov. so please swap to the new URL and verify the implementation before 1st of July.
Posted 23 minutes ago. May 27, 2022 - 11:01 CEST

Chris8er, 2022-05-27:
Ich hätte das Feature auch gerne in der UI. Bei bewölkten Wetter ist es effizienter nur einphasig zu laden. Am besten auch per api, dann könnte ich das mit externer Wetter Logik verknüpfen.

andig, 2022-05-27:
> Ich hätte das Feature auch gerne in der UI. Bei bewölkten Wetter ist es effizienter nur einphasig zu laden. Am besten auch per api, dann könnte ich das mit externer Wetter Logik verknüpfen.

Was hat das eine mit dem anderen zu tun? Wenn nur für 1p Leistung da ist wird auch nur mit 1p geladen. Das entsprechende Api gibts ebenfalls schon.

Chris8er, 2022-05-27:
Ja ich habe den Bug hier falsch verstanden. Ich mach mal ein Feat-Request. Kann hier weg.

oliversoerensen, 2022-06-08:
Die Lösung in der Easee-App die Phasen auf 1 zu begrenzen ist keine Lösung. 
Ich möchte im PV und min+PV Modus bei meiner kleinen Anlage immer nur 1-Phasig laden, hier stört die Phasenumschaltung sehr da sie 1. Zeit kostet und 2. nicht zuverlässig funktioniert mit easee. 
Bei "sofort" brauche ich eben manchmal auch die 3 Phasen und da will dann die vollen 11kW haben.

andig, 2022-06-08:
> 1. Zeit kostet und 2. nicht zuverlässig funktioniert mit easee.

Die Zeit zu 1. holst Du durch schnelleres Laden auf 3p wieder rein. Das Problem zu 2. musst Du bei Easee adressieren, insbesondere da 

> Bei "sofort" brauche ich eben manchmal auch die 3 Phasen

Du hier ja doch wieder erwartest, dass es funktioniert ;)

oliversoerensen, 2022-06-08:
Dann steigen mir die Mitarbeiter aufs Dach, die eben bis Mittags ihren Akku vollhaben müssen weil sie zum Kunden müssen. Klar könnte ich die wieder an die öffentliche Ladesäule schicken, was aber die Akzeptanz des EV im Betrieb nicht fördert.

Lösung wäre phases pro Modus, ich würde es dann so einstellen:

pv_phases:  1
minpv_phases: 1
sofort_phases: 3

gosmarty, 2022-06-25:
> > Bin vielleicht ein Sonderfall mit meiner Anlage, aber so ist es eben für mich nicht zufriedenstellend.
> 
> Fair point. Ich tendiere dazu, künftig auch bei 1p3p Anlagen die Initialkonfiguration zu respektieren, heisst: mit
> 
> ```
> phases: 0
> ```
> 
> gibts Automatik und mit
> 
> ```
> phases: 1/3
> ```
> 
> würde fest eingestellt. Das würde aber genau Dein Problem nicht lösen.
> 
> > Die Lösung in der Easee-App die Phasen auf 1 zu begrenzen ist keine Lösung.
> 
> Ist weiter der beste Weg für Dich.

Also, mir würde das sehr gut helfen und gefallen. Eine meiner Boxen ist nämlich nur 1-phasig angeschlossen und beim Anschließen des Autos startet immer erst der 3p Timer.

andig, 2022-06-26:
Siehe https://github.com/evcc-io/evcc/pull/3714 - kann gerne schon getestet werden

closed 2022-06-27 by andig