evcc-io/evcc #3485: Nil pointer dereference

Chris8er, 2022-05-23:
### Describe the bug

Evcc startet häufig neu (ca. alle 20-30min).

Im Debug log findet sich ein null pointer error.

### Steps to reproduce

Evcc starten und warten.

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 5s # control cycle interval

sponsortoken: xxx

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: xxx
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: xxx
  password: xxx
  name: pv1

chargers:
- type: template
  template: easee
  user: xxx
  password: xxx
  charger: xxx
  name: wallbox2
- type: template
  template: easee
  user: xxx
  password: xxx
  charger: xxx
  name: wallbox3

loadpoints:
- title: Stellplatz
  charger: wallbox3
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
- title: Garage
  charger: wallbox2
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  meters:
    grid: grid1

hems:
  type: sma
  AllowControl: false
  deviceid: 125615551587

### Log details

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xef36c4]

goroutine 419 [running]:
github.com/koron/go-ssdp.(*Advertiser).Alive(0x0)
        /home/runner/go/pkg/mod/github.com/koron/go-ssdp@v0.0.3/advertise.go:156 +0x40
github.com/evcc-io/evcc/hems/semp.(*SEMP).Run(0x3b74040)
        /home/runner/work/evcc/evcc/hems/semp/semp.go:146 +0x320
created by github.com/evcc-io/evcc/cmd.run
        /home/runner/work/evcc/evcc/cmd/root.go:227 +0x10f8

### What type of operating system are you running?

Linux

### Version

0.92

andig, 2022-05-23:
Es ist nicht ganz klar, wo der Fehler liegt. Eine mögliche Ursache in unserem Code habe ich behoben- bitte im Nightly nochmal testen. Wenns wieder auftritt bitte melden.

Chris8er, 2022-05-24:
Nein in den Logs sehe ich nichts weiter. Dann warte ich mal bis gemerged wurde und teste.

```
[site  ] DEBUG 2022/05/23 19:47:27 ----
[lp-1  ] DEBUG 2022/05/23 19:47:27 charge power: 0W
[lp-2  ] DEBUG 2022/05/23 19:47:27 charge power: 0W
[site  ] DEBUG 2022/05/23 19:47:27 grid power: 20W
[site  ] DEBUG 2022/05/23 19:47:27 grid currents: [1.11 0.377 0.374]A
[site  ] DEBUG 2022/05/23 19:47:27 site power: 20W
[lp-2  ] DEBUG 2022/05/23 19:47:27 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2022/05/23 19:47:27 charger status: A
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xef9870]

goroutine 459 [running]:
github.com/koron/go-ssdp.(*Advertiser).Alive(0x0)
        /home/runner/go/pkg/mod/github.com/koron/go-ssdp@v0.0.3/advertise.go:156 +0x40
github.com/evcc-io/evcc/hems/semp.(*SEMP).Run(0x3a65000)
        /home/runner/work/evcc/evcc/hems/semp/semp.go:146 +0x320
created by github.com/evcc-io/evcc/cmd.run

```

andig, 2022-05-24:
Entschuldige, das sollte eigentlich gestern noch mit rein. Könntest Du das morgen im Nightly nochmal testen? Wenn der Fehler wieder/immer noch auftritt bitte nochmal ins Log schauen und wieder melden.

closed 2022-05-24 by andig

Chris8er, 2022-05-26:
Lief jetzt 2 Tage ohne Absturz. 👍🏻

andig, 2022-05-26:
Top. Und auch im Logfile keine Fehlermeldung? Ich hätte erwartet, dass da irgendein Announcement fehlschlägt das später zu dem Fehler führte, jetzt aber nur noch geloggt wird?