evcc-io/evcc #3504: New UI does not automatically refresh after longer period inactivity

hyperbart, 2022-05-25:
### Describe the bug

I have EVCC as a "web app" on the homescreen of iOS (https://www.makeuseof.com/tag/add-website-shortcuts-iphone-home-screen/).

However, after a longer period of time (+12h) when you open the EVCC app it does not update the values anymore and the buttons for the charging points (stop/now/min/pv) also don't work.

If I force close the "app" and restart it the values are updating again

### Steps to reproduce

Install EVCC
Add EVCC local website to your homescreen on iOS (https://www.makeuseof.com/tag/add-website-shortcuts-iphone-home-screen/)
Launch EVCC "app"
Wait +12h
Launch EVCC "app"

### Configuration details

```
interval: 1s

site:
- title: Allspark
  meters:
    grid: HuisMeter
    pvs:
    - pvHuis
    - pvTuinHuis
  residualpower: 100

messaging:
  events:
    start:
      title: Laadbeurt gestart
      msg: Er is een laadbeurt gestart op xxxx
    stop:
      title: Laadbeurt gestopt
      msg: |
        Oplader {{.title}} heeft een laadbeurt voltooid
        Er is {{round (divf .chargedEnergy 1000) 2 }} kWh geleverd op {{.chargeDuration}}.
    connect:
      title: Wagen aangesloten op xxx
      msg: Er is een auto aangesloten op de lader in xxxx
    disconnect:
      title: Wagen ontkoppeld op xxxx
      msg: De wagen is ontkoppeld van de lader in xxxxx
  services:
    - type: telegram
      token: xxxxxxx:xxxxxxxxxxxxxxxxxxxx
      chats:
        - -xxxxxxx

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.28 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.12 # [currency]/kWh

meters:
  - name: HuisMeter
    type: modbus
    model: sdm # specific non-sunspec meter
    id: 1
    energy: Sum # only required for charge meter usage
    #via TCP:
    uri: 192.168.4.184:8899
    rtu: false # serial modbus rtu (rs485) device connected using simple ethernet adapter
  - name: pvHuis
    type: sma
    uri: 192.168.4.230
    password: xxxxxx
  - name: pvTuinHuis
    type: sma
    uri: 192.168.4.231
    password: xxxxxx

sponsortoken: xxxxxx

log: debug
levels:
    core: debug

influx:
  url: http://192.168.y.x:xxxx
  database: EVCC
  token: "xxxxxxxx=="
  org: "xxxxx"

chargers:
- name: ChargingPointAlfen
  type: alfen
  uri: 192.168.4.178 # TCP ModBus address
```

### Log details

/

### What type of operating system are you running?

Linux

naltatis, 2022-05-25:
Thanks for the issue. I've noticed similar effects from time to time - even with the previous design. I'll try to reliably recreate this issue to come up with a solid fix. If you're able to reproduce it (without waiting > 12h) please let me know 😁.

hyperbart, 2022-05-25:
@naltatis with the previous GUI I didn't notice it as to be a thing with EVCC, because I went out of the house, lost connection etc, but even then I didn't experience it (or maybe once that I still know of), but now with the new GUI it's every morning or evening.

RTTTC, 2022-05-25:
I thought it only depends on browser cashing ?
I simply refresh the page, no need to force close it

andig, 2022-05-25:
If anything it would probably be the reconnect of the websocket connection. Haven't found any way to replicate it though.

hyperbart, 2022-05-26:
> I thought it only depends on browser cashing ?
> I simply refresh the page, no need to force close it

You are not reproducing the issue: if you create a shortcut on home screen and you open it the browser controls are not shown: You can’t refresh

naltatis, 2022-05-27:
👆 I've introduces a dedicated offline indicator. It will help debug this issue but is also a general improvement since evcc usually only works in your lokal network.

naltatis, 2022-05-27:
The PR is merged and the code will be part of todays nightly build. Would be great if you could install and test it. Maybe the changes already solve some issues.

hyperbart, 2022-05-28:
@naltatis , sure. I am running EVCC in a Docker environment. Haven't used/built the nightly builds before. How do I instruct my Docker to take the latest nightly?

andig, 2022-05-28:
:nightly

hyperbart, 2022-05-29:
@naltatis : installed the nightly, verified if the "cloudy icon and no connection to server" popped up by disabling wifi, works as expected.

Couple of hours later: EVCC "webpage app on homescreen" hangs (no updates) but the notification does not appear. So I have the issue but the no connection notification is not popping up.

naltatis, 2022-05-30:
Ok, so I guess we don't have a networking/reconnect issue. It seems likely that the web app hags due to some other problem.

@hyperbart Do you have a Mac? If yes it would help if you could USB-connect you iPhone to you Mac in this "hanging"-situation to extract browser logs via macOS Safari.

Here is an good description on how to do that:
**Method 1**
https://www.browserstack.com/guide/how-to-debug-on-iphone

I hope there will be some useful information in the Konsole and Network tab that can help me understand the problem.

RTTTC, 2022-05-30:
Another data point that might be interesting:

Due to my main inverter problems I get no connection to it, so grid meter and one pv meter is not working,  constant flow of errors in GUI.
EVCC also now shows "no connection to server" , yet I know there is a good connection to server (raspberry) since I can see data from my other inverter "pv meter" changing.

hyperbart, 2022-05-31:
> 

No, sorry, no Mac available here.

andig, 2022-06-19:
Mir ist das heute seit langer Zeit mal wieder passiert, gleich 2x. Ich schau mal, ob ich an die Logs komme...

andig, 2022-08-25:
Ich mach hier mal zu- gerne wieder auf wenn wir neue Erkenntnisse haben.

closed 2022-08-25 by andig

naltatis, 2022-09-04:
Hallo Alle, bei mir läuft die Nightly seit dem MR nun problemfrei durch. Ich hatte keine steckengebliebenen Werte mehr und musste auch seitdem die Homescreen-App nicht wieder abschießen weil sie nicht mehr reagierte oder nur weiß angezeigt hat. Habt ihr ähnliche Erfahrungen gemacht?

hyperbart, 2022-09-05:
@naltatis : same experiences here, your MR seems to have fixed the issue, good job.