evcc-io/evcc #3526: Wrong unit of odometer for BMW

toh1000, 2022-05-28:
### Describe the bug

If configured unit (in BMW vehicle) is set to kilometers, the printed odometer value is wrong, because it is given in miles but pretends to be in kms. 
Please note: unit of milage is variable and can be accessed via API.

### Steps to reproduce

1. Configure BMW vehicle (in my case 330e). Assure to set "km" as unit in your car.
2. run `evcc vehicle`

Output is (in my case)
`....
Odometer:      26490km
`
correct value is 42595km which equals 26490 **miles**. 

### Configuration details

vehicles:
\- type: template
  template: bmw
  title: 330e
  user: xxx
  password: xxx
  vin: xxx
  capacity: 11
  name: BMW

### Log details

n.a.

### What type of operating system are you running?

Linux

### Version

0.94

andig, 2022-05-28:
Wouldn't that be something for BMW to fix? If not- what fix do you expect? Multiply everything my km/miles or some more involved logic? Which? note it's impossible to test/diagnose this without access to a physical car's odometer...

rikroe, 2022-05-28:
Does adding the following parameter to all HTTP requests help?
`bmw-units-preferences: d=KM;v=L`

I've seen it while debugging, but didn't think that much of it. And can't reproduce on my own vehicle as that doesn't provide any odometer data anymore...

andig, 2022-05-28:
Check! With https://github.com/evcc-io/evcc/commit/55e045dd936b8bbc30c3fe4f0ee63c1489f61942 I can see the odometer change. Electrical range doesn't though. I'd say it's fixed, thanks for the report and the suggestion!

closed 2022-05-28 by andig