evcc-io/evcc #3528: Support different min/max current per 1p/3p

andig, 2022-05-28:
Der Smart EQ kann 1-phasig mit 6A laden. Bei 3-phasig aber nur mit minimal 8A.
Kann ich das in evcc (mit Sponsor-Token) abbilden?

Falls nicht, müsste ich 8A in evcc als Minimum einstellen. Ich könnte mir vorstellen, dass das 1-phasig im Winter, oder an trüberen Tagen oft genau den Unterschied macht zwischen ausreichend PV-Überschuss, oder nicht.

_Originally posted by @reach0775 in https://github.com/evcc-io/evcc/discussions/3514_

naltatis, 2022-05-28:
Die separate Einstellbarkeit wäre für mein Setup auch interessant. Ich lade mit der easee 1p (6A-20A) und 3p (6A-16A). Da ich das nicht separat einstellen kann gibt er in 3p auch bis 20A frei. Das ist kein wirkliches Problem für mich, da das Auto nicht mehr als 11kW nimmt. In der Phasenvisualisierung sind die Balken bei 3p@16A aber nicht ganz ausgefüllt weil der Max-Wert ja 20A ist. Also eher ein kosmetisches Thema.

premultiply, 2022-05-28:
Dito.

Joegi05, 2022-07-10:
Hallo zusammen, ich bin ganz neu hier und noch recht unvertraut damit, wie das hier so läuft. :-)
Ich habe das gleiche Problem beim Laden meines Renault Twingos (Bj2022). Dieser lädt 1-phasig problemlos von 6A-16A, 3-phasig jedoch erst ab 8A. Das führt zu andauerndem Wechsel der Ladeströme und zu erhöhtem Netzbezug. Aktuell behelfe ich mir damit den minimalen Strom generell auf 8A zu erhöhen, da in der Mittagszeit öfters mal 3-phasig geladen wird. Ich befürchte aber, genau wie andig, dass mir dadurch gerade in der Übergangszeit einige kWh im unteren Bereich entgehen. Ideal wäre es natürlich, wenn man die minimalen Ladeströme für 1- und 3- phasiges Laden separat definieren könnte.

ThomDietrich, 2022-07-11:
Ich würde diese Verbesserung ebenso begrüßen. Aktuell setze ich ähnliche Wünsche erfolgreich aus Home Assistant heraus um.

Rainbow45-pixel, 2022-07-17:
Das wäre bestimmt für alle Zoé Fahrer interessant.

fountainedge, 2022-07-18:
Ich habe auch oft das Phänomen, dass der Smart (mit dem Zoe-22kW-Bordlader) einphasig bis zur Grenze von 3,6kW lädt, und sobald es nach Dreiphasig wechseln möchte, das Laden abbricht. Habe ich genügend Leistung von der PV (z.B. 6-7kW) und es startet dann direkt dreiphasig, lädt er ab 6,4kW problemloser und kann dann auch hochfahren. Aber zwischen 3,6 und 6,4kW fehlt irgendwie ein Range der dem Smart gefällt.

andig, 2022-09-08:
Die Idee ist gut, aber momentan leider erstmal ausserhalb Reichweite da zu viele Edge cases.

closed 2022-09-08 by andig

pvolkert, 2023-09-28:
Moin zusammen, wird an dem Thema noch "indirekt" gearbeitet? Sprich gab es inzwischen vllt eine Idee wie die verschiedenen Stromgrenzen zu realisieren wären? Habe bei meinem Twingo Electric (logischerweise) das selbe Problem. Komme mit 7A als globale Grenze meist durch, aber die Abbrüche sind schon etwas nervig da man ständig das System im Blick behalten muss.

MarkusGH, 2023-10-15:
Möchte das Thema auch nochmals in Erinnerung rufen. Habe hier ein Fahrzeug, das mit 3x32A oder 1x20A laden kann. wegen 1p3p Umschaltung muss ich aber maxCurrent auf 20A stellen, da ansonsten kein Umschaltung auf 3p erfolgt, wenn im PV Modus mehr als 4,6 kW und weniger als 7,4 kW zur Verfügung stehen.

andig, 2023-10-15:
Können wir ggf. angehen wenn https://github.com/evcc-io/evcc/issues/10330 umgesetzt ist

MarkusGH, 2023-10-15:
Wäre super! Aber macht es keinen Sinn, das gleich in #10330 zu integrieren, das ist ja ohnehin ein BC?

ChrisGitHub42, 2024-08-19:
Gibt es hier ein Update? In der Schweiz darf man aus regulatorischen Gründen im 1-Phasig nur mit max. 8A Laden - 3-Phasig kann ich Problemlos auf bis zu 32A (wenn das die Wallbox zu lassen würde), gehen...

So wie es jetzt ist, ist die Umschaltmöglichkeit von 1P auf 3P und zurück wie z.B. mit der GoE Box in der Schweiz in Kombination mit EVCC schlicht nicht legal zu betreiben... (ausser man begrenzt auch das 3-Phasige Laden auf max. 8A, was dann aber EVCC wiederum komplett überflüssig macht...).