evcc-io/evcc #3535: Provide a meaningful error message  for not connected renault cars

ickeundso, 2022-05-29:
### Describe the bug

For some reasons the the my renault account is not connected with the configured car in evcc.
Known reasons: 
- the user doesn't yet connected the car to the my renault account 
- renault removed the car from the account, way is can happen is not known but it was seen

This will produce a http 403 if evcc tries to get the vehicle status values. The vehicle will then not be available in evcc.

The kemeron API provides an additional attribute to determine the car is already connected to the active account.

For the /vehicles request the response contains the "Connected Driver" object.

`"connectedDriver": {
                "role": "MAIN_DRIVER",
                "createdDate": "2022-04-20T14:39:40.557672Z",
                "lastModifiedDate": "2022-05-29T12:36:47.520413Z"
            },`

For the case the car is not connected to the my renault account the response lokks like this:

`"connectedDriver": {
                "createdDate": "2022-04-20T14:39:40.557672Z",
                "lastModifiedDate": "2022-05-29T12:36:47.520413Z"
            },`

The implementation can ask for this attribute and can produce a meaningful error message.

### Steps to reproduce

Go to you my renault app an remove the car form the acount.
"trennen sie ihr auto" on the very bottom left.
You will then get the http 403 when evcc tries to ask the kemereon API.

### Configuration details

vehicles:
  - name: zoe
    type: renault
    title: Zoe # Wird in der Benutzeroberfläche angezeigt # Optional
    user: ****@posteo.de
    password: ****
    vin: VF...
    capacity: 41 # Akku-Kapazität in kWh # Optional 
    onIdentify: # set defaults when vehicle is identified
      minSoC: 65 # charge to at least 65% independent of charge mode
      targetSoC: 80 # limit charge to 80%

### Log details

ERROR 2022/05/29 15:19:17 vehicle odometer: unexpected status: 500
ERROR 2022/05/29 15:19:24 vehicle soc: unexpected status: 500

### What type of operating system are you running?

Docker container

### Version

master

ickeundso, 2022-05-29:
I will create a pull request for that issue. Almost done.

closed 2022-07-03 by andig

andig, 2022-08-20:
@ickeundso leider scheint das mit Dacia nicht zu funktionieren:

		"connectedDriver": {
			"createdDate": "2022-07-25T14:59:59.396452050Z",
			"lastModifiedDate": "2022-08-19T13:47:49.592337747Z"
		},

Hier gibts kein Role Attribut und trotzdem ist das Auto verbunden. Ich würde jetzt createdDate != 0 ebenfalls noch akzeptieren.

andig, 2022-08-20:
Siehe https://github.com/evcc-io/evcc/pull/4166