evcc-io/evcc #3543: 1Phase / 3Phase Charging Lock in UI

mhahn67, 2022-05-30:
Hey,

our PV System isn't so big. We have only 7kwh peak power. We need good conditions to get enough power for 3p charging. 
The current system is switching a lot from 1P to 3P and then back to 1P only when the Sun is shining without clouds it is working a long time with 3P. And that's not so cool when its switching 2-6 Times in hour from 1 to 3p.

Can you add the Feature "Phase Lock" that I can set up in the UI that EVCC should switch to 3p or stay at 1P (also when the power is there for 3P) 

Our System:
-PowerFox and Easee Home Charging Robot

Regards and thanks from Germany 
Marco

dirk1992, 2022-05-31:
We have exactly the same system. It is often the case that the vehicle no longer charges at all. We have an e-tron, an e-golf and a gte. With the GTE, which is 1-phase, it works without problems. With the e-golf and the e-tron, you often have to disconnect and reconnect the vehicles completely. I suspect it is due to the constant switching. One has just such a system so that the fluctuations during the day are intercepted.

Chris8er, 2022-06-02:
see: https://github.com/evcc-io/evcc/issues/3522

Just asked the same question. I was told to "disable 3P charging in the Easee App". But I would prefer using only EVCC to control everything.

Also when I set the limit in the easee app, the UI sometimes displays a "switching to 3P in..." announcment nevertheless.

If I charge our Model 3 I get some weired "phase wiring issue between charger and vehicle" error (can't remember the text exactly and I'm unable to reproduce right now).

andig, 2022-06-03:
Duplikat zu https://github.com/evcc-io/evcc/issues/3482

closed 2022-06-03 by andig

dirk1992, 2022-06-06:
Da man hier auch auf Deutsch schreiben kann hoffe ich ist das in Ordnung.
Ich finde nicht dass es ein Duplikat zu dem verlinkten Post ist.
Eine Phasenumstellung möchte ich nicht im UI machen, auch wenn das ein nützliches Feature ist.
Es geht darum, dass die aktuelle Umstellung nicht richtig funktioniert mit easee und es hier zu Abbrüchen führt und das autounabhängig.
Hier müsste man schauen woran das liegt. Ich hatte vorher clever-pv getestet und dort waren solche Probleme nicht. Daher kann ich ein Problem mit Fahrzeug oder Wallbox ausschließen.

andig, 2022-06-06:
> Es geht darum, dass die aktuelle Umstellung nicht richtig funktioniert mit easee und es hier zu Abbrüchen führt und das autounabhängig.

bitte neues Issue mit Logfile

tribun83, 2023-09-22:
> our PV System isn't so big. We have only 7kwh peak power. We need good conditions to get enough power for 3p charging. 

I would reduce the wallbox to 2P. In Wallbox app oder by phsically disconnecting 3rd Phase and setting EVCC to 2 phases in yaml. This limits the problem...

premultiply, 2023-09-22:
This is no valid configuration.
3p or 1p on charger!

tribun83, 2023-09-22:
> This is no valid configuration. 3p or 1p on charger!

Thanks for the addition. You mean in EVCC or in hardware? It works fine for me (also with Teslas and some cars only have 2P). I don't see a downside of this configuration (beside not beeing valid).