evcc-io/evcc #3567: Nissan Leaf (pre 2019) does not update SOC reliably

Kubiac, 2022-06-02:
### Describe the bug

Ich habe einen Nissan Leaf ZE1 40 kWh (Erstzulassung 2018). Der SOC kann grundsätzlich mit meiner evcc.yaml abgerufen werden, aber tut dies nur seeeeehr selten und ich vermute, dass es nur einmal geht, wenn ich mich zuvor gerade in der Nissan-App oder Website angemeldet habe.

Zigfache Fehlermedlung im evcc-Webinterface ist:
```vehicle soc: rpc error: code = DeadlineExceeded desc = context deadline exceeded```

### Steps to reproduce

keine Interaktion nötig

### Configuration details

Meine config in Auszügen:
```
vehicles:
- type: template
  template: carwings 
  title: Elli  
  user: ...@...  
  password: ************
  vin: S*************
  capacity: 35  
  phases: 1  
  mode: pv  
  minSoC: 20  
  minCurrent: 6  
  maxCurrent: 32  
  cloud: true  
  name: ev2
```

### Log details

reiche ich nach

### What type of operating system are you running?

Linux

### Version

0.94

andig, 2022-06-03:
> context deadline exceeded

Der Server antwortet einfach viel zu langsam. Wir könnten das konfigurierbar machen oder den Default hoch setzen?

andig, 2022-06-04:
@Kubiac funktioniert denn

    evcc vehicle

bei Dir? Gibts ein Logfile aus dem das Problem ersichtlich wäre?

Kubiac, 2022-06-07:
`evcc vehicle` geht.
Hier der Auszug aus dem log mit debug. Hier tritt der Fehler nach dem Verbinden des Fahrzeugs das erste mal auf:
```
Jun 07 19:00:40 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:40 ----
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charge power: 0W
Jun 07 19:00:40 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:40 pv power: 3040W
Jun 07 19:00:40 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:40 site power: -2040W
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charge currents: [0 0 0]A
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charger status: B
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charge start energy: 1.71e+03kWh
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] INFO 2022/06/07 19:00:40 car connected
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 vehicle api refresh
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 pv timer elapse
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 pv timer inactive
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] ERROR 2022/06/07 19:00:50 vehicle soc: rpc error: code = DeadlineExceeded desc = context deadline exceeded
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 pv charge current: 8.87A = 0A + 8.87A (-2040W @ 1p)
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 site power -2040W <= 0W enable threshold
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 pv enable in 0s
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 pv enable timer elapsed
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 max charge current: 6A
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 charger enable
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 wake-up timer: start
Jun 07 19:00:50 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:50 ----
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 charge power: 0W
Jun 07 19:00:50 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:50 pv power: 3026W
Jun 07 19:00:50 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:50 site power: -2026W
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 charge currents: [0 0 0]A
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 charger status: B
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 next soc poll remaining time: 59m49s
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 pv charge current: 8.81A = 0A + 8.81A (-2026W @ 1p)
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 pv timer reset
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 pv timer inactive
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 max charge current: 8A
Jun 07 19:01:00 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:01:00 ----
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:00 charge power: 1224W
Jun 07 19:01:00 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:01:00 pv power: 3025W
Jun 07 19:01:00 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:01:00 site power: -801W
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:00 charge currents: [7.5 0 0]A
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:00 charger status: C
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:00 charge start energy: 1.71e+03kWh
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] INFO 2022/06/07 19:01:00 start charging ->
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] ERROR 2022/06/07 19:01:00 vehicle soc: rpc error: code = DeadlineExceeded desc = context deadline exceeded
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:00 pv charge current: 11.5A = 8A + 3.48A (-801W @ 1p)
Jun 07 19:01:00 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:00 max charge current: 11A
Jun 07 19:00:40 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:40 ----
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charge power: 0W
Jun 07 19:00:40 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:40 pv power: 3040W
Jun 07 19:00:40 raspberrypi evcc[930]: [site  ] DEBUG 2022/06/07 19:00:40 site power: -2040W
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charge currents: [0 0 0]A
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charger status: B
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 charge start energy: 1.71e+03kWh
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] INFO 2022/06/07 19:00:40 car connected
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 vehicle api refresh
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 pv timer elapse
Jun 07 19:00:40 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:40 pv timer inactive
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] ERROR 2022/06/07 19:00:50 vehicle soc: rpc error: code = DeadlineExceeded desc = context deadline exceeded
Jun 07 19:00:50 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:00:50 pv charge current: 8.87A = 0A + 8.87A (-2040W @ 1p)
[... cut ...]
Jun 07 19:01:30 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:30 charger status: C
Jun 07 19:01:30 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:30 next soc poll remaining time: 59m50s
Jun 07 19:01:30 raspberrypi evcc[930]: [lp-1  ] ERROR 2022/06/07 19:01:30 vehicle soc: rpc error: code = DeadlineExceeded desc = context deadline exceeded
Jun 07 19:01:30 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:30 pv charge current: 8.85A = 8A + 0.848A (-195W @ 1p)
Jun 07 19:01:30 raspberrypi evcc[930]: [lp-1  ] DEBUG 2022/06/07 19:01:30 max charge current: 8A
```

andig, 2022-06-07:
Ähhhhh…. wieso eigentlich rpc error? Hast du cloud=1? Warum?

Kubiac, 2022-06-08:
Ja cloud war auf true gesetzt. Deaktiviert ist der Fehler auf jeden Fall weg und es scheint sofort deutlich besser zu funktionieren. Warum ich es an hatte? `evcc configure` hat es mir schmackhaft gemacht und Sponsor war ich auch schon vorher.

1) Gibt es hier jetzt noch was zu klären warum es über die cloud nicht ging? Ich helfe gerne.
2) Ich habe jetzt neue/weitere Fehlermeldungen, die mit der `vehicle`-config in Zusammenhang stehen aber keine Einschränkung der Hauptfunktionen bedeuten. Wollt ihr die dokumentiert haben? Wenn ja, in neuen Issues oder hier?

andig, 2022-06-08:
Wenns ein Fehler ist neues Issue, sonst gerne Discussion.