evcc-io/evcc #3574: SMA connection lost after a while

Matze4711, 2022-06-03:
### Describe the bug

After a while, evcc loses the connection to SMA inverters. 
Somtimes after several hours, sometimes after a few minutes.

Restarting of evcc helps. 

login failed: no packet received in timeout
vor 1 Sekunde
pv meter 0: update timeout: 1h43m56s
vor 1 Sekunde
login failed: no packet received in timeout
vor 5 Sekunden
pv meter 1: update timeout: 1h43m51s
vor 5 Sekunden

### Steps to reproduce

Not reproducable 

### Configuration details

evcc running on home assitant on raspi 4.

meters:

  - name: sma-HM2
    type: template
    template: sma-home-manager
    usage: grid
    host: 192.168.178.xx # IP-Adresse oder Hostname 
  
  - name: WR5Gebraucht
    type: template
    template: sma-inverter
    usage: pv
    host: 192.168.178.xx # IP-Adresse oder Hostname
    password: "Password" # Passwort für Benutzergruppe Benutzer # Optional 
    
  - name: WR6
    type: template
    template: sma-inverter
    usage: pv
    host: 192.168.178.xx # IP-Adresse oder Hostname
    password: "Password" # Passwort für Benutzergruppe Benutzer # Optional     

### Log details

[site  ] DEBUG 2022/06/03 14:41:18 site power: -4695W
[site  ] DEBUG 2022/06/03 14:41:21 ----
[sma   ] ERROR 2022/06/03 14:41:22 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:22 pv meter 0: update timeout: 1h45m26s
[sma   ] ERROR 2022/06/03 14:41:22 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:23 pv meter 1: update timeout: 1h45m26s
[site  ] DEBUG 2022/06/03 14:41:23 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:23 grid power: -4711W
[site  ] DEBUG 2022/06/03 14:41:23 grid currents: [5.79 6.49 7.65]A
[site  ] DEBUG 2022/06/03 14:41:23 site power: -4711W
[site  ] DEBUG 2022/06/03 14:41:26 ----
[sma   ] ERROR 2022/06/03 14:41:27 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:27 pv meter 0: update timeout: 1h45m31s
[sma   ] ERROR 2022/06/03 14:41:27 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:28 pv meter 1: update timeout: 1h45m31s
[site  ] DEBUG 2022/06/03 14:41:28 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:28 grid power: -4363W
[site  ] DEBUG 2022/06/03 14:41:28 grid currents: [5.33 6.05 7.15]A
[site  ] DEBUG 2022/06/03 14:41:28 site power: -4363W
[site  ] DEBUG 2022/06/03 14:41:31 ----
[sma   ] ERROR 2022/06/03 14:41:32 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:32 pv meter 0: update timeout: 1h45m36s
[sma   ] ERROR 2022/06/03 14:41:32 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:33 pv meter 1: update timeout: 1h45m36s
[site  ] DEBUG 2022/06/03 14:41:33 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:33 grid power: -4335W
[site  ] DEBUG 2022/06/03 14:41:33 grid currents: [5.5 6.21 6.75]A
[site  ] DEBUG 2022/06/03 14:41:33 site power: -4335W
[site  ] DEBUG 2022/06/03 14:41:36 ----
[sma   ] ERROR 2022/06/03 14:41:37 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:37 pv meter 0: update timeout: 1h45m41s
[sma   ] ERROR 2022/06/03 14:41:37 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:38 pv meter 1: update timeout: 1h45m41s
[site  ] DEBUG 2022/06/03 14:41:38 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:38 grid power: -4176W
[site  ] DEBUG 2022/06/03 14:41:38 grid currents: [5.34 5.99 6.55]A
[site  ] DEBUG 2022/06/03 14:41:38 site power: -4176W
[site  ] DEBUG 2022/06/03 14:41:41 ----
[sma   ] ERROR 2022/06/03 14:41:42 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:42 pv meter 0: update timeout: 1h45m46s
[sma   ] ERROR 2022/06/03 14:41:42 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:43 pv meter 1: update timeout: 1h45m46s
[site  ] DEBUG 2022/06/03 14:41:43 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:43 grid power: -4052W
[site  ] DEBUG 2022/06/03 14:41:43 grid currents: [5.15 5.87 6.37]A
[site  ] DEBUG 2022/06/03 14:41:43 site power: -4052W
[site  ] DEBUG 2022/06/03 14:41:46 ----
[sma   ] ERROR 2022/06/03 14:41:47 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:47 pv meter 0: update timeout: 1h45m51s
[sma   ] ERROR 2022/06/03 14:41:47 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:48 pv meter 1: update timeout: 1h45m51s
[site  ] DEBUG 2022/06/03 14:41:48 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:48 grid power: -4222W
[site  ] DEBUG 2022/06/03 14:41:48 grid currents: [5.34 6.08 6.58]A
[site  ] DEBUG 2022/06/03 14:41:48 site power: -4222W
[site  ] DEBUG 2022/06/03 14:41:51 ----
[sma   ] ERROR 2022/06/03 14:41:52 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:52 pv meter 0: update timeout: 1h45m56s
[sma   ] ERROR 2022/06/03 14:41:52 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:53 pv meter 1: update timeout: 1h45m56s
[site  ] DEBUG 2022/06/03 14:41:53 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:53 grid power: -4161W
[site  ] DEBUG 2022/06/03 14:41:53 grid currents: [5.28 5.98 6.5]A
[site  ] DEBUG 2022/06/03 14:41:53 site power: -4161W
[site  ] DEBUG 2022/06/03 14:41:56 ----
[sma   ] ERROR 2022/06/03 14:41:57 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:57 pv meter 0: update timeout: 1h46m1s
[sma   ] ERROR 2022/06/03 14:41:57 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:41:58 pv meter 1: update timeout: 1h46m1s
[site  ] DEBUG 2022/06/03 14:41:58 pv power: 0W
[site  ] DEBUG 2022/06/03 14:41:58 grid power: -4101W
[site  ] DEBUG 2022/06/03 14:41:58 grid currents: [5.19 5.92 6.42]A
[site  ] DEBUG 2022/06/03 14:41:58 site power: -4101W
[site  ] DEBUG 2022/06/03 14:42:01 ----
[sma   ] ERROR 2022/06/03 14:42:02 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:42:02 pv meter 0: update timeout: 1h46m6s
[sma   ] ERROR 2022/06/03 14:42:02 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:42:03 pv meter 1: update timeout: 1h46m6s
[site  ] DEBUG 2022/06/03 14:42:03 pv power: 0W
[site  ] DEBUG 2022/06/03 14:42:03 grid power: -4364W
[site  ] DEBUG 2022/06/03 14:42:03 grid currents: [5.54 6.26 6.78]A
[site  ] DEBUG 2022/06/03 14:42:03 site power: -4364W
[site  ] DEBUG 2022/06/03 14:42:06 ----
[sma   ] ERROR 2022/06/03 14:42:07 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:42:07 pv meter 0: update timeout: 1h46m11s
[sma   ] ERROR 2022/06/03 14:42:07 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:42:08 pv meter 1: update timeout: 1h46m11s
[site  ] DEBUG 2022/06/03 14:42:08 pv power: 0W
[site  ] DEBUG 2022/06/03 14:42:08 grid power: -4395W
[site  ] DEBUG 2022/06/03 14:42:08 grid currents: [5.57 6.27 6.83]A
[site  ] DEBUG 2022/06/03 14:42:08 site power: -4395W
[site  ] DEBUG 2022/06/03 14:42:11 ----
[sma   ] ERROR 2022/06/03 14:42:12 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:42:12 pv meter 0: update timeout: 1h46m16s
[sma   ] ERROR 2022/06/03 14:42:12 login failed: no packet received in timeout
[site  ] ERROR 2022/06/03 14:42:13 pv meter 1: update timeout: 1h46m16s
[site  ] DEBUG 2022/06/03 14:42:13 pv power: 0W
[site  ] DEBUG 2022/06/03 14:42:13 grid power: -4356W
[site  ] DEBUG 2022/06/03 14:42:13 grid currents: [5.51 6.23 6.76]A
[site  ] DEBUG 2022/06/03 14:42:13 site power: -4356W

### What type of operating system are you running?

Docker container

andig, 2022-06-04:
/cc @bboehmke hast Du eine Idee was da los sein könnte?

@Matze4711 kannst Du mal bitte den Tei vom Log zeigen wo das Problem anhängt (nicht erst 1:45h später)?

bboehmke, 2022-06-04:
So spontan habe ich keine Idee woran das liegt. 
Ich würde jetzt darauf tippen das es möglicherweise an dem Dockerzeugs von Home Assistant liegen könnte.
Aber ohne genaue Anhaltspunkte oder logs wird es etwas schwierig genaues zu sagen.

andig, 2022-06-05:
Klingt so als könnten wir nicht viel tun. @Matze4711 gerne wieder melden, wenn Du das auch ohne Docker und mit Logfile nachstellen kannst.

closed 2022-06-05 by andig

Matze4711, 2022-06-06:
Ich habe endlich den Moment in den Logs erwischt, wo es auftritt. 
Scheinbar hat es sich sogar von selbst erholt.
Kann man den Timeout anpassen?

Using config file:  /config/evcc.yaml
starting evcc: 'evcc --config  /config/evcc.yaml'
[main  ] ERROR 2022/06/05 16:18:03 `uri` is deprecated and will be ignored. Use `network` instead.
[sma   ] ERROR 2022/06/06 02:35:21 login failed: no packet received in timeout
[sma   ] ERROR 2022/06/06 02:35:26 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:35:26 pv meter 1: update timeout: 11s
[sma   ] ERROR 2022/06/06 02:35:31 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:35:31 pv meter 1: update timeout: 16s
[sma   ] ERROR 2022/06/06 02:35:36 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:35:36 pv meter 1: update timeout: 21s
[sma   ] ERROR 2022/06/06 02:35:41 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:35:41 pv meter 1: update timeout: 26s
[sma   ] ERROR 2022/06/06 02:35:46 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:35:46 pv meter 1: update timeout: 31s
[sma   ] ERROR 2022/06/06 02:35:51 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:35:51 pv meter 1: update timeout: 36s
[sma   ] ERROR 2022/06/06 02:35:56 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:35:56 pv meter 1: update timeout: 41s
[sma   ] ERROR 2022/06/06 02:36:01 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:01 pv meter 1: update timeout: 46s
[sma   ] ERROR 2022/06/06 02:36:06 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:06 pv meter 1: update timeout: 51s
[sma   ] ERROR 2022/06/06 02:36:11 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:11 pv meter 1: update timeout: 56s
[sma   ] ERROR 2022/06/06 02:36:16 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:16 pv meter 1: update timeout: 1m1s
[sma   ] ERROR 2022/06/06 02:36:21 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:21 pv meter 1: update timeout: 1m6s
[sma   ] ERROR 2022/06/06 02:36:26 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:26 pv meter 1: update timeout: 1m11s
[sma   ] ERROR 2022/06/06 02:36:31 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:31 pv meter 1: update timeout: 1m16s
[sma   ] ERROR 2022/06/06 02:36:36 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:36 pv meter 1: update timeout: 1m21s
[sma   ] ERROR 2022/06/06 02:36:41 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:41 pv meter 1: update timeout: 1m26s
[sma   ] ERROR 2022/06/06 02:36:46 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:46 pv meter 1: update timeout: 1m31s
[sma   ] ERROR 2022/06/06 02:36:51 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:51 pv meter 1: update timeout: 1m36s
[sma   ] ERROR 2022/06/06 02:36:56 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:36:56 pv meter 1: update timeout: 1m41s
[sma   ] ERROR 2022/06/06 02:37:01 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:37:01 pv meter 1: update timeout: 1m46s
[sma   ] ERROR 2022/06/06 02:37:06 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:37:06 pv meter 1: update timeout: 1m51s
[sma   ] ERROR 2022/06/06 02:37:11 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:37:11 pv meter 1: update timeout: 1m56s
[sma   ] ERROR 2022/06/06 02:37:16 login failed: no packet received in timeout
[site  ] ERROR 2022/06/06 02:37:16 pv meter 1: update timeout: 2m1s
[sma   ] ERROR 2022/06/06 09:37:51 login failed: no packet received in timeout

bboehmke, 2022-06-06:
Anpassen kann man das bisher nicht und eigentlich sollte es auch nicht nötig sein.
Hinzu kommt das der login 3 mal versucht wird und jedes mal wird 500 ms gewartet (was schon sehr lange ist).

Hast du vielleicht die Möglichkeit evcc mal auf einem anderem Geräte laufen zu lassen um zu schauen ob das reproduzierbar ist?
Eventuell wäre auch so der grobe Netzwerkaufbau (Router, Switche, ...) ganz interessant. Das SMA Speedwire ist hier teilweise etwas sensibel und hat mit einigen Geräten teilweise Probleme.

Matze4711, 2022-06-07:
Kann es daran liegen, dass ich parallel auch eine Modbus Verbindung und eine UDP Verbindung zu SMA Produkten laufen habe?

bboehmke, 2022-06-07:
Ausschließen würde ich es zumindest nicht.
Ich hab hier allerdings keinerlei Erfahrung (ich habe bisher nur via Speedwire mit den Geräten kommuniziert).

premultiply, 2022-06-07:
Würde ich auch nicht ausschließen halte ich aber für unwahrscheinlich. Habe ich auch laufen ohne dass dabei zum genannten Problem kommt.

Wohl aber hatte ich den Fall schonmal als ich einen Wechselrichter neugestartet habe. Da musste ich dann auch anschließend evcc neustarten da es keine Verbindung mehr aufbauen wollte errinnere ich mich jetzt. Es blieb dann in dem oben beschriebenen Zustand hängen.