evcc-io/evcc #3587: New GUI vertauscht Ladepunkt und Verbrauch - manchmal

Swantu, 2022-06-05:
Hallo allerseits, bin seit gestern auf 0.94 und schaue gerade zu, wie die Sonne mein Auto lädt. 
Dabei ist mir ein Fehler aufgefallen - manchmal vertauscht EVCC die Werte für hausverbrauch und für den Ladepunkt - zumindest im GUI und in den Daten die in Influx geschrieben werden:  
[image]
[siehe blau/lila Peaks: charger <1kw, Haus ~ bis zu 11kw]
Das ganze hängt wohl mit Timeouts vom charge-meter zusammen, zumindest korreliert es bei mir mit den Zeitmeldungen von "timeouts", und ist mit dem nächsten 10Sekundenintervall wieder vorbei. 
[image]
[2. Screenshot bei einer Wolke gemacht, da war die Ladeleistung nur 6.3kw anstatt 11]

Das dabei die Charge-Stärke nicht direkt angepasst wird schließe ich aus der fehlenden "Einspeisung", die sonst um bis zu +11kw hochgehen müsste. Bevor die Frage aufkommt: Ich habe keinen 2.Charger oder andere Verbraucher, der ~11kw in 10 Sekunden an oder ausschaltet. 

..bis auf den Darstellungsfehler/Verbindungsfehler, eine schöne Demo, wie EVCC Solarstrom in das Auto pumpt :) 
Ich freue mich weiterhin!

Swantu, 2022-06-05:
Hier noch der Screenshot mit den disconnects dazu. 
[image]
Wenn ich weiter darüber nachdenke, dann wird wohl einfach nur der "fehlende" Chargerwert auf "0" gesetzt, und der Rest ergibt sich dann automatisch. War das bei vorher [0.89 und früher] konservativer, oder ist meine Verbindung heute/gestern schlechter ?

premultiply, 2022-06-05:
Hast du vielleicht die neue und die alte Version noch irgendwie parallel laufen?
Sicherheitshalber und im Zweifelsfall mal das System auf dem evcc läuft neustarten.

andig, 2022-06-05:
> Wenn ich weiter darüber nachdenke, dann wird wohl einfach nur der "fehlende" Chargerwert auf "0" gesetzt, und der Rest ergibt sich dann automatisch. War das bei vorher [0.89 und früher] konservativer, oder ist meine Verbindung heute/gestern schlechter ?

Wüsste nicht was sich da geändert haben sollte. Was stellst Du Dir unter "konservativer" bei fehlenden Meßwerten vor?