evcc-io/evcc #3590: recovering from panic in logger: interface conversion: interface {} is []interface {}, not string

naltatis, 2022-06-05:
### Describe the bug

evcc outputs millions (literally) error messages to the log resulting in GBs of log file and higher than usual cpu and memory load.

this message appears over and over:

```
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
```

evcc itself still does its job correctly and from time to time there are expected log entries.
It seems that the number of errors between control cycles is growing (doubling).

```
[site  ] DEBUG 2022/06/05 17:13:24 grid power: -825W
[site  ] DEBUG 2022/06/05 17:13:24 site power: -825W
[lp-2  ] DEBUG 2022/06/05 17:13:24 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2022/06/05 17:13:24 charger status: A
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
:
[site  ] DEBUG 2022/06/05 17:13:24 grid power: -825W
[site  ] DEBUG 2022/06/05 17:13:24 site power: -825W
[lp-2  ] DEBUG 2022/06/05 17:13:24 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2022/06/05 17:13:24 charger status: A
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
recovering from panic in logger: interface conversion: interface {} is []interface {}, not string
:
[site  ] DEBUG 2022/06/05 17:13:24 grid power: -825W
[site  ] DEBUG 2022/06/05 17:13:24 site power: -825W
[lp-2  ] DEBUG 2022/06/05 17:13:24 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2022/06/05 17:13:24 charger status: A
```

### Steps to reproduce

This behaviour does not happen directly after start and I wasn't able to find what triggers it. I also don't have a log where the "normal" log transforms into this "messy" log.

### Configuration details

Templates: tesla, vzlogger, solaredge-inverter (tcpip), easee

### Log details

See above

### What type of operating system are you running?

Docker container

### Version

latest nightly

andig, 2022-06-06:
Mhhm. Ich hab die grobe Idee, dass es iwie mit Logging der Ströme (=slice) zusammen hängt. Lässt sich testen, ob das beim laden passiert/ wenn das UI verbunden ist? Oder bei Logging nach Influx? Ggf liegts auch an der Easee Implementierung- bei mir sehe ich das trotz Ladezähler nciht.

andig, 2022-06-06:
Ich könnte mir vorstellen, dass es am `SignalrLogger` liegt, ggf an

    if slices.Contains(skipped, v.(string))

Du könntest an der Stelle mal testen ob es wirklich ein String ist und- falls nicht- vorher abbrechen und v.a. ins Logfile schreiben was da ankommt um das zu verstehen bevor wir einen Fix umsetzen.

andig, 2022-06-06:
Yep, da muss es sein: https://github.com/philippseith/signalr/blob/7387d3c3ec4c49011ef2e7e4a55c1037892e564f/options.go#L116

aBua-MUC, 2022-07-11:
Ich schließe mich dem Fehler an. Bei mir ebenso im Docker, Easee Home Wallbox, an Kostal Plenticore Plus. Fahrzeug wird (im Moment) noch nicht daran betrieben, da Lieferverzögerung und ist auch noch nicht konfiguriert.

andig, 2022-07-11:
Da hilft nur logging was da schreckliches kommt. Kann ich mangels Easee leider nicht reproduzieren.

andig, 2022-07-13:
Bitte mal morgen das Nightly beobachten. Die panic wird jetzt eine deutlichere Meldung ausgeben.

andig, 2022-07-21:
Habt ihr irgendwas spannendes im Logfile gefunden?

andig, 2022-07-24:
Bitte melden falls es neue Erkenntnisse gibt, ich mache solange mal zu

andig, 2022-08-15:
https://github.com/evcc-io/evcc/issues/4116#issuecomment-1214940574 has some details:

    Aug 15 12:52:48 raspberrypi evcc[11091]: recovering from panic in logger: expected string, got []interface {} from [ts 2022-08-**15T11:52:48.920933998Z level info connect

Looks like a slice of string from Easee, not sure why we're receiving this. I think we could safely just ignore it.

andig, 2022-08-17:
Per https://github.com/evcc-io/evcc/commit/1d7e25f8664c0f796910a68f24e5242e1a2589af behoben. Falls `key` kein zugehöriges Value hatte verschieben sich dadurch die Logeinträge, das müssten wir dann nochmal ändern.

closed 2022-08-17 by andig