evcc-io/evcc #3594: vehicle not available: cannot create vehicle 'template': yaml: line 5(password?): found character that cannot start any token

Chris591, 2022-06-06:
### Describe the bug

Vehicle isn't working anymore

### Steps to reproduce

create a password with a starting '@...'
didn't changed anything on the config

### Configuration details

vehicles:
- type: template
  template: ford
  title: Mustang Mach-e
  user: e@mail
  password: '@.....'
  vin: **VIN**
  capacity: 88
  name: Mustang
  identifiers:
    - 0d03ef2b
  mode: "pv"
  #minSoC: 10
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 16
  phases: 3

### Log details

[main  ] INFO 2022/06/06 12:57:17 evcc 0.94 (ff1cd415)
[main  ] INFO 2022/06/06 12:57:17 using config file /usr/share/hassio/homeassistant/evcc/evcc.yaml
[mqtt  ] INFO 2022/06/06 12:57:17 connecting evcc-1532395470 at tcp://10.201.1.10:1883
.SoC:     vehicle not available: cannot create vehicle 'template': yaml: line 5: found character that cannot start any token
Capacity: 0kWh

### What type of operating system are you running?

Linux

### Version

evcc version 0.94 (ff1cd415)

andig, 2022-06-06:
@Chris591 ging das in früheren Versionen? Kann sein, dass YAML damit prinzipiell Probleme hat, siehe https://en.wikipedia.org/wiki/YAML. Am einfachsten: Passwort ändern!

Chris591, 2022-06-06:
Ja ging bis heute nach dem Update.
letzte Update hatte ich die Woche gemacht da lief es noch.

andig, 2022-06-06:
Fehler gefunden. Das Problem ist https://github.com/evcc-io/evcc/commit/335dafc890461ab9eb330f18277f87b5e2866d35 wenn der Wert schon in Quotes steht. Ich kann jetzt auf das double-quoting verzichten oder das einfach *nur* setzen wenn der Wert mit `0` anfängt. Wär wohl die bessere Lösung.

Chris591, 2022-06-06:
wer hat schon ein `@` am Anfang von einem Passwort ;-)
ohne Quotes schmeiß er ja schon beim einlesen einen Fehler.
das mit der `0` verstehe ich nicht aber muss ich ja auch nicht. :-)

warwick1902, 2022-06-22:
Kann mir jemand die Löung erklären? Ich habe das gleich Problem und verstehe nicht, was ich wie wo machen muss, um es zu lösen...

andig, 2022-06-22:
Aktuelles evcc nutzen und falls es noch auftritt bitte neues Thema @warwick1902