evcc-io/evcc #3611: Controled charge-stop without defined SOC-values

beocycris, 2022-06-09:
Good morning.

Ive got a MG ZS EV and at the moment no access to the SOC value via my local network.
Because the car automatically reduces the loading power from 16 amps to down to 1-2 amps while reaching a SOC are between 80 percent and 100 percent, I asked myself, if there could be any possibility to stop the charging session, after the car reduces the loading power?
This would lead to a reduction of the loading losses and could improve the battery lifetime.

In my case the go-echarger sends all values via MQTT to my local network. So the loadingpower is accessible via EVCC.

Does anyone implemented anything like this?

premultiply, 2022-06-09:
You can build your own control which implements this function by MQTT or REST API.

beocycris, 2022-06-09:
In EVCC itself?