evcc-io/evcc #3614: Keba und RFID

hungerburg, 2022-06-09:
An der Keba P30 Wallbox sind im evcc.yaml nur name, uri und typ konfiguriert. Ich seh diese Meldung mehrmals die Minute - https://github.com/evcc-io/evcc/blob/c984a19ed000910f91f0baa233ed5639f44882f0/charger/keba.go#L186

Wenn ich einen tag "123" konfiguriere, wird kein Fehler gemeldet. Ich finde nicht mehr, wo ich das gelesen hab, aber RFID ist optional. Die Meldung ist dann zu viel des Guten?

PS: Einen Tag konfigurieren mag ich nicht, denn dann ist die immer offen, auch ohne Karte, https://github.com/evcc-io/evcc/blob/master/templates/definition/charger/keba.yaml

andig, 2022-06-10:
Danke für den Hinweis. Die Abfrage dient ja dazu zu verstehen, ob evcc die WB überhaupt kontrollieren kann. In der Statusabfrage ist das nur bedingt sinnvoll. Bleibt die Frage wo wir die Warnung stattdessen hin packen wollen- in die Aufrufe zur Freigabe der Wallbox? M.E. sollte der Anwender hier weiter einen Hinweis bekommen wenn evcc einschalten will, dies aber gar nicht möglich ist.

Wär das ok?

hungerburg, 2022-06-11:
Wenn ich den path richtig lese, dann wird die Meldung nur mehr dann gedruckt, wenn der evc-Conntoller das Laden nicht erlauben konnte, nicht mehr dann, wenn der Status der Wallbox abgefragt wird, also um einiges seltener. Das macht die Meldung um Einiges nützlicher; In meinem Fall trotzdem nur Rauschen, aber eines mit dem sich besser auskommt.

andig, 2022-06-11:
Du kannst es morgen im nightly ausprobieren. Ich denke die Meldung brauchts weiter für andere Anwender.

closed 2022-06-11 by andig

hungerburg, 2022-06-13:
Die Meldung ist seit dem das nightly läuft nicht mehr ausgegeben worden. Von dieser Seite her gut. Vielen Dank!