evcc-io/evcc #3618: Password starting with '#'

chadibe0, 2022-06-10:
### Describe the bug

I had a password for the access to the car starting with a '#'.
I genereated the yaml-File by the installer script and the password was stored in the yaml file like this:
  password: '#RestOfThePW'

I removed the ' in front and after the password without success. Changing the password made it working finally.

### Steps to reproduce

set a pw beginning with #

### Configuration details

see above

### Log details

.SoC:     rpc error: code = Unknown desc = cannot create vehicle 'template': cannot create vehicle 'id': login.error.throttled

### What type of operating system are you running?

Linux

### Version

0.94

thierolm, 2022-06-10:
Are u using a template config?
Which vehicle template?
Pls provide your `evcc.yaml` vehicle config.

Leading special characters in evcc template string parameters are a known issue.

Solution to mittigate:
a) Create a **raw** configuration for your vehicle using evcc configure expand option: `./evcc configure --expand`
b) Copy and paste the created raw configuration into your `evcc.yaml` file 
c) Manually add leading and trailing single quotes around the affected parameter value (in your case the password)

chadibe0, 2022-06-10:
I used the template for VW ID.

 

The yaml-File looks like this:

 

 

vehicles:

- type: template

  template: id

  title: ID3

  user: myuseremail ***@***.***> 

  password: ‘#RestOfThePassword’

  vin: WV-ID

  capacity: 55

  phases: 3

  mode: pv

  minSoC: 20

  targetSoC: 80

  minCurrent: 1

  maxCurrent: 32

 

Regards

Bernhard

andig, 2022-06-10:
Probier mal bitte das Nightly- geht es damit?

andig, 2022-06-11:
Should be fixed in nightly.

closed 2022-06-11 by andig