evcc-io/evcc #3630: kostal-ksem-inverter: grid falsch

andig, 2022-06-12:
Grid ist völlig falsch. Das wird ja hier https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/kostal-ksem-inverter.yaml#L20 über den WR ausgelesen. Falsches Register oder bei Dir einfach nicht wie passende Konfiguration?

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/3619#discussioncomment-2932866_

andig, 2022-06-13:
From the docs for register 252: 

> Total active power (powermeter)
> Sensor position 1 (home consumption): (+) House consumption, (-) generation
> Sensor position 2 (grid connection): (+) Power supply, (-) feed-in

/cc @stonemaan bei Dir scheint der Sensor/KSEM einfach an anderer Stelle zu sitzen!

/cc @premultiply wie gehen wir damit um?

premultiply, 2022-06-13:
Sensorposition muss geändert werden und in der Software auf grid connection geändert werden.
Hier gibt es nichts zu tun.

stonemaan, 2022-06-13:
> Sensorposition muss geändert werden und in der Software auf grid connection geändert werden. Hier gibt es nichts zu tun.
> 
> Man könnte höchstens noch einen Hinweistext ins Template einfügen.

Meinst du Hardwareseitig? Das wird nicht gehen. Dann muss ich leider wieder auf mein selbst gebasteltes Script umsteigen. Da hat es funktioniert ^^

premultiply, 2022-06-13:
Ja genau.

stonemaan, 2022-06-13:
Ach ich verstehe. EVCC kann nur Sensor Position 1 verarbeiten. Und Sensor Position 2 nicht. Das muss man doch aber programmtechnisch abfangen können.. mit meinem kleinen Blockly Script in iobroker gehts ja auch. Nur ist evcc comfortabler programmiert. Schade

stonemaan, 2022-06-13:
Ok, ganz verstanden habe ich es nicht, weil ja mein KSEM bzw. mein Kostal WR alle Messwerte (vermeintlich?!) richtig anzeigt.. So funktioniert ja letztendlich auch mein Skript.. ob die Messwerte am KSEM zu ungenau sind bzw. sich von den Messwerten am Zähler unterscheiden kann ich nicht beurteilen.. Vielleicht installiere ich mal irgendwas einen Lesekopf am Zähler und komme auf evcc zurück ;-) Dann danke euch trotzdem für die Hilfe und die gute Arbeit am evcc-Projekt!

andig, 2022-06-14:
Deine App scheint den jeweils anderen Wert auszurechnen. Evcc braucht aber immer und zwingend einen Netzzähler. Den Wert gibt der WR in Deinem Einbauszenario nicht aus. Ich schau später nochmal rein, aber da hilft nur Deine Lösung oder Umbau.

andig, 2022-06-16:
Closed in https://github.com/evcc-io/evcc/commit/131d650ee7b3ea3dedbe0794dcdf3f09d5a882f2

closed 2022-06-16 by andig