evcc-io/evcc #3643: Phase current seems incorrect for Dutch DSMR5 / P1 when exporting

rtfmjoey, 2022-06-15:
### Describe the bug

I have a SolarEdge PV installation (3 phase) and a "Dutch Smart Meter" (P1/DSMR5) configured and currently using the hoermto:feature/lastmanagement1_grid version to loadbalance energy usage and noticed that the gridcurrents are noted positive when exporting energy:

"gridConfigured":true,
"gridCurrents":[7,7,6],
"gridPower":-5061,
"homePower":857,

At this moment I'm exporting on all three phases and would expect the gridcurrent to be negative on all phases.

The odd thing is that when the car isn't charging calculations of the currentbudget seem incorrect as the positive values seem to be used and cutting in to the available bidget. As soon as the car is plugged in and is charging the calculation seems correct (see https://github.com/evcc-io/evcc/pull/3453#issuecomment-1154968718 for more information).

### Steps to reproduce

- Configure PV and Dutch Smart Meter
- Review gridcurrents when exporting energy 

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s

meters:
- type: template
  template: dsmr
  host: 192.168.10.7
  port: 2000
  usage: grid
  name: grid1
- type: template
  template: solaredge-inverter
  id: 1
  host: 192.168.10.25
  port: 9000
  usage: pv
  modbus: tcpip
  timeout: 5s
  name: pv1

chargers:
- type: template
  template: evsewifi
  host: 192.168.40.31
  name: wallbox1

loadpoints:
- title: Oprit
  charger: wallbox1
  vehicles:
  - ev1
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Thuis
  meters:
    grid: grid1
    pvs: pv1
  maxCurrent: 25
  residualPower: 500

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.228

  feedin:
    type: fixed
    price: 0.228

mqtt:
  broker: 192.168.10.25:1883
  topic: evcc

vehicles:
- type: template
  template: tesla
  title: Tezla
  accessToken: xxx
  refreshToken: xxx
  vin: xxx
  capacity: 50
  name: ev1

### Log details

[site  ] DEBUG 2022/06/15 14:02:20 ----
[lp-1  ] DEBUG 2022/06/15 14:02:20 charge power: 0W
[site  ] DEBUG 2022/06/15 14:02:20 lm: distribute 18.0A over 0 lp in state B and 0 lp in state C
[site  ] DEBUG 2022/06/15 14:02:20 lm: first loop, remaining budget: 18.0A
[site  ] DEBUG 2022/06/15 14:02:20 pv power: 5684W
[site  ] DEBUG 2022/06/15 14:02:20 grid power: -4942W
[site  ] DEBUG 2022/06/15 14:02:20 grid currents: [6 7 6]A
[site  ] DEBUG 2022/06/15 14:02:20 grid margin: 18.0A @ 25.0A
[site  ] DEBUG 2022/06/15 14:02:20 site power: -4442W
[lp-1  ] DEBUG 2022/06/15 14:02:20 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/06/15 14:02:20 charger status: A

### What type of operating system are you running?

Linux

### Version

evcc version 0.50 (f3f96efb) but as this is hoermto:feature/lastmanagement1_grid version it's actually built on 0.94

premultiply, 2022-06-15:
`evcc meter --log trace` ?

rtfmjoey, 2022-06-15:
I can make a trace log on saturday as the sun is setting now and I'll be out of town for a few days.

rtfmjoey, 2022-06-20:
> The currents are obtained from OBIS codes
> 
> ```
> "1-0:31.7.0", "1-0:51.7.0", "1-0:71.7.0"
> ```
> 
> are these correct? If yes then that's what the meter returns.

That's correct but depending on fields 
"1-0:21.7.0", "1-0:41.7.0", "1-0:61.7.0", "1-0:22.7.0", "1-0:42.7.0", "1-0:62.7.0" 

This means either importing or exporting, only positive values are used:

Only exporting:
`1-0:31.7.0(002*A)
1-0:51.7.0(002*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.182*kW)
1-0:42.7.0(00.375*kW)
1-0:62.7.0(00.290*kW)`

Only importing:
1-0:31.7.0(015*A)
1-0:51.7.0(007*A)
1-0:71.7.0(006*A)
1-0:21.7.0(03.529*kW)
1-0:41.7.0(01.679*kW)
1-0:61.7.0(01.542*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)

Importing on L1 and L3 and exporting on L2:
`1-0:31.7.0(002*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.184*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.068*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.024*kW)
1-0:62.7.0(00.000*kW)`

andig, 2022-06-20:
Thanks, that should be fixable, i.e. if any of

> "1-0:22.7.0", "1-0:42.7.0", "1-0:62.7.0"

is positive the swap sign. Should also use the same logic for the power values (or rather just subtract those powers).

closed 2022-06-20 by andig