evcc-io/evcc #3649: Audi QMAuth parameters have changed.

peetermeos, 2022-06-16:
Current QMAuth params for me yield 403 because both client id and secret have changed. 
In file /vehicle/vag/idproxy/endpoint.go I propose the following:
```

const qmSecret = "e47866378ef0658ce75d71007a809f34616b9635e2ec228245784c1f63e88d06"
const qmClientId = "c95f4fd2"

func qmauth(ts int64) string {
	secret, _ := hex.DecodeString(qmSecret)
	hash := hmac.New(sha256.New, secret)
	hash.Write([]byte(strconv.FormatInt(ts, 10)))
	b := hash.Sum(nil)

	return hex.EncodeToString(b)
}
```

andig, 2022-06-17:
Thank you, closed in https://github.com/evcc-io/evcc/commit/24b0eaa7418717f2dc7ba52d6e4f329314076bdc

closed 2022-06-17 by andig

SebastiaanVanspauwen, 2022-09-04:
Hi @peetermeos,

Sorry to comment on a closed issue but I am wondering how you managed to retrieve the qmSecret? I am currently looking at myAudi app version 4.11.0 and do not see an option to retrieve the qmSecret.

In the request header I can see the following, so I know the qmClientId is '27054350', but the secret is hashed with the current timestamp.
`x-qmauth: v1:27054350:c6256effb1ea9a5d6352254260a5fb147aaf0effb5a12ce910b8f3cf41c8627c` 

Thanks in advance

peetermeos, 2022-09-12:
Hi @SebastiaanVanspauwen

The cocky answer would be "with Android Studio":) Actually the process was quite messy and involved opening the app in debugger and seeing what's going on inside and tracking down the bit of code that did the hashing.

With best,
Peeter