evcc-io/evcc #3650: Keine Bluelink-Verbindung nach boot, erst nach restart kommt die Verbindung zustande

klhedrdic, 2022-06-17:
### Describe the bug

Ein kleiner Bug in diesem großartigen Projekt, ich schreibe das mehr zur Info und nicht als wirkliche  Fehlermeldung, denn der Workarround steht ja schon in der Überschrift.

Wenn der Raspi frisch gebootet hat, kommt nie eine Verbindung zu Bluelink zustande. Die Meldung lautet

`vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get \"https://raw.githubusercontent.com/neoPix/bluelinky-stamps/master/hyundai-014d2225-8495-4735-812d-2616334fd15d.v2.json\": dial tcp [2606:50c0:8002::154]:443: i/o timeout`

Die Logs liefern dasselbe:
```
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-06-16 14:29:08 CEST; 52min ago
   Main PID: 352 (evcc)
      Tasks: 8 (limit: 415)
        CPU: 54.245s
     CGroup: /system.slice/evcc.service
             └─352 /usr/bin/evcc

Jun 16 15:19:54 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:19:54 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:20:04 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:20:04 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:20:14 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:20:14 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:20:24 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:20:24 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:20:34 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:20:34 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:20:44 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:20:44 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:20:54 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:20:54 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:21:04 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:21:04 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:21:14 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:21:14 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
Jun 16 15:21:24 broker evcc[352]: [lp-1  ] ERROR 2022/06/16 15:21:24 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: failed to download stamps: Get "https://raw.github>
```
Wenn in diesem Zustand 

`evcc vehicle`
aufgerufen wird, kommt offensichtlich eine Verbindung zustande:
```
pi@broker:~ $ evcc vehicle
[main  ] INFO 2022/06/16 15:19:44 evcc 0.94
[main  ] INFO 2022/06/16 15:19:44 using config file /etc/evcc.yaml
SoC:           97%
Charge status: C
Range:         442km
Odometer:      610km
Finish time:   2022-06-16 15:39:00 +0200 CEST
Capacity:      72kWh
OnIdentified:  Mode:pv, MinCurrent:6, MaxCurrent:16, MinSoC:20, TargetSoC:100
pi@broker:~ $
```
Mit dem Zustand habe ich mich dann tagelang rumgeschlagen, weil offensichtlich ja alles richtig war und ich auch schon einmal ein SOC über Bluelink erhalten hatte. Mehr oder weniger durch Zufall habe ich dann im laufenden Raspi einen restart ausgelöst:

`sudo systemctl restart evcc` 

und siehe da, alles läuft zur vollsten Zufriedenheit, auch reproduzierbar.

### Steps to reproduce

s.o.

### Configuration details

```
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

#### evcc Version v0.94
#### 

log: info

site:
- title: Zuhause
  meters: 
    grid: grid
    #residualPower: -1000 #1 kW Netzbezug im PV-Modus erlaubt
    pvs: pv
    
loadpoints:
- title: Garage
  charger: wallbox
  meter: charge
  vehicle: ioniq5 
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
    # soc:
    # poll:
      # mode: charging
      # interval: 60m
    # min: 0
    # target: 100
    # estimate: false
    
    # enable:
    # threshold: 0
    # delay: 1m
  guardduration: 5m 
  
chargers:
- name: wallbox
  type: template
  template: wallbe #Wallbe Eco 2.0s mit FirmwareSL-01.04.21
  host: 192.168.178.40
  port: 502

  
  
meters:
- name: grid
  type: custom ##Leistungswert aus solarview EVU-Zähler
  power:
    source: mqtt
    broker: localhost:1883
    topic: solarview/D0/DPAC
    scale: -1
  energy:  
    source: mqtt
    broker: localhost:1883
    topic: solarview/D0/GKT0
    scale: 1
- name: pv
  type: custom 
  power:    ##Leistungswert aus solarview Umrichter
    source: mqtt
    broker: localhost:1883
    topic: solarview/WR0/PAC
    scale: 1
  energy:   ##Gesamtertrag aus solarview Umrichter
    source: mqtt
    broker: localhost:1883
    topic: solarview/WR0/KT0
    scale: 1
- name: charge
  type: custom
  power:
    source: mqtt
    broker: localhost:1883
    topic: Zähler/Wallbox/P0
    scale: 1000 #Leistung in W
  energy:
    source: mqtt
    broker: localhost:1883
    topic: Zähler/Wallbox/E0

  
vehicles:
-  name: ioniq5
   type: template
   template: hyundai
   title: Ioniq5  # Wird in der Benutzeroberfläche angezeigt
   user: xxxxxxxxx  # Benutzername um auf das Gerät zuzugreifen
   password: xxxxxxxxxx # Passwort des Benutzerkontos
   vin: xxxxxxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
   capacity: 72  # Akku-Kapazität in kWh
   phases: 3  # Die maximale Anzahl der Phasen welche genutzt werden können
   mode: pv  # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll
   minSoC: 20  # Lade sofort mit maximaler Geschwindigkeit bis zu dem angegeben Ladestand, wenn der Lademodus nicht auf 'Aus' steht
   targetSoC: 100  # Bis zu welchem Ladestand (SoC) soll das Fahrzeug geladen werden
   minCurrent: 6  # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll
   maxCurrent: 16  # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll
#   identifiers: 987654321 # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/guides/vehicles/#erkennung-des-fahrzeugs-an-der-wallbox
   language: de
    

   
   

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2887 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1638 # EUR/kWh für < 30 %
    #price: 0.12  # EUR/kWh für > 30 %
```

### Log details

s.o.

### What type of operating system are you running?

Linux

### Version

evcc version 0.94

premultiply, 2022-06-17:
Zum Startzeitpunkt von evcc in der systemd-Startsequenz sind offensichtlich einige erforderliche Netzwerkdienste noch nicht (vollständig) verfügbar.

andig, 2022-06-17:
Denke ich auch. Fehler uns da noch eine Dependency im Servicefile?

closed 2022-06-21 by andig