evcc-io/evcc #3675: charger wird im pv mode trotz Netzbezug und thresholds laut Doku aktiviert

Alex-ander-s, 2022-06-20:
### Describe the bug

Ladegerät wird trotz Netzbezug aktiviert, obwohl evcc im PV modus ist und thresholds laut Dokumentation richtig angepasst sind.

### Steps to reproduce

evcc 0.95 installieren
evcc.yaml konfigurieren (siehe configuration details)
evcc starten mit /opt/evcc/bin/evcc -c /opt/evcc/conf/evcc.yaml --log debug
Oberfläche von evcc aufrufen und pv mode aktivieren, während der Netzbezug > 0W ist

### Configuration details

network:
  schema: http
  host: home.net
  port: 7070
interval: 15s # control cycle interval

# log settings
log: debug
levels:
  core: debug
  lp-1: debug
  Home: debug
  site-1: debug
  go-e-1: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
  - name: energyMeter
    type: custom
    power: # power (W)
      source: mqtt
      topic: energyMeter/instantPower
      timeout: 30s
#      scale: 0.001
    energy: # optional energy (kWh)
      source: mqtt
      topic: energyMeter/activeEnergyP
      timeout: 30s
#      scale: 0.001
#    soc: # optional battery soc (%)
#      source: mqtt
#      # ...
    currents:
      - source: mqtt
        topic: energyMeter/currentL1
        timeout: 30s
#        scale: 0.001
      - source: mqtt
        topic: energyMeter/currentL2
        timeout: 30s
#        scale: 0.001
      - source: mqtt
        topic: energyMeter/currentL3
        timeout: 30s
#        scale: 0.001

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: go-e-1
  type: template
  template: go-e
  host: 192.168.124.136

[... 146 lines cut ...]
[lp-1  ] DEBUG 2022/06/20 18:09:28 vehicle range: 201km
[lp-1  ] DEBUG 2022/06/20 18:09:28 pv charge current: 4.44A = 6A + -1.56A (358W @ 1p)
[lp-1  ] DEBUG 2022/06/20 18:09:28 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[site  ] DEBUG 2022/06/20 18:09:43 ----
[lp-1  ] DEBUG 2022/06/20 18:09:43 charge power: 1330W
[site  ] DEBUG 2022/06/20 18:09:43 grid power: 1730W
[site  ] DEBUG 2022/06/20 18:09:43 grid currents: [3.71 2.28 2.06]A
[site  ] DEBUG 2022/06/20 18:09:43 site power: 1730W
[lp-1  ] DEBUG 2022/06/20 18:09:43 charge currents: [6 0 0]A
[lp-1  ] DEBUG 2022/06/20 18:09:43 detected phases: 1p
[lp-1  ] DEBUG 2022/06/20 18:09:43 charger status: C
[lp-1  ] DEBUG 2022/06/20 18:09:43 next soc poll remaining time: 14m44s
[lp-1  ] DEBUG 2022/06/20 18:09:43 vehicle soc: 82%
[lp-1  ] DEBUG 2022/06/20 18:09:43 vehicle range: 201km
[lp-1  ] DEBUG 2022/06/20 18:09:43 pv charge current: 0A = 6A + -7.52A (1730W @ 1p)
[lp-1  ] DEBUG 2022/06/20 18:09:43 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[site  ] DEBUG 2022/06/20 18:09:58 ----

### What type of operating system are you running?

Linux

### Version

evcc version 0.95

premultiply, 2022-06-20:
`[lp-1 ] DEBUG 2022/06/20 18:09:43 climater active: true, target temp: 22.0°C, outside temp: 22.0°C`

Alex-ander-s, 2022-06-20:
[image]

> `[lp-1 ] DEBUG 2022/06/20 18:09:43 climater active: true, target temp: 22.0°C, outside temp: 22.0°C`

Eigentlich ist die Vorklimatisierung seit Tagen nicht aktiviert worden und auch jetzt nicht aktiv 
Scheinbar passt die Erkennung ob gerade klimatisiert wird nicht.
Kann man dieses Feature irgendwie disablen?
[image]

rickcrash, 2022-06-21:
Ich hatte heute morgen das gleiche Phänomen. Bis gestern lief alles gemäß Konfiguration, nach dem Update gestern auf 0.95 wurde das 3p-Laden um 6:31 Uhr bei folgenden Rahmenbedingungen gestartet:

Wallbox ABL eMH-1 started charging Citroën ëC4 in PV mode.
--------------------------
evcc Status (2022-06-21 06:31:32)
Netz-Leistung: 2.59 kW
Solar-Leistung: 0.556 kW
Eigenverbrauch: 0.823 kW

andig, 2022-06-21:
> Ich hatte heute morgen das gleiche Phänomen. 

Inwiefern? Klimatisierung aktiv? Falls nein bitte anderes Thema.

premultiply, 2022-06-21:
`evcc vehicle --log trace` ?

andig, 2022-06-22:
@Alex-ander-s im letzten Auszug gibts auf jeden Fall kein Problem mit false-positive sondern einfach eine (gute) Fehlermeldung. Genau die false Positives müssten wir aber finden, also „Klima an“ obwohl das laut App nicht der Fall und damit falsch ist..

Alex-ander-s, 2022-06-22:
@andig evcc stop, dann rm -f /tmp/evcc hat das Fehlverhalten gestern offenbar bereinigt.
Den vehicle output gestern habe ich danach erst gemacht und reingestellt.

Mit trace sieht es heute so aus (Problem tritt aber aktuell nicht mehr auf):
```
/opt/evcc/bin/evcc -c /opt/evcc/conf/evcc.yaml -l trace vehicle
[main  ] INFO 2022/06/22 08:46:18 evcc 0.95
[main  ] INFO 2022/06/22 08:46:18 using config file /opt/evcc/conf/evcc.yaml
[mqtt  ] INFO 2022/06/22 08:46:18 connecting evcc-1805765246 at tcp://localhost:1883
[mqtt  ] DEBUG 2022/06/22 08:46:18 tcp://localhost:1883 connected
[carwings] TRACE 2022/06/22 08:46:18 POST https://gdcportalgw.its-mo.com/api_v200413_NE/gdc/InitialApp_v2.php
[carwings] TRACE 2022/06/22 08:46:19 initial_app_str=**********
--
{"status":200,"message":"success","baseprm":"**********"}
[carwings] TRACE 2022/06/22 08:46:19 POST https://gdcportalgw.its-mo.com/api_v200413_NE/gdc/UserLoginRequest.php
[carwings] TRACE 2022/06/22 08:46:27 Password=**********&RegionCode=NE&UserId=***&initial_app_str=**********
--
{"status":200,"sessionId":"**********","VehicleInfoList":{"VehicleInfo":[{"nickname":"**********","telematicsEnabled":"true","vin":"***"}],"vehicleInfo":[{"nickname":"**********","telematicsEnabled":"true","vin":"***","custom_sessionid":"**********"}]},"vehicle":{"profile":{"vin":"***","gdcUserId":"","gdcPassword":"","encAuthToken":"**********","dcmId":"**********","nickname":"**********","modelyear":"18"}},"EncAuthToken":"**********","CustomerInfo":{"UserId":"**********","Language":"de-DE","Timezone":"Europe\/Vienna","RegionCode":"NE","OwnerId":"**********","EMailAddress":"***","Nickname":"**********","Country":"AT","VehicleImage":"\/content\/language\/default\/images\/img\/ph_car.jpg","UserVehicleBoundDurationSec":"946771200","VehicleInfo":{"VIN":"***","DCMID":"**********","SIMID":"**********","NAVIID":"**********","EncryptedNAVIID":"**********","MSN":"**********","LastVehicleLoginTime":"","UserVehicleBoundTime":"2020-08-15T10:33:56Z","LastDCMUseTime":"","NonaviFlg":"false","CarName":"LEAF","CarImage":"carimg7.png"}},"UserInfoRevisionNo":"1","ngTapUpdatebtn":"300000","timeoutUpdateAnime":"300000","G1Lw":"5","G1Li":"2","G1Lt":"20","G1Uw":"15","G1Ui":"2","G1Ut":"20","G2Lw":"15","G2Li":"2","G2Lt":"20","G2Uw":"15","G2Ui":"2","G2Ut":"20","resultKey":"**********"}
[carwings] TRACE 2022/06/22 08:46:27 POST https://gdcportalgw.its-mo.com/api_v200413_NE/gdc/BatteryStatusRecordsRequest.php
[carwings] TRACE 2022/06/22 08:46:29 RegionCode=NE&VIN=***&custom_sessionid=**********&tz==Europe%2FVienna
--
{"status":200,"VoltLabel":{"HighVolt":"240","LowVolt":"120"},"BatteryStatusRecords":{"OperationResult":"START","OperationDateAndTime":"2022\/06\/22 08:45","BatteryStatus":{"BatteryChargingStatus":"NORMAL_CHARGING","BatteryCapacity":"240","BatteryRemainingAmount":"201","BatteryRemainingAmountWH":"30640","BatteryRemainingAmountkWH":"","SOC":{"Value":"84"}},"PluginState":"CONNECTED","CruisingRangeAcOn":"215000","CruisingRangeAcOff":"221000","TimeRequiredToFull200_6kW":{"HourRequiredToFull":"2","MinutesRequiredToFull":"30"},"NotificationDateAndTime":"2022\/06\/22 06:45","TargetDate":"2022\/06\/22 06:45"}}
[carwings] TRACE 2022/06/22 08:46:29 POST https://gdcportalgw.its-mo.com/api_v200413_NE/gdc/RemoteACRecordsRequest.php
[carwings] TRACE 2022/06/22 08:46:31 RegionCode=NE&VIN=***&custom_sessionid=**********&tz=Europe%2FVienna
--
{"status":200,"RemoteACRecords":[]}
SoC:           84%
Charge status: C
Range:         215km
Climater:      climate status not available
Capacity:      36kWh
```
 
Ich habe es versucht nochmal zu reproduzieren (Downgrade evcc 0.94, dann wieder auf evcc 0.95 und in den pv modus gehen), schaffe es aktuell aber nicht mehr.
Ich werde es jedenfalls weiter beobachten.

Habt ihr eine Idee bzw. gibt es eine einfache Möglichkeit Benachrichtigungen bei "climater" events zu erzeugen, sodass ich es gleich mitbekomme und tracen kann, wenn es wieder passiert?

Eine Idee wäre (Beispiel):
```
# push messages
messaging:
  events:
    climateStart: # climate start event
      title: Climater started
      msg: Climater target temp: ${targetTemp:%.1fk} C and outside temp: ${outsideTemp}
    climateStop: # climate stop event
      title: Climater stopped
      msg: Climater target temp: ${targetTemp:%.1fk} C and outside temp: ${outsideTemp}
```

Vom Nissan Server bekomme ich ohnehin eine Mailbenachrichtigung, wenn ich die Klimaanlage doch mal einschalte.

andig, 2022-06-22:
> Habt ihr eine Idee bzw. gibt es eine einfache Möglichkeit Benachrichtigungen bei "climater" events zu erzeugen, sodass ich es gleich mitbekomme und tracen kann, wenn es wieder passiert?

Wäre möglich. Bevor wir da was bauen lass uns aber mal schauen, ob es überhaupt wieder auftritt und was dann im TRACE Log steht. Vllt ists ja offensichtlich.

Ich mache hier mal solange zu, wenn Du ein Update schickst sehe ich das.

closed 2022-06-22 by andig