evcc-io/evcc #3685: First OCPP Errors with Wallbe Pro - invalid character '\x00' in string literal

Hofyyy, 2022-06-21:
### Describe the bug

OCPP Mode does not start completely, because of the following error:

[ocpp  ] ERROR 2022/06/21 18:56:01 invalid character '\x00' in string literal

But the first handshake is working:

[2,"1986641971","GetConfiguration",{}]

[3,"1986641971",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":true,"value":"true"},{"key":"ClockAlignedDataInterval","readonly":true,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"0"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":true,"value":"false"},{"key":"LocalPreAuthorize","readonly":true,"value":"false"},{"key":"MeterValuesAlignedData","readonly":true,"value":"false"},{"key":"MeterValuesSampledData","readonly":true,"value":"true"},{"key":"StopTransactionOnEVSideDisconnect","readonly":true,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":true,"value":"true"},{"key":"StopTxnAlignedData","readonly":true,"value":"false"},{"key":"StopTxnSampledData","readonly":true,"value":"false"},{"key":"MeterValueSampleInterval","readonly":false,"value":"300"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"false"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"true"},{"key":"ResetRetries","readonly":false,"value":"3"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"60"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":true,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"1"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"true"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"1"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"1"},{"key":"PricePerKwh","readonly":false,"value":"0"},{"key":"AllowOfflineTxForFreemode","readonly":true,"value":"false"},{"key":"WebSocketPingInterval","readonly":false,"value":"60"},{"key":"UID_HighRegisterFirst","readonly":false,"value":"false"},{"key":"ConnectorMaximumCurrent","readonly":false,"value":"255"},{"key":"ConnectorMinimumCurrent","readonly":false,"value":"1"},{"key":"GlobalMaximumCurrent","readonly":false,"value":"16"},{"key":"MeterValueSendTotalEnergy","readonly":false,"value":"false"},{"key":"EMOC_PublicKey_Conn_1","readonly":true,"value":"unknown

### Steps to reproduce

Configure: Wallbe Pro in OCPP
- ws://192.168.20.144:8887/wallbeP9741

Info:The default ID from the vendor is wallbeP#SerialNumber

Configure: OCPP in evcc config

chargers:
  - name: wallbe
    type: ocpp # Wallbe charger
    stationid: wallbeP9741 # ModBus address
    idtag: 1010

### Configuration details

chargers:
  - name: wallbe
    type: ocpp # Wallbe charger
    stationid: wallbeP9741 # ModBus address
    idtag: 1010

### Log details

./evcc
[main  ] INFO 2022/06/21 18:54:00 evcc 0.95
[main  ] INFO 2022/06/21 18:54:00 using config file /Users/sebastianhofmann/Downloads/evcc_0.95/evcc.yaml
[main  ] INFO 2022/06/21 18:54:00 listening at :7070
[ocpp  ] ERROR 2022/06/21 18:56:01 invalid character '\x00' in string literal
[main  ] FATAL 2022/06/21 18:56:11 cannot create charger 'wallbe': cannot create charger 'ocpp': timeout

### What type of operating system are you running?

Linux

### Version

evcc version 0.95
Mac OS

andig, 2022-06-22:
@Hofyyy 

> {"key":"EMOC_PublicKey_Conn_1","readonly":true,"value":"unknown

sieht so aus als würde die Wallbe kein JSON schicken. Das wär ein Fehler beim Hersteller.

Hofyyy, 2022-06-22:
Das kann ich mir garnicht vorstellen :-).

Ich würde heute mal ein Wireshark capture ziehen, damit wir die Kommunikation vollständig haben. Wenn wir das dann wirklich beim hersteller sehen mache ich da einen case auf.

Hofyyy, 2022-06-22:
Anbei die komplette Kommunikation mit der Box. Einmal als RAW und einmal als TXT schon vorgefiltert.
[wallbe_ocpp_communication.zip](https://github.com/evcc-io/evcc/files/8959039/wallbe_ocpp_communication.zip)

andig, 2022-06-22:
Mir fehlt die Zeit das zu durchforsten- könntest Du den Punkt mit dem Abbruch extrahieren?

Hofyyy, 2022-06-22:
> Mir fehlt die Zeit das zu durchforsten- könntest Du den Punkt mit dem Abbruch extrahieren? Schickt die Wallbe gültiges JSON? Falls nein können wir hier zu machen ;)

Also ich habe es hier geprüft:
- https://jsonformatter.curiousconcept.com/#
- VALID (RFC 8259)

Also ich gucke mir das jetzt mal im Detail an.
Wenn ich wetten müsste verschluckt sich EVCC einfach am JSON der Config.

@andig 
Entweder ist es die JSON vom der Config oder die letzte. Da müsstest du nicht viel gucken. Es ist super schnell im *.rtf zu sehen. Aber ich gucke auch schon mal.

andig, 2022-06-23:
Der Fehler ist ja

>  invalid character '\x00' in string literal

Irgendwo ist also eine `0x00` in einem String und das ist halt nicht zulässig (ohne in den RFC geschaut zu haben). Wenn ich raten müsste dann im `EMOC_PublicKey_Conn_1`.

Hofyyy, 2022-06-23:
> Der Fehler ist ja
> 
> > invalid character '\x00' in string literal
> 
> Irgendwo ist also eine `0x00` in einem String und das ist halt nicht zulässig (ohne in den RFC geschaut zu haben). Wenn ich raten müsste dann im `EMOC_PublicKey_Conn_1`.

Ich hab mal einen Fall bei Wallbe (Compleo) aufgemacht. Die Antworten normalerweise in 24 Stunden. Mal gucken was Sie sagen.

closed 2022-06-23 by andig

xantalor, 2022-06-24:
> @xantalor könntest Du schauen wo der Fehler her kommt? Haben wie eine Chance, das Logging hier auf DEBUG zu ziehen und auf TRACE die JSON Nachrichten auszugeben? Habe nicht gecheckt, ob die Library das raus rückt...

Sorry eben erst dazu gekommen. Wir können auf Trace hoch und genauer reinschauen was wir bekommen. 
```
levels:
  ocpp-{stationid}:{connectorid}: trace
```

Hofyyy, 2022-06-27:
Also Wallbe scheint den "NULL" Wert bei Boxen zurückzugeben, die nicht geeicht sind, da man diese normalerweise nicht per OCPP benutzt. Ich bin mir noch nicht sicher ob das wirklich nicht erlaubt ist. Alle mit denen ich in der FIrma gesprochen habe meinten, das ein Parser aussteigt ist wahrscheinlich, aber es ist wahrscheinlich trotzdem erlaubt :-P

Wenn ich mal langeweile habe debugge ich das mal.

Und ich schreibe mal weiter mit dem Support und kläre, warum sie nicht z.B. "Nicht Geeicht" in den Key schreiben.

andig, 2022-06-27:
In https://datatracker.ietf.org/doc/html/rfc4627 finde ich nichts von `0x00`, also ist es auch nicht erlaubt.

>   The representation of strings is similar to conventions used in the C
   family of programming languages.  A string begins and ends with
   quotation marks.  All Unicode characters may be placed within the
   quotation marks except for the characters that must be escaped:
   quotation mark, reverse solidus, and the control characters (U+0000
   through U+001F).

Auch in C ist 0x00 kein Teil des Strings.

andig, 2022-06-27:
Ich mach hier mal zu. Evcc tut wie erwartet.