evcc-io/evcc #3717: Porsche Taycan

makinitright, 2022-06-26:
### Describe the bug

When adding my Porsche Taycan to evcc, the following log is generated with `sudo evcc -l debug vehicle`:

```
porsche
-------
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'porsche': vehicle is not paired with the My Porsche account
Capacity: 0kWh
```

The configuration was done with the configurator.

### Steps to reproduce

Run config or manually add to the the yaml and start evcc: 

```
vehicles:
- type: template
  template: porsche
  title: Taycan
  user: ...
  password: ...
  vin: ...
  capacity: 71
  phases: 3
  mode: pv
  minSoC: 5
  targetSoC: 100
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev3
```

### Configuration details

```
vehicles:
- type: template
  template: porsche
  title: Taycan
  user: ...
  password: ...
  vin: ...
  capacity: 71
  phases: 3
  mode: pv
  minSoC: 5
  targetSoC: 100
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev3
```

### Log details

[main  ] INFO 2022/06/26 13:37:45 evcc 0.95
[main  ] INFO 2022/06/26 13:37:45 using config file /etc/evcc.yaml
[main  ] INFO 2022/06/26 13:37:45 listening at :7070
[openwb] INFO 2022/06/26 13:37:50 connecting evcc-1263312707 at tcp://10.10.60.5:1883
[openwb] DEBUG 2022/06/26 13:37:50 tcp://10.10.60.5:1883 connected
[openwb] DEBUG 2022/06/26 13:37:50 tcp://10.10.60.5:1883 subscribe openWB/system/Timestamp
[openwb] DEBUG 2022/06/26 13:37:50 openWB/lp/1/boolChargePointConfigured wait for initial value
[site  ] INFO 2022/06/26 13:37:56 site config:
[site  ] INFO 2022/06/26 13:37:56   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/06/26 13:37:56     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/06/26 13:37:56     pv 1:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/06/26 13:37:56 loadpoint 1:
[lp-1  ] INFO 2022/06/26 13:37:56   mode:        pv
[lp-1  ] INFO 2022/06/26 13:37:56   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/06/26 13:37:56   meters:      charge ✓
[lp-1  ] INFO 2022/06/26 13:37:56     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/06/26 13:37:56   vehicles:    ✓
[lp-1  ] INFO 2022/06/26 13:37:56     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/06/26 13:37:56 vehicle updated: unknown -> unavailable
[site  ] DEBUG 2022/06/26 13:37:56 ----
[lp-1  ] DEBUG 2022/06/26 13:37:56 charge power: 5680W

### What type of operating system are you running?

Linux

### Version

evcc version 0.95

premultiply, 2022-06-26:
`vehicle is not paired with the My Porsche account`

Hast du denn das Fahrzeug im My Porsche account hinzugefügt?

andig, 2022-06-26:
Nothing we can do. If it's not paired it cannot be accessed. Please check with Porsche.

makinitright, 2022-06-26:
Danke für die Antworten. Das Auto ist natürlich hinzugefügt und über das äquivalente Porsche Add-On für HASS funktioniert es auch. Sonst noch eine Idee? 😔

andig, 2022-06-26:
Zugangsdaten an info@evcc.io?

makinitright, 2022-06-26:
> Zugangsdaten an [info@evcc.io](mailto:info@evcc.io)?

Mail ging eben raus. Besten Dank! Ich weiß, dass es in der Vergangenheit wunderbar funktioniert hatte. Aber mit der alten evcc.yaml bzw. dem Fahrzeug-Eintrag geht es auch nicht - liegt also vermutlich nicht an der Config.