evcc-io/evcc #3718: Audi E-Tron - Unknown

makinitright, 2022-06-26:
### Describe the bug

When adding my Audi E-Tron to evcc, the following log is generated with sudo evcc -l debug vehicle:

```
audi
----
.SoC:          unknown error
Charge status: unknown error
Range:         unknown error
Odometer:      unknown error
Finish time:   unknown error
Climater:      unknown error
Capacity:      0kWh
OnIdentified:  MaxCurrent:16, MinSoC:5, TargetSoC:100, Mode:pv, MinCurrent:6
```

### Steps to reproduce

Start evcc after configuration.

### Configuration details

```
- type: template
  template: etron
  title: E-Tron
  user: ...
  password: ...
  vin: ...
  phases: 3
  mode: pv
  minSoC: 5
  targetSoC: 100
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev4
```

### Log details

```
[main  ] INFO 2022/06/26 13:37:45 evcc 0.95
[main  ] INFO 2022/06/26 13:37:45 using config file /etc/evcc.yaml
[main  ] INFO 2022/06/26 13:37:45 listening at :7070
[openwb] INFO 2022/06/26 13:37:50 connecting evcc-1263312707 at tcp://10.10.60.5:1883
[openwb] DEBUG 2022/06/26 13:37:50 tcp://10.10.60.5:1883 connected
[openwb] DEBUG 2022/06/26 13:37:50 tcp://10.10.60.5:1883 subscribe openWB/system/Timestamp
[openwb] DEBUG 2022/06/26 13:37:50 openWB/lp/1/boolChargePointConfigured wait for initial value
[site  ] INFO 2022/06/26 13:37:56 site config:
[site  ] INFO 2022/06/26 13:37:56   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/06/26 13:37:56     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/06/26 13:37:56     pv 1:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/06/26 13:37:56 loadpoint 1:
[lp-1  ] INFO 2022/06/26 13:37:56   mode:        pv
[lp-1  ] INFO 2022/06/26 13:37:56   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/06/26 13:37:56   meters:      charge ✓
[lp-1  ] INFO 2022/06/26 13:37:56     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/06/26 13:37:56   vehicles:    ✓
[lp-1  ] INFO 2022/06/26 13:37:56     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/06/26 13:37:56 vehicle updated: unknown -> unavailable
[site  ] DEBUG 2022/06/26 13:37:56 ----
[lp-1  ] DEBUG 2022/06/26 13:37:56 charge power: 5680W
```

### What type of operating system are you running?

Linux

### Version

0.95

andig, 2022-06-26:
Please provide trace log

makinitright, 2022-06-26:
```
Jun 26 18:31:32 energy-app systemd[1]: Started evcc.
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42 loadpoint 1:
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42   mode:        pv
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42   meters:      charge ✓
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42     charge:    power ✓ energy ✓ currents ✓
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42   vehicles:    ✓
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
Jun 26 18:31:42 energy-app evcc[3427]: [lp-1  ] INFO 2022/06/26 18:31:42 vehicle updated: unknown -> E-Tron
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] ERROR 2022/06/26 18:31:44 vehicle odometer: unknown error
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 set charge mode: pv
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 set min current: 6
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 set max current: 16
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 set min soc: 5
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 set target soc: 100
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 charge power: 0W
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 charge currents: [0 0 0]A
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 charger status: B
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] ERROR 2022/06/26 18:31:44 vehicle soc: unknown error
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 max charge current: 16A
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 charger enable
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 wake-up timer: start
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 pv timer elapse
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 pv timer inactive
Jun 26 18:31:44 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:44 charge power: 0W
Jun 26 18:31:45 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:45 charge currents: [0 0 0]A
Jun 26 18:31:45 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:45 charger status: B
Jun 26 18:31:45 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:45 next soc poll remaining time: 59m59s
Jun 26 18:31:45 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:45 pv timer elapse
Jun 26 18:31:45 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:45 pv timer inactive
Jun 26 18:31:54 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:54 charge power: 0W
Jun 26 18:31:54 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:54 charge currents: [0 0 0]A
Jun 26 18:31:54 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:54 charger status: B
Jun 26 18:31:54 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:54 next soc poll remaining time: 59m50s
Jun 26 18:31:54 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:54 pv timer elapse
Jun 26 18:31:54 energy-app evcc[3427]: [lp-1  ] DEBUG 2022/06/26 18:31:54 pv timer inactive
```

Hoffe das ist das, was Du meinst. Das wird in den syslog nach Start von EVCC geschrieben.

andig, 2022-06-26:
Nein, bitte log level trace, relevant ist nur der Audi. Siehe Doku.

andig, 2022-07-01:
Ping @makinitright