evcc-io/evcc #3722: configure: invalid default value applied

andig, 2022-06-26:
Hallo, ich habe im Moment nochmal  evcc configure aufgerufen und musste leider feststellen, dass du hier der Vorgabe-Wert 1 für die ID eingetragen ist. Du solltest hier für die ID den Wert 255 als Vorgabe eintragen.
Dieses Problem war am Anfang auch schon mal und ich hatte es dir damals auch schon mal mitgeteilt. Dann gab es soweit ich noch weiß, eine Version von evcc configure bei der die Vorgabe 255 als Standardwert angezeigt wurde. Aber offenbar ist dann irgendwie wieder die 1 hinein gerutscht. Also hier bitte auf 255 abändern und das Problem sollte gelöst sein.

Gruß Jürgen

_Originally posted by @hambest in https://github.com/evcc-io/evcc/discussions/2803#discussioncomment-3027282_

RalfArduino, 2022-06-28:
Es werden auch für Kostal Wechselrichter und KSEM die falschen Werte vorgeschlagen:
Modbus ID 1 anstatt 71, 502 anstatt 1502.
Für das serielle Interface zur ABL Wallbox sind auch die falschen Werte vorgegeben:
9600 anstatt 38400. Die Konfigurierbarkeit "8N1" ist neu, funktioniert so aber auch nicht.

Vermutlich ist da wohl generell was schiefgegangen mit allen Defaultwerten.

RalfArduino, 2022-06-28:
Noch eine Anmerkung: Bei der Konfiguration der ABL-Wallbox über USB/seriell Wandler wird auch (unsinnigerweise) 
die Modbus ID abgefragt. In jedem Fall kommt man mit der aktuellen Version 0.95 nicht fehlerfrei durch die Konfiguration, 
egal ob man die Defaultwerte oder die richtigen Werte (38400 Baud, etc.) eingibt.

andig, 2022-06-28:
Das Problem ist generell immer da. Kannst Du nachvollziehen ob das mit einer früheren Version fehlerfrei ging?

RalfArduino, 2022-06-28:
Die Installation (mit Defaultwerten für Kostal & ABL mit USB/Seriell) hat in früheren Versionen einwandfrei funktioniert.
Genau deswegen mach ich jetzt darauf aufmerksam.
Ich leider nicht genau nachvollzien mit welcher Version ich zuletzt installiert habe, aber ich vermute 0.92 oder früher.

andig, 2022-06-28:
0.92 funktioniert. Danke für den Hinweis, das macht die Fehlersuche einfacher.

Saharel001, 2022-07-03:
@andig Für Kostal Plenticore sind die Vorgaben (71, 1502) im letzten Release ok, nur in der nightly passen diese nicht. (1, 502)

Saharel001, 2022-07-03:
> 0.92 funktioniert. Danke für den Hinweis, das macht die Fehlersuche einfacher.

0.94.0 Kostal default values = OK

Saharel001, 2022-07-19:
@andig Moin, wann ist die Änderung im Master zu erwarten?

andig, 2022-07-22:
Ich hoffe die Fehler sind jetzt endgültig behoben- Feedback wäre Klasse, ab morgen im Nightly!

PR #3777 (2022-07-22): Templates: fix modbus default values not applied
Fix #3722
closed 2022-07-22 by andig