evcc-io/evcc #3738: Laden des Hyundai Kona (ohne Bluelink) lässt sich nicht über AUS stoppen

Stefhorn, 2022-06-29:
### Describe the bug

1. Wenn der Ladevorgang im Stop- Modus gestartet wird hört man abwechselnd das Relay im Onboardcharger sowie das Leistungsschütz in der Pracht Alpha XT Wallbox ein und ausschalten. Dabei geht auch die Ladelampe des Kona's auf Störung (rot).

2. Wenn der Ladevorgang im Now oder Min-Modus gestartet wird funktioniert alles normal. Will man dann den Ladevorgang mit Stop- Modus beenden wird die Ladeleistung auch herunter gefahren aber dann passiert das gleiche wie unter 1.

3. Wenn der Ladevorgang im PV-Modus gestartet wird funktioniert alles normal solange ausreichend PV-Strom zur Verfügung steht. Sobald dies nicht mehr gegeben ist wird die Ladeleistung auch herunter gefahren aber dann passiert das gleiche wie unter 

Meist erscheint dann nach ein paar Klicks der Relays die Meldung charger logic error: disabled but charging.

EVCC muss dann neu gestartet werden.

### Steps to reproduce

Ladekabel einstecken und entsprechende Modi auswählen

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE2ODU2OTMzMzEsImlhdCI6MTY1NDE1NzMzMSwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlN0ZWZob3JuIn0.nAjL0-RLFqw1LQQPScyxYgKMTYRuw8vLViyrp1qQA7c

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.188.62
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.188.63
  password: StHoHKL2022!
  name: pv2

chargers:
- type: pracht-alpha
  id: 1
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  name: wallbox3
  timeout: 2s

loadpoints:
- title: Carport
  charger: wallbox3
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2

### Log details

pi@raspberrypi:~ $ evcc --log debug
[main  ] INFO 2022/06/29 16:41:46 evcc 0.95
[main  ] INFO 2022/06/29 16:41:46 using config file /etc/evcc.yaml
[main  ] INFO 2022/06/29 16:41:46 listening at :7070
[site  ] INFO 2022/06/29 16:41:50 site config:
[site  ] INFO 2022/06/29 16:41:50   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/06/29 16:41:50     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/06/29 16:41:50     pv 1:      power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/06/29 16:41:50 loadpoint 1:
[lp-1  ] INFO 2022/06/29 16:41:50   mode:        pv
[lp-1  ] INFO 2022/06/29 16:41:50   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/06/29 16:41:50   meters:      charge ✗
[lp-1  ] INFO 2022/06/29 16:41:50   vehicles:    ✗
[lp-1  ] ERROR 2022/06/29 16:41:52 charger: serial: timeout
[site  ] DEBUG 2022/06/29 16:41:52 ----
[server] FATAL 2022/06/29 16:41:52 remove /tmp/evcc: operation not permitted
[lp-1  ] DEBUG 2022/06/29 16:41:52 charge power: 0W

pi@raspberrypi:~ $ sudo journalctl -fau evcc
-- Journal begins at Mon 2022-04-04 14:17:02 CEST. --
Jun 29 14:58:59 raspberrypi evcc[861]: [lp-1  ] INFO 2022/06/29 14:58:59 start charging ->
Jun 29 15:03:02 raspberrypi evcc[861]: [lp-1  ] WARN 2022/06/29 15:03:02 charger logic error: disabled but charging
Jun 29 15:03:09 raspberrypi evcc[861]: [lp-1  ] WARN 2022/06/29 15:03:09 charger logic error: disabled but charging
Jun 29 15:03:19 raspberrypi evcc[861]: [lp-1  ] WARN 2022/06/29 15:03:19 charger logic error: disabled but charging
Jun 29 15:03:22 raspberrypi evcc[861]: [lp-1  ] WARN 2022/06/29 15:03:22 charger logic error: disabled but charging
Jun 29 15:05:20 raspberrypi evcc[861]: [lp-1  ] WARN 2022/06/29 15:05:20 charger logic error: disabled but charging
Jun 29 15:05:30 raspberrypi evcc[861]: [lp-1  ] WARN 2022/06/29 15:05:30 charger logic error: disabled but charging
Jun 29 15:05:40 raspberrypi evcc[861]: [lp-1  ] WARN 2022/06/29 15:05:40 charger logic error: disabled but charging
Jun 29 15:05:49 raspberrypi evcc[861]: [lp-1  ] INFO 2022/06/29 15:05:49 stop charging <-
Jun 29 15:07:49 raspberrypi evcc[861]: [lp-1  ] INFO 2022/06/29 15:07:49 car disconnected

### What type of operating system are you running?

Linux

### Version

pi@raspberrypi:~ $ evcc -v
evcc version 0.95

Stefhorn, 2022-06-29:
Hier noch ein [Video] ( https://photos.app.goo.gl/9MeVGcRAA4cf9iEF7)

andig, 2022-06-29:
Scheint als würde das ausschalten nicht funktionieren.

/cc @premultiply

Stefhorn, 2022-06-30:
Könnte mir bitte jemand  erklären wie die Kommunikation zwischen EVCC, Wallbox und Fahrzeug normalerweise abläuft.
Was passiert eigentlich wenn der Stop- Modus betätigt wird?
Ich habe das Gefühl, dass der Ladevorgang zwar beendet wird, da aber der Ladestecker noch eingesteckt ist will das Fahrzeug sofort erneut laden, was dann EVCC wieder unterbindet. Somit schaltet das Schütz in der Wallbox ständig ein und aus, was dann das Relay im OBC triggert.

premultiply, 2022-06-30:
Die minimalistische Schnittstelle dieser Box ist ja zudem so minimalistisch dokumentiert dass man da kaum etwas falsch machen kann.

Leider finden sich dazu auch keine anderen Implementierungen um zu vergleichen ob und was dort ggf. anders gemacht würde.

Vom Code her sehe ich da also bisher kaum einen Ansatz dass hier wissentlich etwas falsch gemacht wurde bzw. dass hier seitens evcc etwas falsch implementiert ist.

premultiply, 2022-06-30:
Noch eine Idee: Offensichtlich hast du eine Box mit zwei Ladepunkten (Video), oder?

Könntest du mal bitte den anderen Ladepunkt testen?

premultiply, 2022-06-30:
Eine korrekte Konfiguration für beide Ladepunkte könnte so aussehen:

```yaml
chargers:
- type: pracht-alpha
  id: 1
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  name: wallbox3
  timeout: 2s
  vehicle: 1
- type: pracht-alpha
  id: 1
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  name: wallbox4
  timeout: 2s
  vehicle: 2

loadpoints:
- title: Carport 1
  charger: wallbox3
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
- title: Carport 2
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
```

Stefhorn, 2022-07-01:
Richtig, die Wallbox hat 2 Ladepunkte. Und mit der geänderten evcc.yaml habe ich jetzt auch beide auf dem UI. Danke für den Tip.
Ich habe auf dem linken 3 Phasen und auf dem rechten 1 Phase angeschlossen. So kann ich wenigstens manuell zwischen 1 und 3ph-Laden wählen.

loadpoints:
- title: Carport 1
  charger: wallbox3
  mode: pv
  phases: 1    # Von 3 auf 1 geändert

Der Stop-Modus klappt leider aber noch nicht. Ich werde jetzt mal die Logs erstellen.

Stefhorn, 2022-07-01:
2. Log im AUS modus , 1. Phase, Ladesleistung 3,7 kW

pi@raspberrypi:~ $ evcc --log trace
[main  ] INFO 2022/07/01 15:56:51 evcc 0.95
[main  ] INFO 2022/07/01 15:56:51 using config file /etc/evcc.yaml
[main  ] INFO 2022/07/01 15:56:51 listening at :7070
[sma   ] TRACE 2022/07/01 15:56:51 send 192.168.188.62: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:51 recv 192.168.188.62: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:51 new energy meter at 192.168.188.62 - Serial=3011956767
[sma   ] TRACE 2022/07/01 15:56:51 send 192.168.188.62: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 send discover package
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.63: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.62: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.62: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 send discover package
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.63: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.62: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.62: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.62: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 new energy meter at 192.168.188.62 - Serial=3011956767
[sma   ] TRACE 2022/07/01 15:56:52 found device %!s(uint32=3011956767) at 192.168.188.62
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 new inverter at 192.168.188.63 - Serial=1980466877
[sma   ] TRACE 2022/07/01 15:56:52 new inverter at 192.168.188.63 - Serial=1980466877
[sma   ] TRACE 2022/07/01 15:56:52 found device %!s(uint32=1980466877) at 192.168.188.63
[sma   ] TRACE 2022/07/01 15:56:52 login for 192.168.188.63:9522
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 requestValues for 192.168.188.63:9522: 0x5100 0x263F00 0x263FFF
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 requestValues for 192.168.188.63:9522: 0x5100 0x295A00 0x295AFF
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:52 requestValues for 192.168.188.63:9522: 0x5100 0x411E00 0x4120FF
[sma   ] TRACE 2022/07/01 15:56:52 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:53 send discover package
[sma   ] TRACE 2022/07/01 15:56:53 recv 192.168.188.63: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2022/07/01 15:56:53 recv 192.168.188.62: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2022/07/01 15:56:53 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:53 requestValues for 192.168.188.63:9522: 0x5100 0x464000 0x4642FF
[sma   ] TRACE 2022/07/01 15:56:53 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:53 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:53 requestValues for 192.168.188.63:9522: 0x5100 0x464800 0x4655FF
[sma   ] TRACE 2022/07/01 15:56:53 send 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/07/01 15:56:53 recv 192.168.188.63: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[... cut ...]
[pracht] TRACE 2022/07/01 15:59:47 modbus: send 01 04 00 03 00 01 c1 ca
[pracht] TRACE 2022/07/01 15:59:47 modbus: recv 01 04 02 00 00 b9 30
Charge status: A
Enabled:       false

wallbox4
--------
[pracht] TRACE 2022/07/01 15:59:47 modbus: send 01 04 00 6b 00 01 40 16
[pracht] TRACE 2022/07/01 15:59:48 modbus: recv 01 04 02 00 02 38 f1
[pracht] TRACE 2022/07/01 15:59:48 modbus: send 01 04 00 04 00 01 70 0b
[pracht] TRACE 2022/07/01 15:59:48 modbus: recv 01 04 02 00 00 b9 30
Charge status: C
Enabled:       false

Stefhorn, 2022-07-01:
Bemerkung:

[server] FATAL 2022/07/01 15:56:56 remove /tmp/evcc: operation not permitted

Habe ich schon gelöscht aber machte auch kein Unterschied.
Nach jedem Neustart des Raspi ist die Datei wieder da.

andig, 2022-07-01:
Das ist ziemlich unlesbar: bitte formatieren und sma da raus nehmen, das ist für die Kommunikation mit der Wallbox irrelevant.

Stefhorn, 2022-07-01:
Sorry. Mach ich nochmal. Dieser Log trace ist ja nur eine Momentaufnahme. Gibt es auch die Option einen kontinuierlichen Log zu erzeugen?

Stefhorn, 2022-07-02:
pi@raspberrypi:~ $ sudo journalctl -fau evcc
-- Journal begins at Mon 2022-04-04 14:17:02 CEST. --
Jul 02 15:54:21 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:21 pv timer elapse-------------------- Ladung läuft auf PV
Jul 02 15:54:21 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:21 pv timer inactive
Jul 02 15:54:21 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:21 charge power: 3680W
Jul 02 15:54:21 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:21 charger status: C
Jul 02 15:54:21 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:21 pv charge current: 43.6A = 16A + 27.6A (-6339W @ 1p)
Jul 02 15:54:21 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:21 pv timer reset
Jul 02 15:54:21 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:21 pv timer inactive
Jul 02 15:54:25 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:25 charge power: 3680W
Jul 02 15:54:26 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:26 charger status: C
Jul 02 15:54:26 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:26 pv charge current: 43.7A = 16A + 27.7A (-6375W @ 1p)
Jul 02 15:54:35 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:35 charge power: 3680W
Jul 02 15:54:44 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:44 set charge mode: minpv -------------------- Ladung läuft auf Min+PV
Jul 02 15:54:44 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:44 pv timer elapse
Jul 02 15:54:44 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:44 pv timer inactive
Jul 02 15:54:44 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:44 charge power: 3680W
Jul 02 15:54:44 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:44 charger status: C
Jul 02 15:54:45 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:45 pv charge current: 50A = 16A + 34A (-7817W @ 1p)
Jul 02 15:54:45 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:45 pv timer reset
Jul 02 15:54:45 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:45 pv timer inactive
Jul 02 15:54:45 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:45 charge power: 3680W
Jul 02 15:54:46 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:46 charger status: C
Jul 02 15:54:47 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:47 pv charge current: 50A = 16A + 34A (-7817W @ 1p)
Jul 02 15:54:55 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:54:55 charge power: 3680W
Jul 02 15:55:04 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:04 set charge mode: now ----------------------Ladung läuft auf Now
Jul 02 15:55:04 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:04 pv timer elapse
Jul 02 15:55:04 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:04 pv timer inactive
Jul 02 15:55:04 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:04 charge power: 3680W
Jul 02 15:55:05 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:05 charger status: C
Jul 02 15:55:05 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:05 charge power: 3680W
Jul 02 15:55:06 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:06 charger status: C
**Jul 02 15:55:15 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:15 set charge mode: off  ------------------------Ladung gestoppt, Schütz fällt ab und sofort wieder an solange** 
Jul 02 15:55:15 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:15 pv timer elapse
Jul 02 15:55:15 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:15 pv timer inactive				bis wieder in charge mode gewechselt wird. Wallbox geht nach ca 6x
Jul 02 15:55:15 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:15 charge power: 3680W				auf Störung. 
Jul 02 15:55:15 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:15 charger status: C
Jul 02 15:55:16 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:16 charger disable
Jul 02 15:55:16 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:16 wake-up timer: stop
Jul 02 15:55:16 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:16 charge power: 0W
Jul 02 15:55:25 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:25 charge power: 0W
Jul 02 15:55:26 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:26 charger status: C 
Jul 02 15:55:26 raspberrypi evcc[829]: [lp-2  ] WARN 2022/07/02 15:55:26 charger logic error: disabled but charging  ---- Fehlermeldung auf UI
Jul 02 15:55:35 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:34 charge power: 0W
Jul 02 15:55:35 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:35 charge power: 0W
Jul 02 15:55:36 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:36 set charge mode: now ----------------------Ladung läuft auf Now
Jul 02 15:55:36 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:36 pv timer elapse
Jul 02 15:55:36 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:36 pv timer inactive
Jul 02 15:55:36 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:36 charge power: 0W
Jul 02 15:55:37 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:37 charger status: B
Jul 02 15:55:37 raspberrypi evcc[829]: [lp-2  ] INFO 2022/07/02 15:55:37 stop charging <-
Jul 02 15:55:38 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:38 charger enable
Jul 02 15:55:38 raspberrypi evcc[829]: [lp-2  ] DEBUG 2022/07/02 15:55:38 wake-up timer: start

Stefhorn, 2022-07-02:
https://photos.app.goo.gl/wtiFgfdodDmRaMAZ7

andig, 2022-07-02:
Auch im letzten Log fehlt wieder das tace für den Charger

Stefhorn, 2022-07-04:
Sorry, wie bekomme ich das trace für den Charger in den Log?

andig, 2022-07-04:
Schau bitte in die Doku!

andig, 2022-07-04:
Ich sehe da kein Problem. Du schaltest auf "off" und das Auto geht auf Status B (verbunden). Dann schaltest Du wieder auf "now" und das Laden startet. Alles wie erwartet?!

Stefhorn, 2022-07-04:
Diesmal gab es nicht mal den "Fehler charger logic error: disabled but charging "
Leider sieht man auf dem Log nicht, dass sich der Schütz der Wallbox zwischen "off" und "now" 5 mal aus und eingeschaltet hatte. Wenn ich noch länger warte geht die Wallbox auf Störung.
Es scheint so als wenn EVCC das Laden problemlos beendet (charge power: 0W), die Wallbox den Ladevorgang aber sofort wieder in Gang setzt. Da aber EVCC immer noch charge power: 0W sendet, der Ladevorgang sofort wieder abbricht.
Kann man aus den Modbus send / recv codes herauslesen was da passiert?

andig, 2022-07-04:
Kann man wenn du dazu mal ein log hättest. Oder ist das eins vom Fehlerzustand?

Stefhorn, 2022-07-05:
Ich meinte jetzt die Modbusdaten aus dem bestehenden Log.

andig, 2022-07-05:
> Kann man aus den Modbus send / recv codes herauslesen was da passiert?

Nochmal: in dem Log ist alles ok weil Du soweit ich das sehe bewusst den Modus on/off änderst. Was soll ich da raus lesen? Alles wie erwartet. Also entweder machen wir hier zu oder Du produzierst bitte nochmal das angefragte Logfile vom Fehlerzustand und dem Weg dahin.

premultiply, 2022-07-05:
https://docs.evcc.io/docs/guides/setup

andig, 2022-07-05:
In https://github.com/evcc-io/evcc/issues/3738#issuecomment-1173965666 hast du noch ein Logfile, sogar trace mit allen Details. Gleiche Frage: ist das ein Logfile vom Fehlerfall?

Warum frage ich: da sieht alles normal aus. Ich versuche es einzukreisen…

Stefhorn, 2022-07-05:
Beide Logs sind vom Fehlerfall.
Im 1. Log habe ich auch eingetragen wo der Fehler passiert.
Im 2. Log hab ich es nicht markiert, der Fehler passiert aber nach jedem "set charge mode: off". Ja hier sieht alles normal aus, nicht mal ein "Fehler charger logic error: disabled but charging ".  
Dennoch kein Ausschalten möglich.

andig, 2022-07-05:
Dann bitte nochmal zu Deinem 2. Log. Du schaltest aus:

    [lp-2  ] DEBUG 2022/07/04 17:51:35 set charge mode: off
    [lp-2  ] DEBUG 2022/07/04 17:51:35 pv timer elapse
    [lp-2  ] DEBUG 2022/07/04 17:51:35 pv timer inactive
    [site  ] DEBUG 2022/07/04 17:51:35 ----
    [lp-2  ] DEBUG 2022/07/04 17:51:35 charge power: 3680W
    [site  ] DEBUG 2022/07/04 17:51:35 pv power: 7042W
    [site  ] DEBUG 2022/07/04 17:51:35 grid power: -4340W
    [site  ] DEBUG 2022/07/04 17:51:35 grid currents: [4.26 10.1 6.97]A
    [site  ] DEBUG 2022/07/04 17:51:35 site power: -4340W
    [pracht] TRACE 2022/07/04 17:51:35 modbus: send 01 04 00 6b 00 01 40 16
    [pracht] TRACE 2022/07/04 17:51:35 modbus: recv 01 04 02 00 02 38 f1
    [lp-2  ] DEBUG 2022/07/04 17:51:35 charger status: C
    [pracht] TRACE 2022/07/04 17:51:35 modbus: send 01 04 00 04 00 01 70 0b
    [pracht] TRACE 2022/07/04 17:51:35 modbus: recv 01 04 02 00 10 b8 fc
    [pracht] TRACE 2022/07/04 17:51:35 modbus: send 01 06 00 04 00 00 c8 0b
    [pracht] TRACE 2022/07/04 17:51:36 modbus: recv 01 06 00 04 00 00 c8 0b
    [lp-2  ] DEBUG 2022/07/04 17:51:36 charger disable
    [lp-2  ] DEBUG 2022/07/04 17:51:36 wake-up timer: stop
    [site  ] DEBUG 2022/07/04 17:51:40 ----

Die Ladeleistung geht auf 0, es **wird** also tatsächlich abgeschaltet:

    [lp-2  ] DEBUG 2022/07/04 17:51:40 charge power: 0W
    [site  ] DEBUG 2022/07/04 17:51:40 pv power: 7040W
    [site  ] DEBUG 2022/07/04 17:51:40 grid power: -6170W
    [site  ] DEBUG 2022/07/04 17:51:40 grid currents: [9.95 10.1 6.88]A
    [site  ] DEBUG 2022/07/04 17:51:40 site power: -6170W
    [pracht] TRACE 2022/07/04 17:51:40 modbus: send 01 04 00 6a 00 01 11 d6
    [pracht] TRACE 2022/07/04 17:51:40 modbus: recv 01 04 02 00 00 b9 30
    [pracht] TRACE 2022/07/04 17:51:40 modbus: send 01 04 00 03 00 01 c1 ca
    [pracht] TRACE 2022/07/04 17:51:41 modbus: recv 01 04 02 00 00 b9 30
    [site  ] DEBUG 2022/07/04 17:51:50 ----
    [lp-2  ] DEBUG 2022/07/04 17:51:50 charge power: 0W
    [site  ] DEBUG 2022/07/04 17:51:50 pv power: 7019W
    [site  ] DEBUG 2022/07/04 17:51:50 grid power: -6141W
    [site  ] DEBUG 2022/07/04 17:51:50 grid currents: [9.86 10.1 6.87]A
    [site  ] DEBUG 2022/07/04 17:51:50 site power: -6141W
    [pracht] TRACE 2022/07/04 17:51:50 modbus: send 01 04 00 6b 00 01 40 16
    [pracht] TRACE 2022/07/04 17:51:51 modbus: recv 01 04 02 00 01 78 f0

Wir erkennen dass die WB auf Status "B", vebunden, geht:

    [lp-2  ] DEBUG 2022/07/04 17:51:51 charger status: B
    [lp-2  ] INFO 2022/07/04 17:51:51 stop charging <-
    [pracht] TRACE 2022/07/04 17:51:51 modbus: send 01 04 00 04 00 01 70 0b
    [pracht] TRACE 2022/07/04 17:51:51 modbus: recv 01 04 02 00 00 b9 30
    [site  ] DEBUG 2022/07/04 17:52:00 ----
    [lp-2  ] DEBUG 2022/07/04 17:52:00 charge power: 0W
    [site  ] DEBUG 2022/07/04 17:52:00 pv power: 7019W
    [site  ] DEBUG 2022/07/04 17:52:00 grid power: -6134W
    [site  ] DEBUG 2022/07/04 17:52:00 grid currents: [9.88 10.1 6.8]A
    [site  ] DEBUG 2022/07/04 17:52:00 site power: -6134W
    [pracht] TRACE 2022/07/04 17:52:00 modbus: send 01 04 00 6a 00 01 11 d6
    [pracht] TRACE 2022/07/04 17:52:01 modbus: recv 01 04 02 00 00 b9 30
    [pracht] TRACE 2022/07/04 17:52:01 modbus: send 01 04 00 03 00 01 c1 ca
    [pracht] TRACE 2022/07/04 17:52:01 modbus: recv 01 04 02 00 00 b9 30
    [site  ] DEBUG 2022/07/04 17:52:10 ----
    [lp-2  ] DEBUG 2022/07/04 17:52:10 charge power: 0W
    [site  ] DEBUG 2022/07/04 17:52:10 pv power: 7020W
    [site  ] DEBUG 2022/07/04 17:52:10 grid power: -6132W
    [site  ] DEBUG 2022/07/04 17:52:10 grid currents: [9.87 10.1 6.82]A
    [site  ] DEBUG 2022/07/04 17:52:10 site power: -6132W
    [pracht] TRACE 2022/07/04 17:52:10 modbus: send 01 04 00 6b 00 01 40 16
    [pracht] TRACE 2022/07/04 17:52:10 modbus: recv 01 04 02 00 01 78 f0
    [lp-2  ] DEBUG 2022/07/04 17:52:10 charger status: B
    [pracht] TRACE 2022/07/04 17:52:10 modbus: send 01 04 00 04 00 01 70 0b
    [pracht] TRACE 2022/07/04 17:52:11 modbus: recv 01 04 02 00 00 b9 30
    [site  ] DEBUG 2022/07/04 17:52:20 ----

Es bleibt "aus":

    [lp-2  ] DEBUG 2022/07/04 17:52:20 charge power: 0W
    [site  ] DEBUG 2022/07/04 17:52:20 pv power: 7005W
    [site  ] DEBUG 2022/07/04 17:52:20 grid power: -6117W
    [site  ] DEBUG 2022/07/04 17:52:20 grid currents: [9.83 10.1 6.83]A
    [site  ] DEBUG 2022/07/04 17:52:20 site power: -6117W
    [pracht] TRACE 2022/07/04 17:52:20 modbus: send 01 04 00 6a 00 01 11 d6
    [pracht] TRACE 2022/07/04 17:52:20 modbus: recv 01 04 02 00 00 b9 30
    [pracht] TRACE 2022/07/04 17:52:20 modbus: send 01 04 00 03 00 01 c1 ca
    [pracht] TRACE 2022/07/04 17:52:21 modbus: recv 01 04 02 00 00 b9 30

...und genau hier schaltest Du wieder ein:

    [lp-2  ] DEBUG 2022/07/04 17:52:24 set charge mode: now

Deiner Ansicht nach ist das ein Fehler, wir wird aber absichtlich auch "sofort laden" geschaltet. Wo kommt das her? Ist das der Fehler???

premultiply, 2022-07-05:
:point_up: Da die Prachtbox keine Leistungsmessung hat kommt `charge power` intern von evcc und muss daher nicht mit der Realität übereinstimmen. :point_up:

andig, 2022-07-05:
Schau mal https://github.com/evcc-io/evcc/discussions/3192#discussioncomment-3085986. Wenn Du mich fragst ist das ein Fehler der Firmware und der nächste Schritt eine Supportanfrage beim Hersteller.

Stefhorn, 2022-07-05:
https://github.com/evcc-io/evcc/discussions/3192#discussioncomment-3085986  hört sich ganz nach meinem Problem an.
Wenn die Wallbox sich richtig über den Gesamtstrom abzuschalten lässt, wäre dass für mich in Ordnung da ich nur ein EV habe. Kann ich das per Config-datei umsetzen oder bedarf es eine Änderung des Codes?
Ich habe trotzdem schonmal eine Supportanfrage an die Firma Pracht geschickt. 

Btw: Bei mir läuft LP-1 3-phasig und LP-2 1-phasig So kann ich je nach PV-Überschuss den besten LP wählen.

premultiply, 2022-07-05:
Das ist nicht so einfach, da die beiden LPs sich dann gegenseitig den Status beeinflussen würden.

andig, 2022-07-06:
Wir könnten das kurzfristig als Workaround umsetzen. Wirklich schön wäre aber Feedback vom Hersteller.

@Stefhorn magst Du Dich einfach mal an den Support wenden?

andig, 2022-07-06:
Kannst du selbst compilieren? Dann könntest du in https://github.com/evcc-io/evcc/blob/45cbabada985a4b86e0c034494e3deab49a2dda6/charger/prachtalpha.go#L39 mal 40004 durch 40003 ersetzen, das würde zumindest beim ersten Auto funktionieren.

Stefhorn, 2022-07-07:
Hab ich noch nie gemacht aber würde es mal probieren. Hat jemand ein Tip / Anleitung für Anfänger?

andig, 2022-07-07:
Meld Dich mal bei info@evcc.io, dann bekommst Du compilierte Version. Raspi?

DariusM73, 2022-07-07:
> Ok, ich melde mich wieder sobald ich eine Antwort von Pracht erhalten habe.

Da bin ich echt gespannt - habe schon x Anfragen an Pracht gesendet. Die Resonanz war exakt 0. Habe nämlich noch ein weiteres Problem mit der Pracht. Das Modbus Interface schießt sich sporadisch ab, dann muss die Modbus Platine (oder komplette Box) von der Spannung getrennt werden - ist nervig. Konnte es noch nicht genau eingrenzen, aber es passiert, wenn ein Fahrzeug lädt und die Register beschrieben werden, m.E. gibt es hier irgendein Timing Problem. Das sollte beim Überschussladen funktionieren...

Stefhorn, 2022-07-07:
> 

Hab die "/usr/bin/evcc" gegen die abgeänderte Datei getauscht, Owner und Group permission auf root, Rechte auf 755 geändert.

evcc startet nicht mehr

pi@raspberrypi:~ $ sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2022-07-07 20:13:15 CEST; 2s ago
    Process: 1370 ExecStart=/usr/bin/evcc (code=exited, status=1/FAILURE)
   Main PID: 1370 (code=exited, status=1/FAILURE)
        CPU: 118ms

Jul 07 20:13:16 raspberrypi evcc[1363]: While parsing config: yaml: line 1: did not find expected key
Jul 07 20:13:10 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Jul 07 20:13:16 raspberrypi evcc[1370]: While parsing config: yaml: line 1: did not find expected key
Jul 07 20:13:10 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Jul 07 20:13:15 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 7.
Jul 07 20:13:15 raspberrypi systemd[1]: Stopped evcc.
Jul 07 20:13:15 raspberrypi systemd[1]: Started evcc.
Jul 07 20:13:15 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Jul 07 20:13:15 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
pi@raspberrypi:~ $

andig, 2022-07-10:
@Stefhorn klappt das? Wenn nicht bitte mal Service stoppen und dann das nackte evcc von Hand starten.

andig, 2022-07-11:
Mach doch mal bitte langsam. Hast du meinen Vorschlag befolgt? Ergebnis? Ausgabe?

Stefhorn, 2022-07-11:
Nachdem ich die alte evcc Datei gegen die neue getauscht hatte war Laden nicht mehr möglich.
So bin ich vorgegangen:
sudo mv /usr/bin evcc  /usr/bin evcc.old #alte evcc umbenannt
sudo mv evcc /usr/bin # neue evcc eingefügt
sudo chown root:root /usr/bin evcc #Owner geändert
Rechte auf 755 gesetzt

Danach ist im Stop Modus das Problem des Aus-Einschalten nicht mehr aufgetreten.
Aber Laden war weder im Sofort, PV oder Min+PV möglich, statt dessen kontinuierliches anziehen und abfallen des Schützes bis Wallbox auf Störung ging.

Nackte evcc von Hand starten brachte auch kein Erfolg.
Hab leider kein Trace log, kann ich aber bei Bedarf morgen noch nachholen.

andig, 2022-07-11:
Das klingt als würde evcc laufen und wir hätten das Problem von aus auf ein verlagert. Ich schick dir nochmal eine neue Version und probiere mal beide Register zu setzen.

Stefhorn, 2022-07-12:
Hier noch ein Trace Log von gerade eben. Fahrzeug hängt an LP-2, (1-Phase) jede Menge Überschuss vorhanden aber keine Ladung

Jul 12 08:37:39 raspberrypi evcc[2678]: [lp-1  ] DEBUG 2022/07/12 08:37:39 charge power: 0W
Jul 12 08:37:39 raspberrypi evcc[2678]: [lp-2  ] DEBUG 2022/07/12 08:37:39 charge power: 0W
Jul 12 08:37:39 raspberrypi evcc[2678]: [site  ] DEBUG 2022/07/12 08:37:39 pv power: 5806W
Jul 12 08:37:39 raspberrypi evcc[2678]: [site  ] DEBUG 2022/07/12 08:37:39 grid power: -5582W
Jul 12 08:37:39 raspberrypi evcc[2678]: [site  ] DEBUG 2022/07/12 08:37:39 grid currents: [8.12 8.01 8.01]A
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-1/chargePower: 0
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 pvPower: 5806
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 gridPower: -5581.700000000001
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/chargePower: 0
Jul 12 08:37:39 raspberrypi evcc[2678]: [site  ] DEBUG 2022/07/12 08:37:39 site power: -5582W
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 gridCurrents: [8.117 8.009 8.01]
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 gridEnergy: 46.6243
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/mode: pv
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/chargeCurrent: 16
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/chargedEnergy: 0
Jul 12 08:37:39 raspberrypi evcc[2678]: [pracht] TRACE 2022/07/12 08:37:39 modbus: send 01 04 00 6b 00 01 40 16
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/chargeDuration: 0s
Jul 12 08:37:39 raspberrypi evcc[2678]: [pracht] TRACE 2022/07/12 08:37:39 modbus: recv 01 04 02 00 01 78 f0
Jul 12 08:37:39 raspberrypi evcc[2678]: [lp-2  ] DEBUG 2022/07/12 08:37:39 charger status: B
Jul 12 08:37:39 raspberrypi evcc[2678]: [pracht] TRACE 2022/07/12 08:37:39 modbus: send 01 04 00 04 00 01 70 0b
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/connected: true
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/charging: false
Jul 12 08:37:39 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:39 lp-2/enabled: true
Jul 12 08:37:40 raspberrypi evcc[2678]: [pracht] TRACE 2022/07/12 08:37:40 modbus: recv 01 04 02 00 10 b8 fc
Jul 12 08:37:40 raspberrypi evcc[2678]: [lp-2  ] DEBUG 2022/07/12 08:37:40 pv charge current: 24.3A = 0A + 24.3A (-5582W @ 1p)
Jul 12 08:37:40 raspberrypi evcc[2678]: [cache ] TRACE 2022/07/12 08:37:40 homePower: 224.29999999999927

andig, 2022-07-12:
@Stefhorn meldest Du Dich bitte nochmal bei info@evcc.io

andig, 2022-07-16:
@Stefhorn mittlerweile weiß ich ja, dass Du 2 Ladepunkte hast. Das Problem hier hattest Du trotzdem auch wenn nur ein Ladepunkt konfiguriert war so wie oben beschrieben?

closed 2022-07-17 by andig