evcc-io/evcc #3745: Start/Restart von evcc startet sofort den Ladevorgang mit maximal möglicher Leistung

Alex-ander-s, 2022-06-30:
### Describe the bug

Bei einem restart von evcc wird SOFORT der Ladevorgang mit maximal möglicher Leistung gestartet und nach wenigen Sekunden wegen Überschreitung des "disable threshold" wieder beendet.
Das Verhalten treibt leider die Ladezyklen unnötig in die Höhe und führt auch dazu, dass der PV-Überschuss bis zum Ablauf des enable delay, bzw. der guardduration Zeitspanne nicht mehr genutzt wird.

Ein ähnliches Verhalten, aber mit anderer Ursache wurde bereits in #568 festgestellt.

### Steps to reproduce

evcc installieren
config evcc.yaml hinterlegen (siehe unten)
einen Überschuss von ca. 1kW erzeugen/simulieren
evcc starten

### Configuration details

network:
  schema: http
  host: home.net
  port: 7070
interval: 15s # control cycle interval

# log settings
log: debug
levels:
  core: debug
  lp-1: debug
  site-1: debug

meters:
  - name: energyMeter
    type: custom
    power: # power (W)
      source: mqtt
      topic: energyMeter/instantPower
      timeout: 30s
    energy: # optional energy (kWh)
      source: mqtt
      topic: energyMeter/activeEnergyP
      timeout: 30s
    currents:
      - source: mqtt
        topic: energyMeter/currentL1
        timeout: 30s
      - source: mqtt
        topic: energyMeter/currentL2
        timeout: 30s
      - source: mqtt
        topic: energyMeter/currentL3
        timeout: 30s

chargers:
- name: go-e-1
  type: template
  template: go-e
  host: 192.168.124.136

vehicles:
- name: Ellie
  type: template
  template: carwings
  title: Ellie
  user: xxxxx@xxx.xx
  password: XXXXX
  capacity: 36

site:
  title: Home
  meters:
    grid: energyMeter

loadpoints:
- title: go-e-1
  charger: go-e-1
  vehicle: Ellie
  mode: off # charge mode (off, now, minpv, pv)
  resetOnDisconnect: true
  soc:
    poll:
      mode: always
      interval: 15m
    min: 40 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 90 # always charge to 90%
    estimate: false # set true to interpolate between api updates
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: -300 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 1380 # maximum import power (W) - minimum efficient charge power is 1380W @6amps
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 18 # maximum charge current (default 16A)

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.12 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0475 # EUR/kWh

mqtt:
  broker: localhost:1883
  topic: evcc
  user: evcc
  password: XXXXX

influx:
  url: http://localhost:8086
  database: evcc
  token: "XXXXX"
  org: "Home"

### Log details

[main  ] INFO 2022/06/30 11:22:56 evcc 0.95
[main  ] INFO 2022/06/30 11:22:56 using config file /opt/evcc/conf/evcc.yaml
[main  ] INFO 2022/06/30 11:22:56 listening at :7070
[mqtt  ] INFO 2022/06/30 11:22:56 connecting evcc-452345984 at tcp://localhost:1883
[mqtt  ] DEBUG 2022/06/30 11:22:56 tcp://localhost:1883 connected
[lp-1  ] WARN 2022/06/30 11:23:08 poll mode '{always 15m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2022/06/30 11:23:08 poll interval '15m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2022/06/30 11:23:08 site config:
[site  ] INFO 2022/06/30 11:23:08   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2022/06/30 11:23:08     grid:      power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/06/30 11:23:08 loadpoint 1:
[lp-1  ] INFO 2022/06/30 11:23:08   mode:        pv
[lp-1  ] INFO 2022/06/30 11:23:08   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/06/30 11:23:08   meters:      charge ✓
[lp-1  ] INFO 2022/06/30 11:23:08     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/06/30 11:23:08   vehicles:    ✓
[lp-1  ] INFO 2022/06/30 11:23:08     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2022/06/30 11:23:08 vehicle updated: unknown -> Ellie
[site  ] DEBUG 2022/06/30 11:23:08 ----
[lp-1  ] DEBUG 2022/06/30 11:23:08 charge power: 0W
[site  ] DEBUG 2022/06/30 11:23:08 grid power: -966W
[site  ] DEBUG 2022/06/30 11:23:08 grid currents: [5.99 0.38 2.12]A
[site  ] DEBUG 2022/06/30 11:23:08 site power: -966W
[lp-1  ] DEBUG 2022/06/30 11:23:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/06/30 11:23:08 charger status: B
[lp-1  ] DEBUG 2022/06/30 11:23:11 max charge current: 18A
[lp-1  ] DEBUG 2022/06/30 11:23:11 charger enable
[lp-1  ] DEBUG 2022/06/30 11:23:11 wake-up timer: start
[lp-1  ] DEBUG 2022/06/30 11:23:11 pv timer elapse
[lp-1  ] DEBUG 2022/06/30 11:23:11 pv timer inactive
[site  ] DEBUG 2022/06/30 11:23:23 ----
[lp-1  ] DEBUG 2022/06/30 11:23:24 charge power: 4090W
[site  ] DEBUG 2022/06/30 11:23:24 grid power: 3278W
[site  ] DEBUG 2022/06/30 11:23:24 grid currents: [11.9 0.46 2.13]A
[site  ] DEBUG 2022/06/30 11:23:24 site power: 3278W
[lp-1  ] DEBUG 2022/06/30 11:23:24 charge currents: [17.9 0 0]A
[lp-1  ] DEBUG 2022/06/30 11:23:25 charger status: C
[lp-1  ] INFO 2022/06/30 11:23:25 start charging ->
[lp-1  ] DEBUG 2022/06/30 11:23:26 pv timer elapse
[lp-1  ] DEBUG 2022/06/30 11:23:26 pv timer inactive
[site  ] DEBUG 2022/06/30 11:23:38 ----
[lp-1  ] DEBUG 2022/06/30 11:23:38 charge power: 4130W
[site  ] DEBUG 2022/06/30 11:23:38 grid power: 3294W
[site  ] DEBUG 2022/06/30 11:23:38 grid currents: [12 0.4 2.11]A
[site  ] DEBUG 2022/06/30 11:23:38 site power: 3294W
[lp-1  ] DEBUG 2022/06/30 11:23:38 charge currents: [18 0 0]A
[lp-1  ] DEBUG 2022/06/30 11:23:38 detected phases: 1p
[lp-1  ] DEBUG 2022/06/30 11:23:38 charger status: C
[lp-1  ] DEBUG 2022/06/30 11:23:41 vehicle soc: 89%
[lp-1  ] DEBUG 2022/06/30 11:23:41 vehicle range: 228km
[lp-1  ] DEBUG 2022/06/30 11:23:41 pv charge current: 3.68A = 18A + -14.3A (3294W @ 1p)
[lp-1  ] ERROR 2022/06/30 11:23:43 climater: climate status not available
[lp-1  ] DEBUG 2022/06/30 11:23:43 site power 3294W >= 1380W disable threshold
[lp-1  ] DEBUG 2022/06/30 11:23:43 pv disable in 0s
[lp-1  ] DEBUG 2022/06/30 11:23:43 pv disable timer elapsed
[lp-1  ] ERROR 2022/06/30 11:23:43 climater: climate status not available
[lp-1  ] DEBUG 2022/06/30 11:23:43 charger disable
[lp-1  ] DEBUG 2022/06/30 11:23:43 wake-up timer: stop
[site  ] DEBUG 2022/06/30 11:23:53 ----
[lp-1  ] DEBUG 2022/06/30 11:23:53 charge power: 0W
[site  ] DEBUG 2022/06/30 11:23:53 grid power: -934W
[site  ] DEBUG 2022/06/30 11:23:53 grid currents: [5.87 0.41 2.11]A
[site  ] DEBUG 2022/06/30 11:23:53 site power: -934W
[lp-1  ] DEBUG 2022/06/30 11:23:53 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/06/30 11:23:53 charger status: B
[lp-1  ] INFO 2022/06/30 11:23:53 stop charging <-
[lp-1  ] DEBUG 2022/06/30 11:23:53 vehicle soc: 89%
[lp-1  ] DEBUG 2022/06/30 11:23:53 vehicle range: 228km
[lp-1  ] DEBUG 2022/06/30 11:23:53 pv charge current: 4.06A = 0A + 4.06A (-934W @ 1p)
[lp-1  ] ERROR 2022/06/30 11:23:53 climater: climate status not available
[lp-1  ] DEBUG 2022/06/30 11:23:53 site power -934W <= -300W enable threshold
[lp-1  ] DEBUG 2022/06/30 11:23:53 pv enable in 0s
[lp-1  ] DEBUG 2022/06/30 11:23:53 pv enable timer elapsed
[lp-1  ] DEBUG 2022/06/30 11:23:53 max charge current: 6A
[lp-1  ] DEBUG 2022/06/30 11:23:53 charger enable: contactor delay 9m49s

### What type of operating system are you running?

Linux

### Version

evcc version 0.95

andig, 2022-06-30:
Interessant. Du hast `mode: off`, das Fahrzeug wird erkannt, es ist genug Leistung vorhanden und die Ladung startet. Das sollte so vmtl. nicht sein. Könntest Du bitte noch ein `log --trace` machen um zu verstehen was da genau passiert?

Alex-ander-s, 2022-07-01:
Ich hatte es bis jetzt immer mal wieder nach einem restart von evcc nach dem Einspielen einer neuen Version.
Also ich würde sagen schon länger als ein Monat.
Aktuell kann ich es leider wieder nicht nachstellen (Sonne scheint aber gerade nicht genug).
Ich versuche es wieder im "echten Szenario", wenn ich den Akku wieder ein wenig geleert habe und die Sonne mitspielt...

andig, 2022-07-04:
Ich kann das Problem nicht nachvollziehen. Vmtl. liegt es an

     min: 40 # 

wenn Du noch keinen `soc` hast (temporäres Problem etc). Sieht aus wie works as designed.

closed 2022-07-07 by andig