evcc-io/evcc #3759: evcc treats second loadpoint as 3 phase even though it is configured as 1 phase (and physically is)

Caibuk, 2022-07-03:
### Describe the bug

I have a installation with two Wallbe eco 2.0 wallboxes. Loadpoint 1 is physically 3ph up to the socket (Type-2 cable is 1ph) with a 3ph SDM630 meter and configured as 1ph. Loadpoint 2 is physically 1ph with SDM120 meter.

Since the introduction of new phases handling a couple of nightlies ago, my second loadpoint is treated as 3ph altough it is physically 1ph and - in my mind - correctly configured as such.

Due to this treatment charging won't start below (in fact above) roughly -4.4 kW of site power.

When I start a charging session manually using min+pv evcc continues charging quite well and follows the available solar power. Another workaround is setting the loadpoint to 1ph using API.

Loadpoint 1 does not have this problem and is treated as 1ph right from the start!

### Steps to reproduce

See "describe the bug" accordingly.

### Configuration details

```# uri: 0.0.0.0:7070 # uri for ui

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS - evcc.local
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 0.0.0.0
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: thisiswheremysponsortokenbelongsto

# log settings
log: trace
levels:
  #core: trace
  #lp-1: trace
  #site: trace
  #lp-2: trace  
  #cache: trace
  #vw: trace
  #id: debug
  #enyaq: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
  - name: grid
    type: modbus
    model: kostal
    uri: 10.0.1.57:502
    id: 255
    power: Power
    energy: Export
  - name: pv-dc-kostal
    type: custom
    power:
      source: calc
      add: # The add plugin sums up both string values
        - source: modbus
          model: kostal
          value: 160:1:DCW # string 1
          uri: 10.0.1.37:1502
          id: 71 # Configured Modbus Device ID
[... 202 lines cut ...]
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>

```

### Log details

log contains only entries for site and lp-2 as the other may be irrelevant and confusing
- restarted evcc with car connected
- there is plenty of grid power to start 1ph charging (but as evcc thinks lp-2 is 3ph it won't start)
- started charging manually by setting charge mode to minpv at 09:17:53
- evcc starts slowly ramping up the charge power (as current is calculated for 3ph)

[_evcc-nightly_logs (3).txt](https://github.com/evcc-io/evcc/files/9034287/_evcc-nightly_logs.3.txt)

### What type of operating system are you running?

Docker container

### Version

evcc 0.95 (887394fc)

andig, 2022-07-03:
Da hatte ich einen Knoten im Kopf. Könntest Du den Fix kurz bestätigen?

Caibuk, 2022-07-03:
Mach ich. Muss erst einmal mein go auf einen aktuellen Stand bringen. Melde mich dann.

Danke für die schnelle Reaktion!

Caibuk, 2022-07-03:
Passt! Das Phänomen ist weg:
```
[site  ] DEBUG 2022/07/03 11:45:49 ----
[lp-2  ] DEBUG 2022/07/03 11:45:49 charge power: 0W
[site  ] DEBUG 2022/07/03 11:45:50 pv power: 6362W
[site  ] DEBUG 2022/07/03 11:45:50 battery power: -4153W
[site  ] DEBUG 2022/07/03 11:45:50 grid power: -210W
[site  ] DEBUG 2022/07/03 11:45:50 battery soc 0: 61%
[site  ] DEBUG 2022/07/03 11:45:50 site power: -4363W
[lp-2  ] DEBUG 2022/07/03 11:45:50 charger status: B
[lp-2  ] DEBUG 2022/07/03 11:45:50 next soc poll remaining time: 59s
[lp-2  ] DEBUG 2022/07/03 11:45:50 pv charge current: 19A = 0A + 19A (-4363W @ 1p)
[lp-2  ] DEBUG 2022/07/03 11:45:50 climater active: false, target temp: 22.0°C, outside temp: 31.0°C
[lp-2  ] DEBUG 2022/07/03 11:45:50 site power -4363W <= 0W enable threshold
[lp-2  ] DEBUG 2022/07/03 11:45:50 pv enable timer start: 1m0s
[lp-2  ] DEBUG 2022/07/03 11:45:50 pv enable in 1m0s
```

Vielen Dank für den schnellen Fix!

closed 2022-07-03 by andig