evcc-io/evcc #3765: non-anonymous MQTT client for Tinkerforge Warp wallbox

mocaadmin, 2022-07-03:
I have a Warp Pro wallbox installed which works perfectly with the described Mosquitto broker on EVCC. What I want to avoid is to send data through a network without TLS encryption and without a concise user who is subscribing the data. On the wallbox side you can define a user and password which works. TLS seems not to be supported. On EVCC side I cannot configure a user/pw for the wallbox config. At least I did not find it. There seems to be a user/pw support when EVCC itself publishes its data.

I would like to have TLS functionality on EVCC side when connecting to a broker. Second thing I would like to configure user/pw when subscribing to the warp-wallbox data on the broker.

I have checked the github files but did not find any hint on TLS connection and user/pw in the config files.

andig, 2022-07-03:
You can use user/password

mocaadmin, 2022-07-03:
Hi thanks for the fast answer. I tried this and found out that it only works when there are no special characters in the password.

andig, 2022-07-04:
> when there are no special characters in the password.

If there are and they don't fit yaml encoding (most do!) you'll need to quote it.

> Any news on the TLS connection?

It should be possible to prefix the the broker with `tls://` to do this, though I didn't try.