evcc-io/evcc #3769: Nissan Leaf (pre 2019): climate status not available

Kubiac, 2022-07-04:
### Describe the bug

I configured my Leaf 40kWh via evcc configure. When the charge is queried I get the error message "climater: climate status not available" in the evcc UI.

### Steps to reproduce

Everytime the SoC is requested, I get the error message.

### Configuration details

vehicles:
- type: template
  template: carwings 
  title: Elli  
  user: ***  
  password: ***  
  vin: ***  
  capacity: 35  
  phases: 1  
  minCurrent: 6  
  maxCurrent: 32  
  cloud: false  
  name: ev2
  minSoC: 20
  targetSoC: 90
  mode: PV

### Log details

Jul 04 09:10:04 raspberrypi evcc[27080]: [lp-1  ] ERROR 2022/07/04 09:10:04 climater: climate status not available

### What type of operating system are you running?

Linux

### Version

0.95

Kubiac, 2022-07-05:
Meine Erwartung: Keine Fehlermeldung zu bekommen, wenn alles wie erwartet (wenn auch mit reduziertem Feature-Set) läuft. Versetze ich mich in einen unbedarften Nutzer, dann sieht der ein rotes Ausrufezeichen in seiner UI und denkt, dass Handlungsbedarf besteht (vgl. rote Lampe im Auto). In diesem Fall könnte ich mir eine Warnung beim initialisieren des vehicle vorstellen: sinngemäß "Achtung, erwarte nicht, dass dieses Feature funktioniert". Es ist ja auch nichts was sich einfach so ändern könnte und es deswegen Sinn macht immer wieder zu kommunizieren.

andig, 2022-07-05:
> Meine Erwartung: Keine Fehlermeldung zu bekommen, wenn alles wie erwartet (wenn auch mit reduziertem Feature-Set) läuft. 

Es läuft ja nicht wie erwartet. Konkret wird es nicht möglich sein, die WB bei Vorklimatisierung freizuschalten. Wir können aber den Fehler gerne unterdrücken?

Kubiac, 2022-07-05:
Gegenfrage: geht das Feature bei keinem Leaf oder ist das jetzt ein Problem, das nur manche Nutzer haben?

andig, 2022-07-05:
Da fragst Du den Falschen- ich hab ja keinen Leaf. Dein Ticket ist jedenfalls das Einzige dazu. Ich hab keine Zeit für eine Grundsatzdiskussion, aber die Optionen liegen ja auf dem Tisch:

> Wir können aber den Fehler gerne unterdrücken?

Was machen wir jetzt damit?

Kubiac, 2022-07-05:
Ich schließe nicht gerne von mir auf die Allgemeinheit und habe vor allem das carwings-Modul nicht geschrieben. Aber hier ist was ich jetzt machen würde:
Beim Hochfahren loggen, dass Klimaanlage nicht unterstützt ist. Ich vermute mal in dieser Logzeile:
`Jun 27 22:21:09 raspberrypi evcc[660]: [lp-1  ] INFO 2022/06/27 22:21:09     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗`
Und dann keine Meldung mehr im Betreib ausgeben.

Ich versuche die Funktionalität wieder herzustellen und würde die Änderungen, dann in einem PR einreichen. Muss dieses ganze GO aber erstmal einrichten und kennenlernen.

Kubiac, 2022-07-05:
So langsam verstehe ich erst, welchen input du von mir haben möchtest.

> ich hab ja keinen Leaf. Dein Ticket ist jedenfalls das Einzige dazu. ... Was machen wir jetzt damit?

Ich hab dich so verstanden, dass ich im Grunde der einzige Nutzer bin und das von meiner Erfahrung aus verallgemeinern soll.

Habe in der carwings-lib folgenden Kommentar über dem Fehler gefunden:
```
// Sometimes the RemoteACRecords field is an empty array
// instead of a struct value.  This API... ¯\_(ツ)_/¯
```
Also anscheinend kein Bug in evcc oder der library. Mein Vorschlag: Die Meldung von ERROR auf WARNING runterstufen, weil es für die Grundfunktionen nicht kritisch ist. Wenn du es anders siehst, lass es einfach wie es ist.

closed 2022-07-10 by andig