evcc-io/evcc #3810: Neuer Konfigurationsparameter minimalSoC

markus-wetzel, 2022-07-11:
**Is your feature request related to a problem? Please describe.**
Problem ist hier bereits diskutiert:
https://github.com/evcc-io/evcc/discussions/3796

**Describe the solution you'd like**
Neuer Konfigurationsparameter:
minimalSoC: 35 # Im PV-Modus wird der Ladevorgang unterbrochen, wenn Hausbatterie eine SoC von kleiner gleich 35% hat. Hausbatterie bekommt anschließend wieder Ladevorrang bis zum Wert prioritySoC.

Passt thematisch zu https://github.com/evcc-io/evcc/discussions/3530 ist aber unabhängig davon ein weiterer Parameter für eine wie auch immer ganannte Funktion.
Für den Modus PV sollte die Abschlatung immer greifen. Für den "Smart" oder "Min+PV" Modus oder wie der auch immer zukünftig genannt werden wird (#3530), wäre das dann Konfigurierbar? Für mein Setup wäre es in erster Linie wichtig für die PV-Funktion.

**Describe alternatives you've considered**
Der Workaround aus 
https://github.com/evcc-io/evcc/discussions/3796
funktioniert zwar, ist aber keine "schöne" Dauerlösung, da die Umstellung in meiner Konfiguration immer heißt: Auto Parken - Ins Haus an den Zählerschrank - wieder raus ans Auto und Ladevorgang mit RFID-Karte starten.

premultiply, 2022-07-11:
`residualPower` auf 100 setzen und mit bufferSoc arbeiten?

markus-wetzel, 2022-07-11:
Ja, ist eine Möglichkeit, wenn man genügend Leistung vom Dach bekommt oder eine Wallbox hat, in der man die Phasen nicht am Zählerkasten abschalten muss.  Habe ich nur beides nicht (#3796). Der Workaround funktioniert ja auch grundsätzlich mal, ist aber aus meiner Sicht nicht Praxistauglich für meine Anlage und meine Nutzer.

Daher die Bitte sich zu überlegen ob der neue Konfigurationsparameter nicht eine sinnvolle Erweiterung wäre, der einem Teil der Nutzer etwas bringen würde - allen sicherlich nicht, dazu ist mein Setup viel zu speziell.

andig, 2022-07-11:
@markus-wetzel könntest Du statt der Lösung bitte beschreiben, welches Problem eigentlich adressiert werden soll?

markus-wetzel, 2022-07-11:
Das Problem ist https://github.com/evcc-io/evcc/discussions/3796 hier bereits beschrieben. Ich wollte das nicht nochmal wiederholen, wenn ich es aber hier nochmal posten soll, kann ich das gerne machen.

andig, 2022-07-11:
Ich versteh es nicht, hab ich dort kommentiert. Deshalb bitte klare Problembeschreibung. Eine immer tiefere Schachtelung sich gegenseitig übersteuernder Parameter ist jedenfalls kein Lösung.

markus-wetzel, 2022-07-11:
Problemstellung:
Anlage mit Nord-Westausrichtung und Mennekes Amtron Premium Wallbox, die nur 3-phasig laden kann (min. 4,1 kW).
Anlage liefert erst am Nachmittag genügend Power um über die 4,1 kW zu kommen - ich möchte aber bereits am Vormittag ab einer gewissen PV-Leistung (z.B. 2,5 kW) laden können, auch wenn der Rest aus der Hausbatterie kommt.
Allerdings soll das nur passieren, wenn die Hausbatterie einen SoC von mindestens 50% hat.
Das lässt sich mit prioritySoC: 50 einstellen und funktioiniert auch.

Sprich die Hausbatterie wird erst bis 50% geleaden und erst dann beginnt der Ladevorgang, der die Hausbatterie dann wieder entläd.
Allerdings wird sie dann ggf. bis 0% entladen, was ich gerne verhindern möchte.
Im PV-Modus sollte die Entladung der Hausbatterie zwar zulässig sein, aber nicht bis 0% sondern vielleicht bis 20% SoC.

andig, 2022-07-11:
Im PV Modus wird die Batterie nicht entladen. Falls doch ists ein Bug, siehe Kommentar in der Discussion. Da es ab jetzt doppelt wird gerne dort weiter.

closed 2022-07-11 by andig