evcc-io/evcc #3813: Wrong data type for MQTT

chadibe0, 2022-07-11:
The type of the datapoint evcc/loadpoints/1/mode/set is generated as "number" instead of "string".
Setting this datapoint to "off", "no" or "pv" leads into an error in EVCC. After changing the type to string it is working but everytime the MQTT connection is being restarted the data type is back to "number"

andig, 2022-07-13:
> type of the datapoint evcc/loadpoints/1/mode/set is generated as "number" instead of "string".

Mqtt is untyped. What do you mean?

> After changing the type to string

What do you mean by that?

chadibe0, 2022-07-13:
I don't know where the bug is really coming from. I have an MQTT client running on my ioBroker. As soon as the connection to EVCC is established I get all the datapoints in my ioBroker. But the datapoint for setting the mode is always of the wrong datatype. I changed the datatype in ioBroker to String and everything was fine. But afer reestablishing the connection the old datatype was back.

andig, 2022-07-13:
It's not clear that this has todo with evcc, closing.

closed 2022-07-13 by andig