evcc-io/evcc #3823: Alfen Wallbox not mA capable

schrotrf, 2022-07-12:
According to the template documentation for the Alfen Eve wallboxes and the fact that the ChargerEx interface is implemented, they are supposed to be capable of controlling charging currents in mA granularity:
https://github.com/evcc-io/evcc/blob/5c719fcb03f2cf776a08b9937f3b072509875d4b/templates/docs/charger/alfen_0.yaml#L4

Alfen's Modbus holding register 1210 is typed with Float32, however, according to their documentation its stepsize is 1A, i.e. there is de-facto no possibility for mA level current control.

I'm not sure if this has any negative effect, or if the wallbox controller truncates/rounds the values accordingly, but maybe it would be advisable to supply the maximum currents according to the docu.

andig, 2022-07-13:
Thank you. If thats the case we should update the interface implementation. Can anyone confirm this using the `ramp` command?

andig, 2022-07-13:
/cc @premultiply

premultiply, 2022-07-13:
IMHO its fine to set float values if the charger interface supports float type. If the charger would round to any PWM step internally this would be fine.

andig, 2022-07-13:
No, we should remove it or we‘ll have the same effect as when charging a Tesla. The non-linear behaviour does not fit our algorithm that evaluates delta power and converts it to delta current. Will provide a PR asap.

andig, 2022-07-13:
See
    
    evcc charger ramp --help

schrotrf, 2022-07-14:
Checking the actuals via the ramp command was rather helpful, it seems that the wallbox in fact does consider increments of less than 1A.

I ran the following command with a KIA PHEV connected to the Alfen box: 
`evcc charger ramp --delay 30s --digits 1`

These are the results:
[image]

So while there are some small plateaus, which I would attribute to the OBC it is a rather linear increase of the power over the 0.1A increments in current.

Sorry for having caused the confusion here, I'm closing the issue.

closed 2022-07-14 by schrotrf