evcc-io/evcc #3824: Add KSE wBX16 charger support

premultiply, 2022-07-13:
### Discussed in https://github.com/evcc-io/evcc/discussions/3820

<div type='discussions-op-text'>

<sup>Originally posted by **ChristianFleissner** July 12, 2022</sup>
Hallo zusammen,

ich habe bei mir eine KSE wBX16 Wallbox installiert. Diese ist über RS485 und einen entsprechenden USB-Adapter an einem Raspberry Pi angeschlossen, auf dem ich EVCC zum laufen bringen möchte.
Mein Alter ist leider schon etwas fortgeschritten so dass ich ein wenig vor der Wand stehe, wenn ich mit Hilfe der Dokumentation versuche, diese Wallbox anzulegen (wobei inzwischen einiges der allgemeinen Modbus-RTU-Dokumentation verschwunden ist).
Ich sollte eigentlich alle notwendigen Informationen inklusive der entsprechenden Register haben:

Allgemeine Bedienungsanleitung:
https://www.kse-wallbox.com/alle-downloads?file=files/images/Inhaltsbilder/Service/Alle%20Downloads/KSE%20Wallbox%20Bedienungsanleitung.pdf&cid=137

Registerset:
https://www.kse-wallbox.com/alle-downloads?file=files/images/Inhaltsbilder/Service/Alle%20Downloads/KSE%20Wallbox%20Registerset.pdf&cid=137

Anleitung für das KSE-eigene Lastmanagement (habe ich aus Platzgründen nicht genommen, da dieses nicht mit SMA Sunny Home Manager kommuniziert und ich nicht den Platz für zusätzliche Summenstromwandler im Verteilerkasten habe), da steht die Baudrate usw. drin:
https://www.kse-wallbox.com/alle-downloads?file=files/images/Inhaltsbilder/Service/Alle%20Downloads/Anleitung%20Lademanagement.pdf&cid=137

Was muss ich wo eintragen, damit das halbwegs funktionieren kann? Oder bin ich zu naiv und das wird nix?

Vielen Dank!</div>

premultiply, 2022-07-13:
@ChristianFleissner Könntest du mal #3828 testen?

ChristianFleissner, 2022-07-13:
Grundsätzlich gerne, nur wie mach ich das? Also wie bekomme ich die jetzt von dir (wirklich überaschend schnell, vielen Dank) gemachten Anpassungen in "mein" EVCC?

Noch ein paar Zusatzinformationen: 
- meine Wallbox hat keinen RFID-Reader, die Version kam leider für mich zu spät auf den Markt.
- Das Umschalten von 1- auf 3-phasig kann ich nicht testen, da mein Auto (Skoda Superb iV) nur 1phasig laden kann.

premultiply, 2022-07-14:
Ich merge es mal. Dann ist es morgen im Nightly Build.
Bisher nicht Vorhandenes kann ja nicht kaputt gehen.

ChristianFleissner, 2022-07-15:
Alles klar. Dann mach ich morgen früh ein Update und probier schau mal, ob ich es zum laufen kriege.

ChristianFleissner, 2022-07-16:
Entweder klappt es nicht oder ich bin zu doof. Mit ein wenig Google habe ich es geschafft, den nightly Build zu installieren und die KSE taucht dann auch in der Liste auf. Bei der Konfiguration bekomme ich dann aber immer einen Fehler:

 Choose the ModBus interface Serial (USB-RS485 Adapter)
 Modbus ID 100
 Device address /dev/ttyUSB0
 Baudrate 9600
 ComSet 8N1

Testing the configuration of KSE wBX16 ...
   Error: serial: timeout

Die Modbus ID müsste nach der Anleitung des KSE-Lademanagements zwischen 100 und 115 liegen (ich habe alle ausprobiert, auch wenn ich hoffe, dass mein Elektriker nicht an den Jumpern rumgespielt hat und die daher alle off sein sollten). Die Baudrate steht eindeutig drin, die müsste stimmen. Die device address müsste auch stimmen, wenn ich da was anderes eingebe kommt eine andere Fehlermeldung.
Beim Comset habe ich sowohl "8N1" als auch "8E1" ausprobiert. das Ergebnis ist bei beiden das gleiche. Die Kabel habe ich nochmal überprüft und ich habe auch andere USB-Ports auf dem Pi ausprobiert. (Allerdings habe ich nicht bei jedem Port alle 32 Möglichkeiten durchgespielt, werde ich wahrscheinlich den Tag über machen).

Was könnte ich sonst noch falsch machen?

premultiply, 2022-07-16:
>     Ok, einen kleinen Schritt kam ich weiter. 8E1 scheint das richtige comset zu sein. Damit kommt wenigstens ein wenig Information beim Debug, bei 8N1 steht alles auf timout:
> 
> pi@raspberrypi:~ $ sudo evcc -l debug charger

Sehr gut. Hauptsache der Charger spricht überhaupt erstmal.

Könntest du bitte `evcc charger --log trace` nochmal ausführen wenn ein Fahrzeug angeschlossen ist und lädt?

ChristianFleissner, 2022-07-17:
Bitte sehr, war zwar nicht am Laden, sondern nur am klimatisieren, sollte für die Wallbox aber keinen Unterschied machen:

pi@raspberrypi:~ $ evcc charger --log trace
[main  ] INFO 2022/07/17 11:35:00 evcc 0.96 (7ce0297f)
[main  ] INFO 2022/07/17 11:35:00 using config file /home/pi/evcc.yaml
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 03 00 18 00 01 0d f8
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 83 02 d0 ee
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 03 00 14 00 03 4c 3a
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 83 02 d0 ee
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 04 00 10 00 01 39 fa
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 04 02 00 05 35 3b
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 03 00 03 00 01 7d ff
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 03 02 00 10 f5 80
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 03 00 17 00 01 3d fb
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 83 02 d0 ee
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 03 00 67 00 04 fc 23
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 83 02 d0 ee
Power:          modbus: exception '2' (illegal data address), function '3'
Current L1..L3: modbus: exception '2' (illegal data address), function '3'
Charge status:  C
Enabled:        true
Charged:        modbus: exception '2' (illegal data address), function '3'
Identifier:     modbus: exception '2' (illegal data address), function '3'
Diagnostic dump:
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 04 00 30 00 01 38 30
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 04 02 01 04 f5 6b
        Firmware:       1.4
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 04 00 0e 00 01 59 fc
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 84 02 d2 de
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 04 00 10 00 01 39 fa
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 04 02 00 05 35 3b
        State:  5
[kse   ] TRACE 2022/07/17 11:35:01 modbus: send 64 02 00 31 00 01 e1 f0
[kse   ] TRACE 2022/07/17 11:35:01 modbus: recv 64 02 01 00 bf 44
panic: runtime error: index out of range [1] with length 1

goroutine 1 [running]:
encoding/binary.bigEndian.Uint16(...)
        /opt/hostedtoolcache/go/1.18.3/x64/src/encoding/binary/binary.go:101
github.com/evcc-io/evcc/charger.(*KSE).Diagnose(0x41a2b60)
        /home/runner/work/evcc/evcc/charger/kse.go:234 +0x38c
github.com/evcc-io/evcc/cmd.(*dumper).Dump(0x4339d80, {0x41c27b8, 0x8}, {0x1340160, 0x41a2b60})
        /home/runner/work/evcc/evcc/cmd/dumper.go:199 +0x16dc
github.com/evcc-io/evcc/cmd.(*dumper).DumpWithHeader(0x4339d80, {0x41c27b8, 0x8}, {0x1340160, 0x41a2b60})
        /home/runner/work/evcc/evcc/cmd/dumper.go:32 +0x74
github.com/evcc-io/evcc/cmd.runCharger(0x248c9a0, {0x3fdbc60, 0x0, 0x2})
        /home/runner/work/evcc/evcc/cmd/charger.go:126 +0xc68
github.com/spf13/cobra.(*Command).execute(0x248c9a0, {0x3fdbc50, 0x2, 0x2})
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.4.0/command.go:860 +0x688
github.com/spf13/cobra.(*Command).ExecuteC(0x248c338)
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.4.0/command.go:974 +0x414
github.com/spf13/cobra.(*Command).Execute(...)
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.4.0/command.go:902
github.com/evcc-io/evcc/cmd.Execute()
        /home/runner/work/evcc/evcc/cmd/root.go:135 +0x20
main.main()
        /home/runner/work/evcc/evcc/main.go:33 +0x3c
pi@raspberrypi:~ $

premultiply, 2022-07-18:
Danke, schaue ich mir an.

premultiply, 2022-07-18:
Kann es sein dass deine Box neben dem RFID-Zähler auch keinen internen Energiezähler hat?

ChristianFleissner, 2022-07-18:
Meines Wissens hat die keinen internen Zähler, auf dem Datenblatt steht auch nichts darüber.

premultiply, 2022-07-18:
Ok, dann sind die derzeitigen Lesefehler bei den betreffenden Datensätzen nachvollziehbar.
Ich baue das dann so um, dass Zähler und RFID optional sind und automatisch erkannt werden.

ChristianFleissner, 2022-07-19:
Ok, scheint jetzt grundsätzlich zu funktionieren:

pi@raspberrypi:~ $ sudo evcc -l debug charger
[main  ] INFO 2022/07/19 11:28:20 evcc 0.96 (97262a65)
[main  ] INFO 2022/07/19 11:28:20 using config file /home/pi/evcc.yaml
Power:          1010W
Current L1..L3: 3.96A 0.029A 0.119A
Charge status:  C
Enabled:        true
Charged:        5.9kWh
Diagnostic dump:
        Firmware:       1.4
        State:  5

Jetzt muss ich nur noch die Benutzeroberfläche ans laufen kriegen. Dachte bisher, dass das an dem nicht funktionierendem Charger liegt, anscheinend liegt das Problem woanders.

ChristianFleissner, 2022-07-21:
Ok, einen kleinen Bug habe ich noch. Das PV und auch das "Min+PV"-Laden funktioniert. Wenn ich aber auf "Fast" laden will (also zumindest mit den vollen 3,6 kW, wenn keine Sonne da ist), kommt die u.a. Fehlermeldung und er lädt gar nicht. Könnte das mit dem 1p-3p Umschalten zu tun haben, das bei meinem Auto ja nicht geht?

Es wäre hilfreich, wenn er in diesem Fall zumindest mit den vollen 3,6 kW laden würde (bei min+PV lädt er nur mit 1,5, was ja auch richtig ist).

switch phases: modbus: exception '2' (illegal data address), function '6'

premultiply, 2022-07-22:
Ich vermute die Box kann auch keine Umschaltung.
Werde das dann erstmal ausbauen...

closed 2022-07-22 by premultiply

ChristianFleissner, 2022-07-22:
Ok, hier einmal der logtrace bei dem PV-laden (da funktioniert es seit ich in der yaml beim Auto eingestellt habe, dass er nur 1 Phase hat):

pi@raspberrypi:/etc $ evcc charger --log trace
[main  ] INFO 2022/07/22 16:49:11 evcc 0.96 (595a2ca8)
[main  ] INFO 2022/07/22 16:49:11 using config file /etc/evcc.yaml
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 02 00 31 00 01 e1 f0
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 02 01 00 bf 44
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 04 00 18 00 01 b8 38
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 04 02 0e a5 31 23
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 04 00 14 00 03 f9 fa
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 04 06 3a 06 00 59 01 fa 2b 5b
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 04 00 10 00 01 39 fa
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 04 02 00 05 35 3b
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 03 00 03 00 01 7d ff
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 03 02 00 10 f5 80
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 04 00 17 00 01 88 3b
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 04 02 00 e6 74 b2
Power:          3749W
Current L1..L3: 14.9A 0.089A 0.506A
Charge status:  C
Enabled:        true
Charged:        2.3kWh
Diagnostic dump:
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 04 00 30 00 01 38 30
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 04 02 01 04 f5 6b
        Firmware:       1.4
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 04 00 0e 00 01 59 fc
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 84 02 d2 de
[kse   ] TRACE 2022/07/22 16:49:11 modbus: send 64 04 00 10 00 01 39 fa
[kse   ] TRACE 2022/07/22 16:49:11 modbus: recv 64 04 02 00 05 35 3b
        State:  5

Und hier einmal, wenn ich auf "Fast" schalte:

pi@raspberrypi:/etc $ evcc charger --log trace
[main  ] INFO 2022/07/22 16:50:59 evcc 0.96 (595a2ca8)
[main  ] INFO 2022/07/22 16:50:59 using config file /etc/evcc.yaml
[kse   ] TRACE 2022/07/22 16:50:59 modbus: send 64 02 00 31 00 01 e1 f0
[kse   ] TRACE 2022/07/22 16:50:59 modbus: recv 64 02 01 00 bf 44
[kse   ] TRACE 2022/07/22 16:50:59 modbus: send 64 04 00 18 00 01 b8 38
[kse   ] TRACE 2022/07/22 16:50:59 modbus: recv 64 04 02 00 00 f5 38
[kse   ] TRACE 2022/07/22 16:50:59 modbus: send 64 04 00 14 00 03 f9 fa
[kse   ] TRACE 2022/07/22 16:51:00 modbus: recv 64 04 06 00 00 00 00 00 00 f7 c1
[kse   ] TRACE 2022/07/22 16:51:00 modbus: send 64 04 00 10 00 01 39 fa
[kse   ] TRACE 2022/07/22 16:51:00 modbus: recv 64 04 02 00 04 f4 fb
[kse   ] TRACE 2022/07/22 16:51:00 modbus: send 64 03 00 03 00 01 7d ff
[kse   ] TRACE 2022/07/22 16:51:00 modbus: recv 64 03 02 00 00 f4 4c
[kse   ] TRACE 2022/07/22 16:51:00 modbus: send 64 04 00 17 00 01 88 3b
[kse   ] TRACE 2022/07/22 16:51:00 modbus: recv 64 04 02 00 ee 75 74
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Charged:        2.4kWh
Diagnostic dump:
[kse   ] TRACE 2022/07/22 16:51:00 modbus: send 64 04 00 30 00 01 38 30
[kse   ] TRACE 2022/07/22 16:51:00 modbus: recv 64 04 02 01 04 f5 6b
        Firmware:       1.4
[kse   ] TRACE 2022/07/22 16:51:00 modbus: send 64 04 00 0e 00 01 59 fc
[kse   ] TRACE 2022/07/22 16:51:00 modbus: recv 64 84 02 d2 de
[kse   ] TRACE 2022/07/22 16:51:00 modbus: send 64 04 00 10 00 01 39 fa
[kse   ] TRACE 2022/07/22 16:51:00 modbus: recv 64 04 02 00 04 f4 fb
        State:  4

Gemäß Anleitung, Datenblatt und auch dem Registerset sollte die Wallbox eigentlich zwischen 1- und 3-phasig schalten können. Ich werde das aber frühestens in der dritten Septemberwoche testen können, da bekomme ich einen vollelektrischen Ersatzwagen für ein paar Tage.

premultiply, 2022-07-22:
In der Anleitung steht aber auch dass dazu eine bestimmte Softwareversion erforderlich wäre.
Mit dem Register laut Doku scheint es jedenfalls mit deiner Box nicht zu gehen.
Daher habe ich die Automatik erstmal ausgebaut.

ChristianFleissner, 2022-07-22:
Interessant. Ich werde das mal mit dem Hersteller bequatschen. Als ich gefragt habe, ob ich meine Firmware aktualisieren muss hieß es nein.

ChristianFleissner, 2022-07-26:
Insgesamt funktioniert jetzt alles, vielen Dank. Der Charger schmeißt nur öfter Fehlermeldungen, die aber anscheinend keine Auswirkung auf den Betrieb haben. Nur der Vollständigkeit halber:

[image]

premultiply, 2022-07-26:
Da würde ich erstmal pauschal von Übertragungsproblemen auf dem RS485-Bus ausgehen, welche vielfältige Ursachen haben kann. Falsche Kabel, falsche Adern verwendet, Kontaktprobleme, schlechte Leitungstreiber, falsche Schirmung, falsche oder fehlende Terminierung usw. Da müsstest du mal selber schauen ob das alles sauber ist.

Natürlich kann es theoretisch auch sein dass die Box auch mal durch seltsame Firmware mit irgendwas beschäftigt ist und daher tatsächlich nicht antwortet.

ChristianFleissner, 2022-07-26:
Es wird wahrscheinlich daran liegen, dass das Datenkabel im gleichen Leerrohr wie das 5x6mm Stromkabel liegt. Daran werde ich nichts ändern können. Wie gesagt, es funktioniert ja, daher kann ich es auch ignorieren.

4rz0, 2023-08-09:
@premultiply 
> Daher habe ich die Automatik erstmal ausgebaut.

Ich greife das Thema mal auf.
Ist das noch der aktuelle Stand?
Ich könnte wohl Zugriff auf soeine Box bekommen demnächst und würde diese gerne mit evcc zusammen in Betrieb nehmen.

immofan, 2024-05-29:
> In der Anleitung steht aber auch dass dazu eine bestimmte Softwareversion erforderlich wäre.
> Mit dem Register laut Doku scheint es jedenfalls mit deiner Box nicht zu gehen.
> Daher habe ich die Automatik erstmal ausgebaut.

Ich greife das Thema Phasenumschaltung bei der KSE Wallbox nochmal auf. Wie ich gesehen habe, ist das im Code zumindest schon vorgesehen. Ob vollständig kann ich überhaupt nicht beurteilen. Wäre es möglich diesen Teil zu aktivieren?
Die Box kann das ja und die Funktion wäre für meine Anlage recht hilfreich. 
Stehe auch gerne für Testläufe etc. zur Verfügung.
Vielen Dank schon mal...

premultiply, 2024-06-07:
Probier es einfach mal einem Modbus-Tool deiner Wahl "zu Fuß" aus.
Wie oben zu sehen gab es damit ja Probleme.

4rz0, 2024-06-08:
Umschaltung funktioniert bei mir problemlos über das Register 0x35 (modbusRelayMode).
Wallbox-Version ist 0x3452.

premultiply, 2024-06-10:
Hier bitte weiter #14219

PR #3828 (2022-07-14): Add KSE wBX16 charger support
Closes #3824 

Features:
- [x] RFID
- [x] 1p3p - zumindest soll da irgendein "Relais" ansteuerbar sein - zu testen!
- [ ] mA
- [ ] Wakeup - vielleicht automatisch?
PR #3892 (2022-07-18): KSE: Fix registers and detect RFID
Should fix #3824
PR #3919 (2022-07-22): KSE: Disable 1p3p support
Fix #3824