evcc-io/evcc #3846: Recognize connection of vehicles at a switchable socket (e.g. Fritz!DECT)

patbab, 2022-07-14:
Die Beschreibung mache ich mal auf deutsch:

**Is your feature request related to a problem? Please describe.**
Dieses issue greift die in https://github.com/evcc-io/evcc/discussions/3430 beschriebene Problematik auf. Nochmal in neuen Worten und kürzer formuliert:
Eine normale Wallbox erkennt das Anstecken des Fahrzeugs. Der charger status wechselt von „A“ nach „B“ („connected“). In diesem Moment wird ein Abruf des SOC forciert.
An einer schaltbaren Steckdose bleibt der charger status jedoch immer bei „B“, weil die Steckdose das Ab- und Anstecken nicht erkennen kann. Folglich wird aber auch beim Anstecken des Fahrzeugs keine Aktualisierung des SOC forciert.
Ist nun „target SOC“ < 100% eingestellt, so schaltet evcc die Steckdose beim Erreichen des target SOC ab. evcc kann aber das Wiederanstecken des Fahrzeugs nach einer Fahrt nicht erkennen. evcc geht deshalb dauerhaft davon aus, dass der target SOC erreicht ist, und starten keinen Ladevorgang. Ein Ladevorgang kann nur gestartet werden, indem man den SOC auf 100% und ggf. auf „Sofort“ (wenn die PV-Leistung nicht ausreicht) stellt. Wenn der erstmalige Abruf des SOC nicht klappt (siehe dazu https://github.com/evcc-io/evcc/discussions/2772#discussioncomment-2835182 und https://github.com/evcc-io/evcc/discussions/3430), muß man >15 min warten, ehe man wieder in den „PV"-Modus und zum gewünschten Ziel-SOC zurückwechseln kann.
Relevant ist die Problematik v.a. bei kleineren EVs, die über Schuko geladen werden, und die man zur Schonung des Akkus nicht immer auf 100% laden möchte.

**Describe the solution you'd like**
An der Fritz!DECT gibt es eine Taste, mit der man die Steckdose manuell anschalten kann. Allerdings erkennt evcc im nächsten Zyklus sofort, dass die Steckdose „unerwünscht“ an ist und schaltet sie wieder aus („charger out of sync: expected disabled, got enabled“). Die Zeit reicht hier nicht einmal jedes 2. Mal (vermutlch auch abhängig von `interval`) aus, dass sich evcc einen aktuellen SOC holt. Dennoch schaltet auch in dem „erfolgreichen“ Fall evcc kurz die Steckdose aus, „sieht“ dann erst den neuen SOC, und schaltet dann nach den entspr. Delay-Zeiten wieder ein.
Eine Lösung wäre, dass evcc bei einem erkannten „charger out of Sync“ bei einer schaltbaren Steckdose grundsätzlich eine SOC-Abfrage beim angesteckten Fahrzeug anstößt. Der „out of sync“ sollte dann so lange toleriert werden, bis der SOC aktualisiert wurde, um zu entscheiden, ob der Benutzer bewusst über den manuellen Taster den Ladevorgang starten wollte oder tatsächlich ein Sync-Fehler vorliegt. Es wäre sehr komfortabel, wenn man einfach per Knopfdruck den Ladevorgang starten könnte, denn man steht beim Anstecken eh neben der Steckdose.

andig, 2022-07-14:
"charger out of Sync" ist ein Fehler der alle Charger betrifft. Für Steckdosen kann es keine Sonderbehandlung geben.

andig, 2022-07-14:
Ich sehe hier noch kein Muster, dass sich auf alle Steckdosen abstrahieren ließe. Was hält Dich davon ab, den Ladepunkt extern per Api zu steuern?

patbab, 2022-07-14:
Dass ich dann extra das Handy oder dergl. rauskramen muß. Ich stehe aber direkt neben der Steckdose, habe das Kabel in der Hand zum Einstecken und drücke dann einfach noch den Taster an der Steckdose, damit evcc weiß, da hat sich was geändert.

patbab, 2022-07-14:
Oder anders formuliert, letztlich geht es um die Frage, wie man auf möglichst einfachem Weg evcc dazu bringen kann, zu realisieren, dass ein Fahrzeug neu angesteckt wurde und dass der SOC aktualisiert werden sollte.

andig, 2022-07-15:
> Dass ich dann extra das Handy oder dergl. rauskramen muß. Ich stehe aber direkt neben der Steckdose, habe das Kabel in der Hand zum Einstecken und drücke dann einfach noch den Taster an der Steckdose, damit evcc weiß, da hat sich was geändert.

Du kannst auch in Homeassisstant auf den Knopf reagieren.

> letztlich geht es um die Frage, wie man auf möglichst einfachem Weg evcc dazu bringen kann, zu realisieren, dass ein Fahrzeug neu angesteckt wurde und dass der SOC aktualisiert werden sollte.

 Ich sehe hier keinen allgemeingültigen Ansatz der sich implementieren liesse. Die Steckdosen sind halt Krückenlösung.

patbab, 2022-07-15:
Sehe ich etwas anders. Ich müsste ja dann in einem beliebig hohen Takt aus einer separaten Anwendung heraus abfragen, ob die Steckdose manuell angeschalten wurde. Denn wenn ich zufällig 1 sek vor Ablauf des Aktualisierungsintervalls von evcc den Knopf drücke, schaltet evcc die Steckdose binnen 1 sek. wieder aus.
Die Frage ist ja auch, ob die Steckdose überhaupt mitteilt, ob sie manuell geschalten wurde, oder per app, oder von evcc.
Als Krückenlösung würde ich die Steckdosen nicht bezeichnen, es ist doch die einzige Lösung für Schuko-versorgte EVs.

Bin eher am überlegen, ob ich per skript das log von evcc auswerte und von dort entsprechend reagiere. Kann man über die REST API mehr schreiben, als das, was in der Doku steht? Z.b. einen poll der Fahrzeugdaten forcieren oder den poll.interval ändern? Dieses issue hier ist ja mit https://github.com/evcc-io/evcc/issues/3847 verwoben, was mögliche Lösungsansätze betrifft.