evcc-io/evcc #3869: BMW wird nicht mehr erkannt

pk1057, 2022-07-17:
### Describe the bug

Hallo,
wie auch schon in Bug Report #3829 beschrieben wird keine meiner BMWs mehr erkannt.
weder 0.95, latest oder auch nightlies.

BMW hat ja vor kurzem seine App aktualisiert, evtl. eine Änderung der API oder der URL

### Steps to reproduce

start app

### Configuration details

ubuntu docker container

### Log details

/app # evcc -l debug vehicle
[main  ] INFO 2022/07/17 11:11:21 evcc 0.96
[main  ] INFO 2022/07/17 11:11:21 using config file /etc/evcc.yaml
ev3
---
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot find vehicle: WBAXXXXXXXX
Capacity: 0kWh

ev4
---
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot find vehicle: WBYXXXXXXXXXXX
Capacity: 0kWh

### What type of operating system are you running?

Linux

### Version

0.95, latest, nightly

andig, 2022-07-22:
Wie in https://github.com/evcc-io/evcc/issues/3847#issuecomment-1191792849 gezeigt, schickt BMW eine Antwort die keine Fahrzeuge (=leere Liste) enthält, sonst aber fehlerfrei ist.

closed 2022-07-22 by andig

DanielGlaas, 2022-07-22:
Wow, danke für den schnellen Fix!