evcc-io/evcc #3870: OCPP: log unknown station ids

andig, 2022-07-17:
@xantalor könnten wir mal überlegen, die OCPP Implementierung einfach "auf" zu machen und die erste Station die sich meldet zuzulassen oder zumindest im INFO Level zu loggen? Es scheint, als wäre die Ermittlung des Station IDs für Laien je nach Box fast unmöglich...

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/3806#discussioncomment-3164627_

xantalor, 2022-07-17:
Habe ich schon mitbekommen, find die Idee nicht so schlecht. 
Vorschlag: Wenn man keine id angibt, wird die erste genommen und auch gelogged und wenn man eine angibt wird nur die zugelassen. Könnte mich glaub die Tage darum kümmern.

StefanSchoof, 2022-07-17:
Könnte man wenn es mehre gibt auch alle loggen? Kann mir vorstellen, dass dies hilfreich ist, wenn man mehre hat und nicht genau weiß welche. Dann kann man die zumindest durchprobieren.

closed 2022-07-17 by andig