evcc-io/evcc #3885: Lademodi Schnell und PV laden nur mit Mindestleistung

DanielGlaas, 2022-07-18:
### Describe the bug

Mit evcc 0.96 lädt das Auto über goe-charger nur mit der Mindestleistung, obwohl jede Menge Solarüberschuss vorhanden ist (siehe Screenshot):
[image]

Leider habe ich zur Umstellung der Modi keine Infos gefunden, "Schnell" ist in der aktuellen Doku nicht erklärt ([#181](https://github.com/evcc-io/docs/issues/181)). Meine Erwartung ist aber, dass es so viel lädt wir möglich wenn PV-Überschuss vorhanden ist. Mit "PV" ist das Verhalten identisch.

P.S.:
In den Changelogs finde ich keinen Hinweis auf den neuen Modus "Schnell". Weder die Suche nach "mode", "modus" noch nach "schnell" gibt einen Treffer. Daher kann ich auch nicht nachvollziehen, mit welcher Version das reinkam. Bei 0.93 waren die Modi definitiv noch anders und das Laden hat auch mit voller 1-phasiger Leistung funktioniert.

### Steps to reproduce

Umschalten des Modus zwischen "PV" und "Schnell" hat keinen Einfluss, für beide gibt es das identische Verhalten.

### Configuration details

[evcc.txt](https://github.com/evcc-io/evcc/files/9131021/evcc.txt)

### Log details

[main  ] INFO 2022/07/18 11:50:20 evcc 0.96
[main  ] INFO 2022/07/18 11:50:20 using config file /home/pi/evcc.yaml
[main  ] INFO 2022/07/18 11:50:20 listening at :7070
[go-e  ] WARN 2022/07/18 11:50:21 automatic 1p3p phase switching requires sponsor token
[site  ] INFO 2022/07/18 11:50:23 site config:
[site  ] INFO 2022/07/18 11:50:23   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/07/18 11:50:23     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/07/18 11:50:23     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/07/18 11:50:23     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/07/18 11:50:23 loadpoint 1:
[lp-1  ] INFO 2022/07/18 11:50:23   mode:        pv
[lp-1  ] INFO 2022/07/18 11:50:23   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/07/18 11:50:23   meters:      charge ✓
[lp-1  ] INFO 2022/07/18 11:50:23     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/07/18 11:50:23   vehicles:    ✓
[lp-1  ] INFO 2022/07/18 11:50:23     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-2  ] INFO 2022/07/18 11:50:23 loadpoint 2:
[lp-2  ] INFO 2022/07/18 11:50:23   mode:        off
[lp-2  ] INFO 2022/07/18 11:50:23   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2022/07/18 11:50:23   meters:      charge ✗
[lp-2  ] INFO 2022/07/18 11:50:23   vehicles:    ✓
[lp-2  ] INFO 2022/07/18 11:50:23     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/07/18 11:50:23 vehicle updated: unknown -> Opel Elektrisch
[lp-1  ] DEBUG 2022/07/18 11:50:24 vehicle odometer: 2026km
[lp-1  ] DEBUG 2022/07/18 11:50:24 max charge current: 6A
[lp-2  ] INFO 2022/07/18 11:50:24 vehicle updated: unknown -> unavailable
[site  ] DEBUG 2022/07/18 11:50:24 ----
[server] FATAL 2022/07/18 11:50:24 remove /tmp/evcc: operation not permitted

### What type of operating system are you running?

Linux

### Version

evcc version 0.96

VolkerK62, 2022-07-18:
Bei der Layout-Änderung wurde `sofort` zu `schnell`. Aber an der Funktionalität hat sich nix geändert. `schnell` ist völlig unabhängig von PV-Leistung. 
Ist evtl im Auto oder in der Wallbox die Ladeleistung limitiert?

andig, 2022-07-18:
Dein Charger hat lauf Logfile keine Phasenumschaltung?

closed 2022-07-18 by andig

DanielGlaas, 2022-07-18:
@andig 
Dann sollte aber doch zumindest die einphasige Ladung bis 4,2 kW gehen. Das ging in den vergangenen Versionen auch ohne Sponsortoken.

DanielGlaas, 2022-07-18:
@VolkerK62 
Danke für den Hinweis mit der Änderung von `sofort` zu `schnell`. Dadurch, dass sich die Reihenfolge der Modi in der GUI auch geändert hat, war ich davon ausgegangen, dass da mehr dahinter steckt. Ist vielleicht etwas ungünstig gelaufen, dass diese Umbenennung nirgends steht.

Das mit der limitierten Ladeleistung im Auto oder Wallbox werde ich mal nachschauen. Die Wallbox würde ich aber eigentlich ausschließen, denn die bekommt ja von evcc den Befehl für mehr Leistung.

andig, 2022-07-18:
> @andig
> Dann sollte aber doch zumindest die einphasige Ladung bis 4,2 kW gehen. Das ging in den vergangenen Versionen auch ohne Sponsortoken.

Ja, sollte es! Wenns das nicht tut gerne entsprechendes Logfile, dann machen wir hier wieder auf!