evcc-io/evcc #3891: Cannot create OCPP Charger

ExecuteGK, 2022-07-18:
### Describe the bug

In order to control a Daheimladen Wallbox within the local network and without the need of an active Internet connection, I redirected the OCPP Backend from the daheimladen cloud to the EVCC Server.

After some tries, I am getting the following error during startup:

 FATAL 2022/07/18 22:42:41 cannot create charger 'daheimladen': cannot create charger 'ocpp': smart charging key 'ChargingScheduleAllowedChargingRateUnit' not found

The TCP Dump shows the following configurationKeys in the Websocket traffic of the WB:

[3,"3913962819",{"configurationKey":[{"key":"QrCode","value":"serialOfWb","readonly":false},{"key":"ChargeRate","value":"16","readonly":false},{"key":"chargingStationType","value":"SingleSocket","readonly":true},{"key":"firmwareVersion","value":"AC_DG3_1.RSDA","readonly":true},{"key":"maxChargingPower","value":"11","readonly":true},{"key":"namePlateNumber","value":"WYSC22-41620644","readonly":true},{"key":"HeartbeatInterval","value":"60","readonly":false},{"key":"LocalAuthorizeOffline","value":"true","readonly":false},{"key":"MeterValueSampleInterval","value":"5","readonly":false},{"key":"NumberOfConnectors","value":"1","readonly":true},{"key":"SupportedFeatureProfiles","value":"Core,Local Auth List Management,Reservation,Smart Charging,Remote Trigger","readonly":true},{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register,Power.Active.Import,Temperature,Voltage,Current.Import","readonly":true},{"key":"ConnectionTimeout","value":"60","readonly":false},{"key":"LocalAuthorizeOffline","value":"true","readonly":false},{"key":"SendLocalListMaxLength","value":"10","readonly":true},{"key":"LocalAuthListMaxLength","value":"200","readonly":true},{"key":"ChargeProfileMaxStackLevel","value":"7","readonly":true},{"key":"ChargingScheduleMaxPeriods","value":"5","readonly":true},{"key":"MaxChargingProfilesInstalled","value":"5","readonly":true}]}]

Using the cloud, the WB can be managed. However, the goal is to have a cloud independent setup so that all charging and home automation works without an active internet connection.

Please let me know if you need more data.

Viele Grüße

Sebastian

### Steps to reproduce

Configure ocpp wallbox
Start evcc

### Configuration details

evcc.yaml entry for the box looks like:

- type: ocpp
  stationid: serialOfWb # Die Stations-ID der Wallbox
  meterSupported: true
  name: daheimladen

### Log details

Jul 18 22:42:41 flowcontrol evcc[20280]: [main  ] FATAL 2022/07/18 22:42:41 cannot create charger 'daheimladen': cannot create charger 'ocpp': smart charging key 'ChargingScheduleAllowedChargingRateUnit' not found

### What type of operating system are you running?

Linux

### Version

evcc version 0.96

premultiply, 2022-07-18:
According to the OCPP 1.6 Specification `ChargingScheduleAllowedChargingRateUnit` is a required key.
Ref. https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf

/cc @utsavanand2

Hofyyy, 2022-07-19:
Hello. @ExecuteGK,

I had also some trouble with wallbe regarding OCPP. Bit they helped me with a patch and a new firmware. They only tested against their own cloud. 

I would contact the support from daheimladen and ask about their input regarding this topic.

Best regards
Sebastian

andig, 2022-07-20:
Thank you for the confirmation. @ExecuteGK then the error is as expect. We require the Smart Charging Profile.

ExecuteGK, 2022-07-20:
Well,

then the implementation is flawed. In the response, the WB states: SupportedFeatureProfiles", "value": "Core,Local Auth List Management,Reservation,Smart Charging,Remote Trigger", "readonly": true

This  means that the box states that it supports Smart Charging Profile. 

If it doesn't, the Box shouldn't report it as supported.

Regards,

Sebastian