evcc-io/evcc #3918: S01: initial soc request fails

andig, 2022-07-22:
Nur so als Gedenken von mir dazu. Vorweggeschickt habe ich jedoch keinen Einblick, wie diese Authentifikation genau funktioniert. Ich sehe bei dem BMW-Fall aber gewisse Parallelen zu dem, wie es bei mir beim S01 ist.

So wie ich das verstehe, ist der poll-Vorgang Zweistufig. Im ersten Schritt wird vom Authentifiierungs-Server das bearer-token geholt. Im zweiten Schritt dann vom eigentlichen Daten-Server die Daten.

Mit "evcc vehicle" hatte ich bisher nie ein problem. Sehr wohl aber, wenn ich evcc vollständig neu starte. Wenn in dem Zustand mit dem S01 ein Fehler passiert, so kommen auch diese Fehlermeldungen "vehicle not available" und "cannot create vehicle". Siehe Anhang.
[log_s01_220601.txt](https://github.com/evcc-io/evcc/files/9166418/log_s01_220601.txt)

Was passiert eigentlich Server-seitig zwischen den beiden Servern (Auth, Daten) bei so einer Abfrage?

Ich kann nicht oft genug betonen, dass der fehler immer nur bei eriner Erst-Abfrage passiert. Das bearer-Token gilt ja eine Zeit lang und wird nicht jedes Mal neu erzeugt laut Log. Bei einer solchen Nach-Abfrage ist bei mir bisher nie ein Fehler aufgetreten, sondern immer nur, wen vorher das bearer-token abgefragt wird.

Eine naive Idee meinerseits: Evtl. kommt die eigentliche Anfrage der Daten zu schnell nach der bearer-Token-Antwort, und der Daten-Server ist noch nicht weit genug?

Patrick

_Originally posted by @patbab in https://github.com/evcc-io/evcc/issues/3847#issuecomment-1192367546_

andig, 2022-07-22:
> Ich kann nicht oft genug betonen, dass der fehler immer nur bei eriner Erst-Abfrage passiert. Das bearer-Token gilt ja eine Zeit lang und wird nicht jedes Mal neu erzeugt laut Log. Bei einer solchen [...]

Das ist interessant. Könntest Du info@evcc.io mal bitte Zugangsdaten zukommen lassen damit ich das testen kann?

patbab, 2022-07-22:
ich checke gerade nochmal alle logs durch. Hier noch eine andere Variante der Fehlermeldung, die wesentlich seltener auftritt als die 500: Beim nächsten Versuch um 14:13 dann nur der POST-Befehl, der erfolgreich ist.
Ich suche mal noch, ob das auch beim 500er-Fehler so ist, dass der nächste Post-Befehl direkt erfolgreich ist, ohne dass die Sache neu gestartet wurde.

```
Jun 03 13:58:15 evcc evcc[21618]: [s01   ] TRACE 2022/06/03 13:58:15 POST https://www.googleapis.com/identitytoolkit/v3/relyingparty/verifyPassword?alt=json&key=AIzaSyCQYZCPvfl-y5QmzRrbUrCwR0RVNbyKqwI&prettyPrint=false
Jun 03 13:58:15 evcc evcc[21618]: [cache ] TRACE 2022/06/03 13:58:15 lp-2/enabled: false
Jun 03 13:58:15 evcc evcc[21618]: [s01   ] TRACE 2022/06/03 13:58:15 {"email":"***","password":"***","returnSecureToken":true}
Jun 03 13:58:15 evcc evcc[21618]: --
Jun 03 13:58:15 evcc evcc[21618]: {"kind":"identitytoolkit#VerifyPasswordResponse","localId":"OuLTDAxOtVWXhBU4siGPxyjKyus1","email":"***","displayName":"","idToken":"eyJhbGciOiJSUzI1NiIsImtpZCI6ImY0ZTc2NDk3ZGE3Y2ZhOWNjMDkwZDcwZTIyNDQ2YTc0YjVjNTBhYTkiLCJ0eXAiOiJKV1QifQ.eyJodHRwOi8vc2NoZW1hcy5taWNyb3NvZnQuY29tL3dzLzIwMDgvMDYvaWRlbnRpdHkvY2xhaW1zL3JvbGUiOlsiTWdtdC5DdXN0b21lciJdLCJpc3MiOiJodHRwczovL3NlY3VyZXRva2VuLmdvb2dsZS5jb20vc2Nvb3Rlci1jb25uZWN0aXZpdHktcHJvZCIsImF1ZCI6InNjb290ZXItY29ubmVjdGl2aXR5LXByb2QiLCJhdXRoX3RpbWUiOjE2NTQyNTc0OTUsInVzZXJfaWQiOiJPdUxUREF4T3RWV1hoQlU0c2lHUHh5akt5dXMxIiwic3ViIjoiT3VMVERBeE90VldYaEJVNHNpR1B4eWpLeXVzMSIsImlhdCI6MTY1NDI1NzQ5NSwiZXhwIjoxNjU0MjYxMDk1LCJlbWFpbCI6InBhdHJpY2suYmFiaW5nZXJAdXIuZGUiLCJlbWFpbF92ZXJpZmllZCI6dHJ1ZSwiZmlyZWJhc2UiOnsiaWRlbnRpdGllcyI6eyJlbWFpbCI6WyJwYXRyaWNrLmJhYmluZ2VyQHVyLmRlIl19LCJzaWduX2luX3Byb3ZpZGVyIjoicGFzc3dvcmQifX0.SDYBpGIyNSC9eE5lWj7OVnc3IltOw1CdW03ys5tlcfjB9MFrxHNazDideCER6tbmfYbh1fnymsFQwlJh2Cc_6K1XqNfoo20I9LpqjKULXoVQC_lHkMRqf5KpHEQQu3hr2hHgamSC4d_0UgUQq-q1OcHYcG8FqorN8qdeWfoJ3fb51H6AjNyzpbsOmVf2de9K8BkOvZIZd9mNY89r0oVyaw5vwcQ0bdGGVE8YtV1ecWOCBXyz9OXj8C6v2pXkxRXa2lPXtRm1Lyo-QjQbSe6yYRe5NiYtprw3KFGgOyQvn9sXM5oJsWT6v4ht702nPev5LMgpSCcBJPDNSG0zrlRQjQ","registered":true,"refreshToken":"AIwUaOlg27xd5MxLwctpRa9R2X8RLH_JgyUmHkBrzYIsmAQIq9J_26Ka3YRkxdFCW4vY5ZD9TVn_FTtXMUwUU_6F8L2jdxEDtXhGiMNgNwEgsH3loyIVY09mddD1ajnP76cLVSE51BE6aQveUCPpN5s3xSdSC0Wak-nJAc73kJWDpsRSpW8dpLaBPrOHsCi3C3WuOhiAI_UcexdzkCqwibVXKQr-OURRGLjNYIlJQVQG90tIMQrV3MRII_PVj5DHB1pJ0dgglQrt","expiresIn":"3600"}
Jun 03 13:58:15 evcc evcc[21618]: [s01   ] TRACE 2022/06/03 13:58:15 GET https://api.connectivity.silence.eco/api/v1/me/scooters?details=true&dynamic=true
Jun 03 13:58:25 evcc evcc[21618]: [lp-2  ] WARN 2022/06/03 13:58:25 vehicle soc: Get "https://api.connectivity.silence.eco/api/v1/me/scooters?details=true&dynamic=true": context deadline exceeded (ignored by estimator)

[...]
Jun 03 14:13:35 evcc evcc[21618]: [cache ] TRACE 2022/06/03 14:13:35 lp-2/charging: true
Jun 03 14:13:35 evcc evcc[21618]: [s01   ] TRACE 2022/06/03 14:13:35 GET https://api.connectivity.silence.eco/api/v1/me/scooters?details=true&dynamic=true
Jun 03 14:13:35 evcc evcc[21618]: [cache ] TRACE 2022/06/03 14:13:35 lp-2/enabled: true
Jun 03 14:13:36 evcc evcc[21618]: [s01   ] TRACE 2022/06/03 14:13:36 [{"id":"5ee89ea297a1c000013d3e5b","model":"S01","revision":"0.1","color":"black","name":"Patrick","sharedToMe":false,"batteryOut":false,"alarmActivated":false,"charging":true,"friendSharing":null,"imei":"358887095159065","btMac":"aa:00:00:00:00:00","frameNo":"UCYS01300LB001305","plate":null,"manufactureDate":"2020-06-15","lastLocation":{"latitude":48.916416,"longitude":12.012244,"altitude":380,"currentSpeed":0,"time":"2022-06-03T12:12:55Z"},"batteryId":425023877,"batterySoc":73,"odometer":1928,"batteryTemperature":31,"motorTemperature":0,"inverterTemperature":0,"range":75,"velocity":0,"status":6,"lastReportTime":"2022-06-03T12:12:56Z","lastConnection":"2022-06-03T12:12:56Z"}]
Jun 03 14:13:36 evcc evcc[21618]: [lp-2  ] DEBUG 2022/06/03 14:13:36 vehicle soc: 73%
```

patbab, 2022-07-22:
Hier haben wir so einen Fall.
Erster Abruf scheitert. evcc läuft einfach weiter. 15 min später dann nur der Post-Befehl, erfolgreich ist.

```
May 27 14:26:19 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:26:19 charger status: C
May 27 14:26:19 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:26:19 charge start energy: 195kWh
May 27 14:26:19 evcc evcc[9087]: [lp-2  ] INFO 2022/05/27 14:26:19 start charging ->
May 27 14:26:19 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:26:19 lp-2/chargeCurrent: 3
May 27 14:26:19 evcc evcc[9087]: [s01   ] TRACE 2022/05/27 14:26:19 POST https://www.googleapis.com/identitytoolkit/v3/relyingparty/verifyPassword?alt=json&key=AIzaSyCQYZCPvfl-y5QmzRrbUrCwR0RVNbyKqwI&prettyPrint=false
May 27 14:26:19 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:26:19 lp-2/connected: true
May 27 14:26:19 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:26:19 lp-2/charging: true
May 27 14:26:19 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:26:19 lp-2/enabled: true
May 27 14:26:20 evcc evcc[9087]: [s01   ] TRACE 2022/05/27 14:26:20 {"email":"***","password":"***","returnSecureToken":true}
May 27 14:26:20 evcc evcc[9087]: --
May 27 14:26:20 evcc evcc[9087]: {"kind":"identitytoolkit#VerifyPasswordResponse","localId":"OuLTDAxOtVWXhBU4siGPxyjKyus1","email":"***","displayName":"","idToken":"eyJhbGciOiJSUzI1NiIsImtpZCI6IjZmOGUxY2IxNTY0MTQ2M2M2ZGYwZjMzMzk0YjAzYzkyZmNjODg5YWMiLCJ0eXAiOiJKV1QifQ.eyJodHRwOi8vc2NoZW1hcy5taWNyb3NvZnQuY29tL3dzLzIwMDgvMDYvaWRlbnRpdHkvY2xhaW1zL3JvbGUiOlsiTWdtdC5DdXN0b21lciJdLCJpc3MiOiJodHRwczovL3NlY3VyZXRva2VuLmdvb2dsZS5jb20vc2Nvb3Rlci1jb25uZWN0aXZpdHktcHJvZCIsImF1ZCI6InNjb290ZXItY29ubmVjdGl2aXR5LXByb2QiLCJhdXRoX3RpbWUiOjE2NTM2NTQzODAsInVzZXJfaWQiOiJPdUxUREF4T3RWV1hoQlU0c2lHUHh5akt5dXMxIiwic3ViIjoiT3VMVERBeE90VldYaEJVNHNpR1B4eWpLeXVzMSIsImlhdCI6MTY1MzY1NDM4MCwiZXhwIjoxNjUzNjU3OTgwLCJlbWFpbCI6InBhdHJpY2suYmFiaW5nZXJAdXIuZGUiLCJlbWFpbF92ZXJpZmllZCI6dHJ1ZSwiZmlyZWJhc2UiOnsiaWRlbnRpdGllcyI6eyJlbWFpbCI6WyJwYXRyaWNrLmJhYmluZ2VyQHVyLmRlIl19LCJzaWduX2luX3Byb3ZpZGVyIjoicGFzc3dvcmQifX0.Vcm_hoh-dVLOy_KF4ocnGEYJOJsJX5etQljPmvximTKk0FpsQ5_u-b1Pzt7BCV_poaFW8plhwXwC-Ov7yhsL8DVc75aFDLaA4yDFFWZ9h7_eGfp1WljTNyIqwU5b5WaujK-1zqBi-xqrF9O90UmpPSDanRrazCEhDy84437BPjfgpSDASFScHCrVjcKIFd4aQzznj-WoR7B7FE5onE6c6U8MuCEEKpoZ88r-WzXO_muGrC9PTx9pELW_siCmea4V2ZeXpavtSG0VJz9vpw-kSRAwUWgrpWnf37NO2_4yRlPy85sVjtkxiq4hZXhiqE1LW6HFCAaK-c3-27E_J72vjw","registered":true,"refreshToken":"AIwUaOlkXIw1-G8ML0Y6rsnNH5ui8GQP815jQBbjDbdrF3JJceWHJtKdYJad6F1NEcBpsPUOibH0yLD7fSt-s03dL4djkSirfvVC2WEV43MPucFi-Jx1gcnBiZzTU0qYgnNLQnNK9GDSUpAldzIKTPqLajIxT0p8r5Fj0E_djkcPxxI01-9SpruGU13T3GpTl5sNA_u51Ye02H685EglW1Ddp1kxdy1TQLCShQ0E-cdGK17oDl_m_KyjebGCMv8CByeMlKdiqOGS","expiresIn":"3600"}
May 27 14:26:20 evcc evcc[9087]: [s01   ] TRACE 2022/05/27 14:26:20 GET https://api.connectivity.silence.eco/api/v1/me/scooters?details=true&dynamic=true
May 27 14:26:20 evcc evcc[9087]: [s01   ] TRACE 2022/05/27 14:26:20 {"code":500,"message":"Internal server error"}
May 27 14:26:20 evcc evcc[9087]: [lp-2  ] WARN 2022/05/27 14:26:20 vehicle soc: unexpected status: 500 (ignored by estimator)
May 27 14:26:20 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:26:20 soc estimated: 92.00% (vehicle: 92.00%)
May 27 14:26:20 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:26:20 vehicle soc: 92%
May 27 14:26:20 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:26:20 lp-2/vehicleSoC: 92

[…]

May 27 14:41:39 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:41:39 charger status: C
May 27 14:41:39 evcc evcc[9087]: [s01   ] TRACE 2022/05/27 14:41:39 GET https://api.connectivity.silence.eco/api/v1/me/scooters?details=true&dynamic=true
May 27 14:41:39 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:41:39 lp-2/connected: true
May 27 14:41:39 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:41:39 lp-2/charging: true
May 27 14:41:39 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:41:39 lp-2/enabled: true
May 27 14:41:40 evcc evcc[9087]: [s01   ] TRACE 2022/05/27 14:41:40 [{"id":"5ee89ea297a1c000013d3e5b","model":"S01","revision":"0.1","color":"black","name":"Patrick","sharedToMe":false,"batteryOut":false,"alarmActivated":false,"charging":true,"friendSharing":null,"imei":"***","btMac":"aa:00:00:00:00:00","frameNo":"UCYS01300L***","plate":null,"manufactureDate":"2020-06-15","lastLocation":{"latitude":48.916435,"longitude":12.012264,"altitude":380,"currentSpeed":0,"time":"2022-05-27T12:35:46Z"},"batteryId":425023877,"batterySoc":90,"odometer":1818,"batteryTemperature":18,"motorTemperature":0,"inverterTemperature":0,"range":93,"velocity":0,"status":6,"lastReportTime":"2022-05-27T12:35:47Z","lastConnection":"2022-05-27T12:35:47Z"}]
May 27 14:41:40 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:41:40 vehicle soc: 90%
May 27 14:41:40 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:41:40 vehicle range: 93km
May 27 14:41:40 evcc evcc[9087]: [lp-2  ] DEBUG 2022/05/27 14:41:40 vehicle odometer: 1818km
May 27 14:41:40 evcc evcc[9087]: [cache ] TRACE 2022/05/27 14:41:40 lp-2/vehicleSoC: 90
```

andig, 2022-07-23:
Ich kanns leider nicht nachvollziehen. Auch in 20x `evcc vehicle` klappt es immer und sofort im ersten Anlauf. Was ist bei Dir anders???

patbab, 2022-07-24:
Bei `evcc vehicle` klappt es ja auch bei mir immer. 
Nochmals in aller kürze:
a) Weit über 50 Tests unter verschiedenen Bedingungen, 80h laden.
b) `evcc vehicle` hat immer geklappt, bis auf 1x. Fehler war aber andersartig, hab leider kein Log dazu. Auch bestimmt 20x probiert. Allerdings, vielleicht ist das noch ein Punkt, auf einer anderen Maschine (Windows). Das laufende evcc läuft au einem RaspPi.
c) Beim Start eines Ladevorgangs im laufenden Betrieb kommt aber der beschriebene Fehler jedes ca. 2. Mal.
d) Beim Neustart von evcc kommt der Fehler auch öfters, wurde aber nicht konsequent getestet.
e) Es ist noch nicht klar, ob der Roller direkt vorher gefahren sein muß, oder ob es reicht, dass der Roller z.B. >1h steht. (vgl. g)
f) Fehler innerhalb laufender Ladevorgänge sind ebenso extrem selten (1x glaube ich geobachtet). Das würde aber heißen, dass es nicht allein der vollständige Auth-Prozess ist (g), der das Problem macht, weil das wassierr ja jede Stunde.
g) Der Fehler kommt immer nur, wenn nacheinander der POST- und GET-Befehl kommen. Wenn das bearer Token schon da ist, klappt alles. (vgl. Hinweis bei e), weil token dann abgelaufen). Das ist aber natürlich beim Start des Ladevorgangs immer so.

Nicht zuletzt deshalb auch mein Vorschlag, das über einen simplen retry zu lösen (https://github.com/evcc-io/evcc/issues/3847), denn es könnte schwierig werden, die Ursache zu finden. Oder mal zwischen POST und GET eine kurze Pflichtwartezeit einbauen,
Hauptproblematik ist, dass ein nicht erfolgreicher Abruf des SOC zu Ladebeginn eine wesentlich höhere Problematik bietet. V.a. in Kombination mit den schaltbaren Steckdosen ist dann nämlich über 15 min kein Laden ohne manuellen Eingriff möglich (https://github.com/evcc-io/evcc/issues/3846)

andig, 2022-07-24:
> Oder mal zwischen POST und GET eine kurze Pflichtwartezeit einbauen

Das könnte man machen wenns mal einen reproduzierbaren Case gäbe, kannst Du aber auch selber schnell ausprobieren. Ich fürchte wir kommen nicht weiter :(

andig, 2022-07-24:
> Beim Start eines Ladevorgangs im laufenden Betrieb kommt aber der beschriebene Fehler jedes ca. 2. Mal.

Ergänzung: Dein Logfile bezog sich konkret auf den neu bezogenen Token. Der hat aber mit Start eines Ladevorgangs absolut nix zu tun bzw. passiert nur zufällig falls eben genau dann ein Token benötigt wird.

patbab, 2022-07-24:
Das ist klar. Neubezug des Tokens und Ladestart gehen i.d.R. immer Hand in Hand.

Auffällig bleibt, dass bei `evcc vehicle` der Fehler nie kommt, beim Neustart eines Ladevorgangs jedoch schon. Also muß da irgendwas verschieden sein. Sei es auf Server-Seite, oder auf evcc-Seite. Auffällig ist auch, dass  `evcc vehicle` manchmal viel länger braucht und auf irgendwas "wartet".

Ich habe gerade mal das log von `evcc vehicle` genauer angeschaut. Diesmal auf dem raspberry ausgeführt. Auffällig ist, dass beim S01 der Get-Befehl 2x ausgeführt wird (beim Fiat aber anscheinend auch einiges doppelt). Siehe Anhang. 1x bei 13:47:55, 1x bei 13:48:07. Wenn dies tatsächlich immer so ist, und meiner Theorie folgend nur der Erstabruf das Problem ist, so wäre hiermit erklärbar, warum `evcc vehicle` i.r.R. ohne "finalen" Fehler läuft.  `evcc vehicle` macht also anscheinend einen retry?

[log_vehicle.txt](https://github.com/evcc-io/evcc/files/9175719/log_vehicle.txt)

andig, 2022-07-24:
Hier gehts nicht um Fiat.

andig, 2022-07-24:
> Auffällig ist auch, dass evcc vehicle manchmal viel länger braucht und auf irgendwas "wartet".

Wo passiert das beim S01?

patbab, 2022-07-24:
Meine Beobachtung über die letzten Wochen, wenn ich immer wieder `evcc vehicle` probiert habe. Nicht im Fall des obigen log-files. Habe leider keine weiteren logs parat, werde die nächste Tests genauer drauf achten und dann auch das log speichern.

andig, 2022-07-25:
> Meine Beobachtung über die letzten Wochen, wenn ich immer wieder `evcc vehicle` probiert habe. Nicht im Fall des obigen log-files. Habe leider keine weiteren logs parat, werde die nächste Tests genauer drauf achten und dann auch das log speichern.

Ich kanns leider wie oben geschrieben nicht nachvollziehen:

> Ich kanns leider nicht nachvollziehen. Auch in 20x evcc vehicle klappt es immer und sofort im ersten Anlauf. Was ist bei Dir anders???

andig, 2022-07-25:
These: Anbieterproblem dass sie gelöst haben. Wenns wieder auftritt gerne hier melden!

closed 2022-07-25 by andig

patbab, 2022-07-26:
Gelöst worden ist da nichts. Ich hatte erst gestern wieder den Fehler.

Du hast bei deinen Tests zwangsläufig eine wesentliche Voraussetzung für den Fehler nicht erfüllen können:

> e) Es ist noch nicht klar, ob der Roller direkt vorher gefahren sein muß, oder ob es reicht, dass der Roller z.B. >1h steht. 

Meine Tests deuten mehr und mehr darauf hin, dass der Fehler immer nur beim ersten Abrufen der Daten nach einer Bewegung des Fahrzeugs kommt. Also man kommt heim, startet das Laden, Fehler. Ursache natürlich unbekannt.

Gern kann ich Dir mal mails schreiben und ansonsten die Sache unberührt lassen, wenn ich gefahren bin, damit Du in dem Zustand testen kannst.

Gerade aufgrund dieses dubiosen Vehaltens halte ich nach wie vor eine pragmatische Lösung über https://github.com/evcc-io/evcc/issues/3847 für ebenso zielführend.

andig, 2022-07-26:
Yep, lass uns gerne adhoc mailen.