evcc-io/evcc #3927: Daheimladen mit Version ab 0.96

mergeli, 2022-07-23:
### Describe the bug

ab 0.96 wird die Daheimladen trotz RFID-Kartenmodus nicht erkannt.

### Steps to reproduce

reset der wallbox
 

### Configuration details

aus der evcc.yaml:
chargers:
- type: template
  template: daheimladen 
  token: hier ist der token von daheimladen.de --> ausgegraut
  stationid: hier steht in echt die Nr -->ausgegraut  
  name: wallbox5

### Log details

sudo evcc -l debug charger
[main  ] INFO 2022/07/23 16:56:06 evcc 0.97
[main  ] INFO 2022/07/23 16:56:06 using config file /etc/evcc.yaml
Power:          unexpected status: 401
Energy:         unexpected status: 401
Current L1..L3: unexpected status: 401
Charge status:  unexpected status: 401
Enabled:        unexpected status: 401

### What type of operating system are you running?

Linux

### Version

0.96 und 0.97

andig, 2022-07-23:
/cc @utsavanand2

utsavanand2, 2022-07-23:
Thanks @andig 

Hi @mergeli Please send a token request to info@daheimladen.de with the email address you have registered your account with.

andig, 2022-07-23:
I‘ll close this here in favor of a direct support discussion. Let me know if I can help with anything.

closed 2022-07-23 by andig

mergeli, 2022-07-24:
Danke schön. 
Kann man zwischenzeitlich auch ein downgrade auf zb 0.89 machen? Ist das irgendwo beschrieben wie man das macht?

Liebe Grüße 
Rainer Pieper

andig, 2022-07-24:
@premultiply können wir das dokumentieren?
@mergeli du kannst dir alle Versionen unter den Releases runter laden und von Hand installieren